juribeparada/MMDVM_HS #8: Teensy support

g4klx, 2017-05-22:
Is there any chance of support for the Teensy platform based on the Teensy support in the main MMDVM code base?

juribeparada, 2017-05-22:
Yes of course, but my free time is limited for a while. I bought a Teensy 3.6 last year, but the board never arrived... I think now Teensy is sold in my country, I will buy one again. If this code run on Arduino Due or Arduino Zero, it should run also on Teensy.

HL1RR, 2017-05-28:
[MMDVM@pi]~/source/DMRGateway$ ./DMRGateway ./DMRGateway.ini
I: 2017-05-28 00:19:58.712 This software is for use on amateur radio networks only,
I: 2017-05-28 00:19:58.713 it is to be used for educational purposes only. Its use on
I: 2017-05-28 00:19:58.713 commercial networks is strictly prohibited.
I: 2017-05-28 00:19:58.713 Copyright(C) 2017 by Jonathan Naylor, G4KLX and others
M: 2017-05-28 00:19:58.713 DMRGateway-20170522 is starting
M: 2017-05-28 00:19:58.713 Built 08:00:03 May 28 2017 (GitID #6f0c543)
I: 2017-05-28 00:19:58.713 MMDVM Network Parameters
I: 2017-05-28 00:19:58.713     Rpt Address: 127.0.0.1
I: 2017-05-28 00:19:58.713     Rpt Port: 62032
I: 2017-05-28 00:19:58.713     Local Address: 127.0.0.1
I: 2017-05-28 00:19:58.713     Local Port: 62031
M: 2017-05-28 00:19:58.713 DMR, Opening MMDVM Network
M: 2017-05-28 00:19:58.714 Waiting for MMDVM to connect.....

I wonder why not.
Tell us how to use it

HL1RR, 2017-05-28:
[General]
Timeout=10
RptAddress=127.0.0.1
RptPort=62032
LocalAddress=127.0.0.1
LocalPort=62031
Daemon=0
Debug=0

[Log]
# Logging levels, 0=No logging
DisplayLevel=1
FileLevel=1
FilePath=.
FileRoot=DMRGateway

[Voice]
Enabled=1
Language=en_GB
Directory=./Audio

[XLX Network 1]
Enabled=1
Address=xlx950.epf.lu
Port=62030
# Local=3351
# Options=
Slot=1
TG=8
Base=84000
Password=passw0rd
Debug=0

[XLX Network 2]
Enabled=0
Address=44.131.4.1
Port=62030
# Local=3351
# Options=
Slot=1
TG=7
Base=74000
Password=passw0rd
Debug=0
# BrandMeister
[DMR Network 1]
Enabled=0
Address=bm.dv.or.kr
#Address=44.131.4.1
Port=62031
# Local=3352
# Local cluster
TGRewrite=1,9,1,9,1
# Reflector TG on to slot 2 TG9
TGRewrite=2,9,2,9,1
# Reflector control command slot 2 94000->4000 to 95000->5000
PCRewrite=2,94000,2,4000,1001
# Echo on RF slot 1 TG9990 to network slot 1 9990
TypeRewrite=1,9990,1,9990
SrcRewrite=1,9990,1,9990,1
# Reflector status returns
SrcRewrite=2,4000,2,9,1001
# Pass all of the other private traffic on slot 1 and slot 2
PassAllPC=1
PassAllPC=2
Password=passw0rd
Debug=0

# DMR+
[DMR Network 2]
Enabled=0
Address=44.131.4.1
Port=55555
# Local=3352
# Reflector TG on to slot 2 TG8
TGRewrite=2,8,2,9,1
# Echo on slot 2 TG9990
TGRewrite=2,9990,2,9990,1
# Reflector control command slot 2 84000->4000 to 85000->5000
PCRewrite=2,84000,2,4000,1001
# Pass all of the other talk group traffic on slot 1 and slot 2
PassAllTG=1
PassAllTG=2
Password=PASSWORD
Debug=0

HL1RR, 2017-05-28:
Please help me.

juribeparada, 2017-05-28:
Hi HL1RR, sorry, this thread was open for Teensy platform support. This is not the correct place for DMRGateway. Please visit MMDVM Yahoo Group to get more information about DMRGateway.ini. MMDVM_HS + MMDVMHost with DMRGateway work OK. Anyway, probably you problem is because you didn't define Local=62032 in [DMR Network] section of MMDVM.ini.

Regards,
Andy.

HL1RR, 2017-05-28:
Thank you very much.de HL1RR 73 !

juribeparada, 2017-07-01:
Added Teensy pinout definitions to MMDVM_HS. I only tested with Teensy 3.5.

closed 2017-07-15 by juribeparada