juribeparada/MMDVM_HS #12: Display problem with serial firmware

farnouche, 2017-07-03:
Hi,

I am building my mmdvm_hs and try to reduce cables between RPi and STM32

previously I built the firmware for STM32 with USBLoader and all works very well (display etc...)

now I build the firmware for STM32 without USB bootloader and ... the law of Murphy is coming, I lost the correct display on the nextion LCD.

the LCD don't display the time line, when a call incoming through the net or from RF, the time line display the call with an offset of the correct position !!!! (without BER) 

I think after several tests that it's a bug in the STM32 serial firmware that uses the 2 uart (PA9/PA10 and PA2/PA3)

thank you for help

juribeparada, 2017-07-03:
Hi Farnouche,

at the moment I can't associate the bug that you are describing with the last changes in the MMDVM_HS firmware. Are you sure it is not the MMDVMHost version?. I will need more information: what previous MMDVM_HS version was OK?, what MMDVMHost version were you using?. What display?, etc. If you can identity the commit that causes the problem, that will help a lot.

Andy.

farnouche, 2017-07-03:
Hi Andy,

yes I am sure that the display bug is inside the mmdvm_hs firmware without usb bootloader.
unfortunely, I can't help you to identify the commit because I am a new user and I don't compiled the old versions.

the modem part work well.

I use the last version of mmdvmhost and mmdvm_hs
I compiled the firmware with the same source code (without update it) 
I only change the config.h and mmdvm.ini to use the correct interface.

- firmware with usb bootloader work well (no display problem with nextion LCD)
- firmware without usb bootloader (serial interface) display information with offset or "ghost" characters

may be a buffer overflow?

if you want, I can make you a videos ;-)

73 QRO

my hardware : 
Raspberry pi 3 + STM32F103 + ADF7021 + Nextion 3.5"
the nextion LCD is connected to the STM32

juribeparada, 2017-07-03:
OK, that makes more sense. A picture of the LCD with the problem would be fine. Unfortunately I haven't used any Display in my setup, I only added the original Jonathan's MMDVM support for serial repeater. What kind of host communication port are you using (direct serial connection to RPi or USB connection)?
You also mention this problem happen after firmware update. What firmware version were you using that it works OK?. Moreover, you may use "git checkout [revision]" command to try to find the specific commit with the problem.

farnouche, 2017-07-03:
Hi,
the version of MMDVM firmware in Gitversion.h is : c187670
the version of MMDVMHost in Gitversion.h is : d399ba9ebb5cfb85f06f0006903b2c605262a7e6
I connected directly the STM32 to the RPi with the serial connection

Firmware with "USB host"
in idle : 
[image]

when network or radio call incoming : 
[image]
[image]

*************************************

Firwmare with "USART1 host" 
in idle (first start) : 
[image]

in idle (after incoming call) : 
[image]

when network or radio call incoming : 
[image]
[image]

I hope this pictures help you to find the issue

except this little bug, the modem work very well in DMR mode ;-)

juribeparada, 2017-07-03:
OK, now I realized that Jonathan added a ring buffer to the serial repeater in MMDVM. I haven't added that here yet. I will do here during the next days.

juribeparada, 2017-07-04:
I increased the serial port buffer, please test with the "devel" branch and tell me if you find any improvement:
https://github.com/juribeparada/MMDVM_HS/tree/devel

farnouche, 2017-07-04:
Hi,

no, sorry, but the increase of buffer don't change anything.
Same bug on display

farnouche, 2017-07-04:
Hi Andy,

I am regularly your repository and I see this new version in devel branch
But the new source (6486BF4) has the same bug

I think you should investigate the difference between Virtual Com through USB and the real UART and why the UART2 don't work correctly when UART1 is used.

;-)

juribeparada, 2017-07-04:
Hi, thank you for testing. That means this bug is not related with buffer ring problems. Sorry, but his bug have to wait until buying a Nextion display for testing by myself and to perform other priority task, due to my limited time. Regards.

farnouche, 2017-07-04:
Ok Andy
I am not in a hurry and can wait ;-)
You can buy a Nextion LCD on Aliexpress or other sites 

Hope you find the bug 
best regards
73 QRO

juribeparada, 2017-07-09:
Hi, now the serial LCD should work OK. Please test.

farnouche, 2017-07-09:
Hi Andy,

Yes, you found the bug ...

Now the firmware seems to work with the 2 uart
I used the source from the master tree not the devel tree
congratulations ;-)

now I have a other issue but I think it is just because I use wires and not a real pcb,sometime I lose communication with the modem after a few secondes
I will made a real pcb

juribeparada, 2017-07-09:
Hi, yes probably that is produced by long interconnection wires. With real PCB there is no such issue.

closed 2017-07-09 by juribeparada