juribeparada/MMDVM_HS #15: MMDVM_HS dated 20/08 Sketch to large on Arduino

bill2e0, 2017-08-24:
Andy
I have just downloaded the MMDVM_HS from git hub. When I try to load it to STM32 boards with Arduino, I am getting a sketch to large message. I have tried it on 2 different PC's with the same result.
Cheers
Bill
[image]

phl0, 2017-08-24:
You are using an STM32F103 board obviously. AFAIK it has to be an STM32F108.

2e0dqd, 2017-08-24:
It should run on a STM32F103

N4IRS, 2017-08-24:
I am wondering why your result is bigger then 64K Here is my result I got from DLing the master branch.

Sketch uses 64088 bytes (97%) of program storage space. Maximum is 65536 bytes.
Global variables use 7104 bytes of dynamic memory.

73, Steve N4IRS

juribeparada, 2017-08-24:
Hi All, compiled with gcc under Linux or Mac the bin size is 49136 bytes. Last weekend I added some additional runtime debug messages, that uses a standard C library. Probably this is too bulky under Arduino / stm32duino. I will correct this soon. Try to compile a previous commit.

73, Andy CA6JAU.

bill2e0, 2017-08-24:
Previous commit compiles fine, bin size around 49k bytes. Just wanted this with the quiet option behavior to mode leds. 
73, Bill  2E0FEI.

juribeparada, 2017-08-24:
Try again, I removed the bulky function.  I will fix better later.

bill2e0, 2017-08-24:
Thank you Andy, perfect now.
73, Bill 2E0FEI
[image]

closed 2017-09-14 by bill2e0