juribeparada/MMDVM_HS #16: Compile Firmware Errors

w1msg, 2017-09-16:
I have been following the directions for compiling the firmware for my ZumSpot Libre USB. However after following the instructions when I do make bl I get a long stream of not declared errors and it Fails. Here is the output ;
root@pi-star3(rw):MMDVM_HS# make bl
echo "#define GITVERSION \"2beb3d7\"" > GitVersion.h
arm-none-eabi-gcc -c -mcpu=cortex-m3 -march=armv7-m -mthumb -Wall -Wextra -I. -ISTM32F10X_Lib/CMSIS/ -ISTM32F10X_Lib/Device/ -ISTM32F10X_Lib/ST                                                                                  M32F10x_StdPeriph_Driver/inc/ -ISTM32F10X_Lib/usb/inc/ -DCUSTOM_NEW -DNO_EXCEPTIONS -Wno-unused-parameter -DUSE_STDPERIPH_DRIVER -DSTM32F10X_MD                                                                                   -DHSE_VALUE=8000000 -DVECT_TAB_OFFSET=0x2000 -DMADEBYMAKEFILE -Os -ffunction-sections -fdata-sections -nostdlib STM32F10X_Lib/STM32F10x_StdPer                                                                                  iph_Driver/src/misc.c -o STM32F10X_Lib/STM32F10x_StdPeriph_Driver/src/misc.o
Compiled STM32F10X_Lib/STM32F10x_StdPeriph_Driver/src/misc.c!

arm-none-eabi-gcc -c -mcpu=cortex-m3 -march=armv7-m -mthumb -Wall -Wextra -I. -ISTM32F10X_Lib/CMSIS/ -ISTM32F10X_Lib/Device/ -ISTM32F10X_Lib/ST                                                                                  M32F10x_StdPeriph_Driver/inc/ -ISTM32F10X_Lib/usb/inc/ -DCUSTOM_NEW -DNO_EXCEPTIONS -Wno-unused-parameter -DUSE_STDPERIPH_DRIVER -DSTM32F10X_MD                                                                                   -DHSE_VALUE=8000000 -DVECT_TAB_OFFSET=0x2000 -DMADEBYMAKEFILE -Os -ffunction-sections -fdata-sections -nostdlib STM32F10X_Lib/STM32F10x_StdPer                                                                                  iph_Driver/src/stm32f10x_iwdg.c -o STM32F10X_Lib/STM32F10x_StdPeriph_Driver/src/stm32f10x_iwdg.o
Compiled STM32F10X_Lib/STM32F10x_StdPeriph_Driver/src/stm32f10x_iwdg.c!

arm-none-eabi-gcc -c -mcpu=cortex-m3 -march=armv7-m -mthumb -Wall -Wextra -I. -ISTM32F10X_Lib/CMSIS/ -ISTM32F10X_Lib/Device/ -ISTM32F10X_Lib/ST                                                                                  M32F10x_StdPeriph_Driver/inc/ -ISTM32F10X_Lib/usb/inc/ -DCUSTOM_NEW -DNO_EXCEPTIONS -Wno-unused-parameter -DUSE_STDPERIPH_DRIVER -DSTM32F10X_MD                                                                                   -DHSE_VALUE=8000000 -DVECT_TAB_OFFSET=0x2000 -DMADEBYMAKEFILE -Os -ffunction-sections -fdata-sections -nostdlib STM32F10X_Lib/STM32F10x_StdPer                                                                                  iph_Driver/src/stm32f10x_usart.c -o STM32F10X_Lib/STM32F10x_StdPeriph_Driver/src/stm32f10x_usart.o
Compiled STM32F10X_Lib/STM32F10x_StdPeriph_Driver/src/stm32f10x_usart.c!

arm-none-eabi-gcc -c -mcpu=cortex-m3 -march=armv7-m -mthumb -Wall -Wextra -I. -ISTM32F10X_Lib/CMSIS/ -ISTM32F10X_Lib/Device/ -ISTM32F10X_Lib/ST                                                                                  M32F10x_StdPeriph_Driver/inc/ -ISTM32F10X_Lib/usb/inc/ -DCUSTOM_NEW -DNO_EXCEPTIONS -Wno-unused-parameter -DUSE_STDPERIPH_DRIVER -DSTM32F10X_MD                                                                                   -DHSE_VALUE=8000000 -DVECT_TAB_OFFSET=0x2000 -DMADEBYMAKEFILE -Os -ffunction-sections -fdata-sections -nostdlib STM32F10X_Lib/STM32F10x_StdPer                                                                                  iph_Driver/src/stm32f10x_exti.c -o STM32F10X_Lib/STM32F10x_StdPeriph_Driver/src/stm32f10x_exti.o
Compiled STM32F10X_Lib/STM32F10x_StdPeriph_Driver/src/stm32f10x_exti.c!

arm-none-eabi-gcc -c -mcpu=cortex-m3 -march=armv7-m -mthumb -Wall -Wextra -I. -ISTM32F10X_Lib/CMSIS/ -ISTM32F10X_Lib/Device/ -ISTM32F10X_Lib/ST                                                                                  M32F10x_StdPeriph_Driver/inc/ -ISTM32F10X_Lib/usb/inc/ -DCUSTOM_NEW -DNO_EXCEPTIONS -Wno-unused-parameter -DUSE_STDPERIPH_DRIVER -DSTM32F10X_MD                                                                                   -DHSE_VALUE=8000000 -DVECT_TAB_OFFSET=0x2000 -DMADEBYMAKEFILE -Os -ffunction-sections -fdata-sections -nostdlib STM32F10X_Lib/STM32F10x_StdPer                                                                                  iph_Driver/src/stm32f10x_gpio.c -o STM32F10X_Lib/STM32F10x_StdPeriph_Driver/src/stm32f10x_gpio.o
Compiled STM32F10X_Lib/STM32F10x_StdPeriph_Driver/src/stm32f10x_gpio.c!

arm-none-eabi-gcc -c -mcpu=cortex-m3 -march=armv7-m -mthumb -Wall -Wextra -I. -ISTM32F10X_Lib/CMSIS/ -ISTM32F10X_Lib/Device/ -ISTM32F10X_Lib/ST                                                                                  M32F10x_StdPeriph_Driver/inc/ -ISTM32F10X_Lib/usb/inc/ -DCUSTOM_NEW -DNO_EXCEPTIONS -Wno-unused-parameter -DUSE_STDPERIPH_DRIVER -DSTM32F10X_MD                                                                                   -DHSE_VALUE=8000000 -DVECT_TAB_OFFSET=0x2000 -DMADEBYMAKEFILE -Os -ffunction-sections -fdata-sections -nostdlib STM32F10X_Lib/STM32F10x_StdPer                                                                                  iph_Driver/src/stm32f10x_rcc.c -o STM32F10X_Lib/STM32F10x_StdPeriph_Driver/src/stm32f10x_rcc.o
Compiled STM32F10X_Lib/STM32F10x_StdPeriph_Driver/src/stm32f10x_rcc.c!

arm-none-eabi-gcc -c -mcpu=cortex-m3 -march=armv7-m -mthumb -Wall -Wextra -I. -ISTM32F10X_Lib/CMSIS/ -ISTM32F10X_Lib/Device/ -ISTM32F10X_Lib/ST                                                                                  M32F10x_StdPeriph_Driver/inc/ -ISTM32F10X_Lib/usb/inc/ -DCUSTOM_NEW -DNO_EXCEPTIONS -Wno-unused-parameter -DUSE_STDPERIPH_DRIVER -DSTM32F10X_MD                                                                                   -DHSE_VALUE=8000000 -DVECT_TAB_OFFSET=0x2000 -DMADEBYMAKEFILE -Os -ffunction-sections -fdata-sections -nostdlib STM32F10X_Lib/usb/usb_regs.c -                                                                                  o STM32F10X_Lib/usb/usb_regs.o
Compiled STM32F10X_Lib/usb/usb_regs.c!

arm-none-eabi-gcc -c -mcpu=cortex-m3 -march=armv7-m -mthumb -Wall -Wextra -I. -ISTM32F10X_Lib/CMSIS/ -ISTM32F10X_Lib/Device/ -ISTM32F10X_Lib/ST                                                                                  M32F10x_StdPeriph_Driver/inc/ -ISTM32F10X_Lib/usb/inc/ -DCUSTOM_NEW -DNO_EXCEPTIONS -Wno-unused-parameter -DUSE_STDPERIPH_DRIVER -DSTM32F10X_MD                                                                                   -DHSE_VALUE=8000000 -DVECT_TAB_OFFSET=0x2000 -DMADEBYMAKEFILE -Os -ffunction-sections -fdata-sections -nostdlib STM32F10X_Lib/usb/usb_cdcacm.c                                                                                   -o STM32F10X_Lib/usb/usb_cdcacm.o
Compiled STM32F10X_Lib/usb/usb_cdcacm.c!

arm-none-eabi-gcc -c -mcpu=cortex-m3 -march=armv7-m -mthumb -Wall -Wextra -I. -ISTM32F10X_Lib/CMSIS/ -ISTM32F10X_Lib/Device/ -ISTM32F10X_Lib/ST                                                                                  M32F10x_StdPeriph_Driver/inc/ -ISTM32F10X_Lib/usb/inc/ -DCUSTOM_NEW -DNO_EXCEPTIONS -Wno-unused-parameter -DUSE_STDPERIPH_DRIVER -DSTM32F10X_MD                                                                                   -DHSE_VALUE=8000000 -DVECT_TAB_OFFSET=0x2000 -DMADEBYMAKEFILE -Os -ffunction-sections -fdata-sections -nostdlib STM32F10X_Lib/usb/usb_mem.c -o                                                                                   STM32F10X_Lib/usb/usb_mem.o
Compiled STM32F10X_Lib/usb/usb_mem.c!

arm-none-eabi-gcc -c -mcpu=cortex-m3 -march=armv7-m -mthumb -Wall -Wextra -I. -ISTM32F10X_Lib/CMSIS/ -ISTM32F10X_Lib/Device/ -ISTM32F10X_Lib/ST                                                                                  M32F10x_StdPeriph_Driver/inc/ -ISTM32F10X_Lib/usb/inc/ -DCUSTOM_NEW -DNO_EXCEPTIONS -Wno-unused-parameter -DUSE_STDPERIPH_DRIVER -DSTM32F10X_MD                                                                                   -DHSE_VALUE=8000000 -DVECT_TAB_OFFSET=0x2000 -DMADEBYMAKEFILE -Os -ffunction-sections -fdata-sections -nostdlib STM32F10X_Lib/usb/usb_core.c -                                                                                  o STM32F10X_Lib/usb/usb_core.o
Compiled STM32F10X_Lib/usb/usb_core.c!

[... cut ...]
       if (m_poPtr >= m_poLen) {
           ^
DMRDMOTX.cpp: At global scope:
DMRDMOTX.cpp:78:1: error: 'uint8_t' does not name a type
 uint8_t CDMRDMOTX::writeData(const uint8_t* data, uint8_t length)
 ^
DMRDMOTX.cpp:93:27: error: variable or field 'writeByte' declared void
 void CDMRDMOTX::writeByte(uint8_t c)
                           ^
DMRDMOTX.cpp:93:27: error: 'uint8_t' was not declared in this scope
Makefile:163: recipe for target 'DMRDMOTX.o' failed
make: *** [DMRDMOTX.o] Error 1
root@pi-star3(rw):MMDVM_HS#

juribeparada, 2017-09-17:
It looks like a problem with your Config.h file, could you copy this file here?. I don't have any problem to compile the firmware here.

w1msg, 2017-09-17:
Yes a mistake on my part, I didnt understand the instructions fully. It 
compiled just fine here after I left the Config.h file alone.

Question does the serial repeater work for connection of a Nextion 
display to the ZumSpot board ?

Thanks

Craig

> Andy CA6JAU <mailto:notifications@github.com>
> September 17, 2017 at 9:08 AM
>
> It looks like a problem with your Config.h file, could you copy this 
> file here?. I don't have any problem to compile the firmware here.

juribeparada, 2017-09-17:
Yes, no problem with Nextion and ZUMspot (Serial Repeater).

w1msg, 2017-09-17:
Thanks, sorry for the confusion and my errors !

All set up and working Great . Will be posting a video on the Pi-Star 
group on Facebook shortly.

Thanks

Craig

> Andy CA6JAU <mailto:notifications@github.com>
> September 17, 2017 at 9:21 AM
>
> Yes, no problem with Nextion and ZUMspot (Serial Repeater).

closed 2017-09-23 by juribeparada