juribeparada/MMDVM_HS #17: How to calibrate frequency?

mattames, 2017-10-16:
The documentation states:-

"The common solution is to adjust RX frequency offset (RXOffset) in your MMDVM.ini file. Please try with steps of +-100 Hz until you get low BER. If you don't have test equipment, the only procedure is trial and error. "

I have plenty of great test gear - but how on earth do you put the MMDVM-HS into a calibration mode? adjusting the RX offset by trial error is a bit time consuming, when I know that the offset is going to be due to the clock running the 7021.

How can I get the MMDVM_HS to transmit a constant tone or similar so I can measure it's frequency offset?

juribeparada, 2017-10-16:
For the few ones that have calibrated test equipment, you can enable TEST_TX feature. Please see "Hidden functions" in BUILD.md document.

mattames, 2017-10-16:
Thanks for the pointer. Just to clarify, the process is to build the firmware with the TEST_TX define enabled, load this firmware into the MMDVM_HS, the radio then transmits the carrier constantly in this mode (?), then fetermine the frequency offset, then load the regular firmware back into the unit?

juribeparada, 2017-10-17:
Yes, exactly. The only important thing is to have DStar enabled in MMDVM.ini. This is because TX carrier only works in GMSK mode for ADF7021.

closed 2017-10-30 by juribeparada

narspt, 2018-04-21:
Sorry to "bump" this, but I don't want to create a new issue for my question... I did already adjust some mmdvm_hs hotspots for DMR, either by using TEST_TX or now using the easy and great mmdvmcal tool, that is now possible with the latest fw, I can tell that I could always easily adjust them successfully to very consistent 0.1-0.3% BER... however yesterday I tried the same procedure to adjust a friend's hotspot for YSF instead and I'm really unable to get consistent values... the audio is apparently ok... but I get almost random BER like 0.5-3.x... I did also try the method of just trying values in +/-50Hz steps, but I never get consistent BER values like the ones I get on DMR... do you have any idea? Thanks.

narspt, 2018-04-21:
Thank you for your reply and confirmation Andy, good to know that it's not only with me.