juribeparada/MMDVM_HS #19: Is CW ID implemented ?

rogerclarkmelbourne, 2017-12-06:
As far as I can tell CW ID is not implemented 

https://github.com/juribeparada/MMDVM_HS/blob/74f9bcf33d124872c56261ed0b14f6d2675ad316/SerialPort.cpp#L463-L464

Is this correct, or am I looking at the wrong bit of code?

I searched on CW but that's all I could find.

If its not implemented, is there a technical reason why not ? Or simply lack of time to do it ?

rogerclarkmelbourne, 2017-12-07:
Hi Andy. I did a bit more research and it looks like I need to port over Jonathan's cwidTx files to your repo

I may try adding them if i have time.

If not I may just do a PR which has

```
          case MMDVM_CAL_DATA:
			DEBUG2("MMDVM_CAL_DATA NOT SUPPORTED", err);
			sendNAK(err);
            break;

          case MMDVM_SEND_CWID:
		      DEBUG2("MMDVM_SEND_CWID NOT SUPPORTED", err);
			  sendNAK(err);
            break;
```

marrold, 2017-12-07:
I could be _completely_ wrong, but because MMDVM effectively communicates over 'Analog' to the attached radios it can generate the 'tones', where as MMDVM_HS is communicating over SPI, so I don't think it's as straight forward as porting the cwidTx files over.

I'm sure someone with a better knowledge of the underlying code will be able to advise shortly.

rogerclarkmelbourne, 2017-12-07:
Thanks @marrold 

I'd not looked in detail about how MMDVM does it, but I see what you are saying.

I don't know enough about the ADF7021 to know if its TX can be turned on and not send any data, but I suspect its possible because that sort of thing is required when doing FCC certification testing.

I'll take a look at the datasheet for the 7021

rogerclarkmelbourne, 2017-12-07:
Interesting..

Page 29 of the data sheet describes

```
TEST PATTERN GENERATOR

000	Transmit Carrier	
001	Transmit +	 f	DEV	 tone 	
010	Transmit 	− f	DEV	 tone 	
011	Transmit 1010 	pattern	
100	Transmit PN9 sequence	
101	Transmit SWD pattern repeatedly	
```

etc

So it should be possible to put it into that mode and use codes 001 and 000 to alternate , tone and no tone

rogerclarkmelbourne, 2017-12-07:
Just a bit of an update.

I found the existing TX test mode, and have taken the code and used it when the CW_ID is requested, and I can turn on the TX and just send a carrier.

I've not had time to do much else, but in the spec it seems to say a tone can be generated if one of the other registers has the tone value.

So I'll need to investigate this, because at the moment if I enable carrier + tone I just appear to get carrier on my FM receiver.

juribeparada, 2017-12-22:
Hi Roger and Matthew,

sorry for my late response, but I was not connected to Internet for a long time because I was in a Antarctic expedition (it's part of my work). I've just returned yesterday, and probably I will take some days for resting.

Actually I didn't implement CW ID because I thought it will not be so useful for a personal hotspot, and there were other priority task to complete at the time I was writing the firmware code. Now I think I can implement. As Matthew said, you need to generate a sine tone FM modulation for CW ID, and that doesn't exist directly in ADF7021. But I think it will be possible to generate a sine approximation using the -3, -1, +1 and +3 symbol generation of 4FSK. There will not be a way to filter this tone to eliminate any harmonics in the FM modulation, and then probably will sound a little distorted, depending on the frequency tone choice. I will try to test in a followings days.

Regards,
Andy.

Radioguy817, 2017-12-25:
While FM would be the preferred method for the CW ID, in the US, a legal ID can be simple OOK CW, so you might be able to just turn the carrier ON and OFF using tge callsign

juribeparada, 2017-12-25:
OK, good to know, but I think I will implement the most known method using FM for the CW ID.

Radioguy817, 2017-12-25:
Certainly the preferred method as that is how a commercial Mot repeater works.
Just wanted to give u an option

rogerclarkmelbourne, 2017-12-26:
Guys

Sorry I've not replied... Things have been a bit busy in the lead up to Christmas

Re: FM CW.

I thought perhaps the test signal generation commands would be able to generate FM, but perhaps it generates FSK not FM.

I only did some very basic tests and was not able to generate any tone at all using the Test mode, i.e just CW.

I presumed perhaps I needed to also set the frequency in one of the other registers, but I've not had time to experiment.

juribeparada, 2017-12-27:
Hi All,

I've just added the CW ID support (FM), based on Jonathan code and using a 4FSK sequence to simulate a sine waveform. It sounds good. Please test and report any bugs, using one mode or using scan modes.

Regards

rogerclarkmelbourne, 2017-12-28:
I pulled the latest version and it built OK.

But it wasnt compatible with the version of MMDVMHost.exe that I was running.

So I had to rebuild MMDVMHost exe using Visual Studio (in my case I have VS2012 so I had to downgrade the project build target, but it did build in the end)

Anyway. The CW is definitely working, very well and sounds very clear on my FM transceiver. (Baofung UV-82)

The only thing I've not done is test what happens during a QSO. I presume that he Host would not send the CW ID command if it was sending data

juribeparada, 2017-12-28:
Good to know the CW ID is working OK, thank you for testing. The CW ID will work only when the modem is in IDLE state, then it should not be a problem during a QSO. Anyway, it is good to test in this condition.

Radioguy817, 2017-12-28:
Please excuse my ignorance, but i have a question. I have a Zumspot RPi Zero board. Will this FW work on that board? Am i to upgrade the board FW to get CW ID? Or PiStar? Do i have to do some linux build process to get this FW? I am not a sw person so hence the questions.

juribeparada, 2017-12-28:
Yes, this firmware is for ZUMspot, MMDVM_HS_Hat and for any other MMDVM_HS homebrew hotspots. For the moment you need to build the firmware, please see "Build the firmware and upload to ZUMspot RPi" in the README.md file, or wait until I launch a binary release of the firmware soon (after more testing of CW ID), for automated upload process.

Radioguy817, 2017-12-28:
Thanks for the explanation. I will wait for the binary. I am terrible with sw. Good with RF hardware and system engineering.

rogerclarkmelbourne, 2017-12-28:
One thing I noticed was that after I updated to the latest sources (firmware) that my existing PC MMDVMHost software would not communicate with latest firmware.

It could be I was using a very old version of MMDVMHost (emailed to me by G4TSN), but its potentially an issue on the RPi Zero depending on when that software was built

N4IRS, 2017-12-28:
Roger,
Just in case, the current source is at https://github.com/g4klx/MMDVMHost
73, Steve N4IRS

juribeparada, 2017-12-28:
Roger, probably your MMDVMHost is too old. This firmware is updated using the latest G4KLX modem protocol, probably the problem is due to the new length of the MMDVM_SET_CONFIG command.

rogerclarkmelbourne, 2017-12-28:
@N4IRS 
Thanks

I downloaded the latest sources and rebuilt them yesterday. (I had to make a slight modification to the project file as I have to use VS 2012 for some contract work, so can't install VS2017. But its only a minor difference in the settings, and it seems to build and run fine using VS2012

@juribeparada 
Yes. It was the set config command.

I'll need to let G4TSN know his MMDVM Host exe is quite old and he should update it.
I'll probably send him the exe I built.

I'm also in the process of updating the ZumSpot branch of the Arduino_STM32 core, to merge in loads of commits, so its a bit more up to date.
The only difference between the ZumSpot branch and the master is a minor change to the USB Serial handling because the MMDVMHost does not set the control signals (I think its DTR), and the normal Arduino_STM32 core will not send data unless it thinks the PC is ready

So it was easy to merge in all 330 commits that the ZumSpot branch was behind, then manually change a few lines in the USB code.

However, I've not actually tested having a QSO with the updated firmware, I have only listened, and channels were too busy with QSO's for me to do a test transmission.
Hence I've not pushed the change to the Zumspot branch of Arduino_STM32 back to github

juribeparada, 2017-12-29:
Thank you Roger to update the ZUMspot branch of Arduino_STM32.

rogerclarkmelbourne, 2017-12-29:
Hi Andy

I hope to fully test the Zumspot today, then I will push the updates

rogerclarkmelbourne, 2017-12-29:
Andy

I've pushed the updates to the ZumSpot branch, after testing it with my Zumspot

Seems to work fine.

juribeparada, 2017-12-30:
Thank you Roger. I also restored the set config length, because in this firmware I am not using the new parameters added by Jonathan. Now I am preparing a RF power level control from MMDVMHost.

rogerclarkmelbourne, 2017-12-30:
OK.

I've now clicked the button to "Watch" this repo, so I'll see next time you update ;-)

rogerclarkmelbourne, 2018-01-05:
The CW ID (missing) is now fixed, so I will close this issue

closed 2018-01-05 by rogerclarkmelbourne