juribeparada/MMDVM_HS #20: power adjust?

jamesbrownfr, 2017-12-31:
Hi,

I see your new release for the power adjust of the hotspot
but I think you make a little mistake in your code (or not LOL)

in your last release 
IO.cpp : `m_power = rf_power;` 

and in G4klx mmdvmhost
MMDVMHost.cpp : `LogInfo("    RF Level: %.1f%%", rfLevel);`
Modem.cpp  : `buffer[12U]  = (unsigned char)(m_rfLevel * 2.55F + 0.5F); `

ok great, we can adjust the power in pourcentage with the maximum power
100% ==> 256 in m_power value

but in ADF7021.cpp 
`ADF7021_REG2 |= (uint32_t) m_power       << 13;  // power level`

so : 
256 ($FF) ... FF << 13 ==> REG2 : xxxxxxxx xxxxx111 111xxxxx xxxxxxxx
63 ($3F) ... 3F << 13 ==> REG2 : xxxxxxxx xxxxx111 111xxxxx xxxxxxxx
oups ;-)

have I forget some codes?

best regards

juribeparada, 2017-12-31:
Sorry, I cant see the mistake, you mean the <<13 ?

juribeparada, 2017-12-31:
OK, I see, it's an old bug, power register is 6 bits, not 8 bits. I will fix.

jamesbrownfr, 2017-12-31:
Hi,

yes, in my mind, it's clear LOL
the rf power in REG2 is only 6 bits (between 0 to 63)
but in the mmdvmhost, the rf power is from 0 to 255

so we have 3 value for +13dBM (REG2 PA LEVEL is 111111)

I think a little change of scale must be better in IO.cpp file :-p (8 bits to 6 bits mean to divise by 4)
`m_power = rf_power / 4;` 
it would work ... or not

like this 
100% : rf_power = 255 ---> m_power = 63 ---> +13 dBm
50% : rf_power = 128 ---> m_power = 32

4% in the minimum because 
- 0%  : rf_power = 0 ---> m_power = 0 ---> PA off
- 4% : rf_power = 4 ---> m_power = 1 ---> -16 dBm

best regard and happy new year

juribeparada, 2017-12-31:
OK, I fixed the problem, please check. I put the change inside ADF7021.cpp instead of IO.cpp, because IO.cpp should be more generic (for future RF chip support).

Thank you and happy new year!

rogerclarkmelbourne, 2018-01-01:
Happy New Year Andy..

I know its not New Year for you get. Its been 2018 here for over 11 hours.

I'm putting up a new antenna at the moment, but I'll try pulling and building later.

BTW. How do test the power ?

I see in the init file, mine has 

Power=1

Is this minimum power ?

Is max power 255 or 100 in this setting in MMDVMHost ?

juribeparada, 2018-01-01:
Hi Roger,
Happy New Year! .Yes, here is still 2017, only left 2 hours.
"Power" parameter is for info to the network (in Watts). Please see the new parameter "RFLevel" in the Modem section. The value range is 0-100 %.

rogerclarkmelbourne, 2018-01-01:
Thanks Andy

Looks like my ini file needs to be update to match the version of MMDVMHost I am now using

rogerclarkmelbourne, 2018-01-01:
I tried to do some range testing with setting

RFLevel=100 and  RFLevel=1 in the MMDVM HOST ini file

But it didnt seem to make any difference to the range.

The signal dropped at a distance of around 500m from home (when I walked around the neighborhood with my GD-77

But I know whether changing the power setting would make a big difference to the range, or whether I have a problem with my MMDVM Host settings etc

rogerclarkmelbourne, 2018-01-01:
Just looked at my MMDVM Host output and I get

```
M: 2018-01-01 04:37:14.962 MMDVMHost-20170719 is starting
M: 2018-01-01 04:37:14.962 Built 16:29:42 Dec 28 2017 (GitID #0514f06)
I: 2018-01-01 04:37:14.962 General Parameters
I: 2018-01-01 04:37:14.962     Callsign: VK3KYY
I: 2018-01-01 04:37:14.962     Id: 0
I: 2018-01-01 04:37:14.962     Duplex: no
I: 2018-01-01 04:37:14.962     Timeout: 600s
I: 2018-01-01 04:37:14.962     D-Star: enabled
I: 2018-01-01 04:37:14.962     DMR: enabled
I: 2018-01-01 04:37:14.962     YSF: enabled
I: 2018-01-01 04:37:14.962     P25: disabled
I: 2018-01-01 04:37:14.962 Modem Parameters
I: 2018-01-01 04:37:14.962     Port: \\.\COM9
I: 2018-01-01 04:37:14.962     RX Invert: no
I: 2018-01-01 04:37:14.962     TX Invert: yes
I: 2018-01-01 04:37:14.962     PTT Invert: no
I: 2018-01-01 04:37:14.962     TX Delay: 100ms
I: 2018-01-01 04:37:14.962     RX Offset: 0Hz
I: 2018-01-01 04:37:14.962     TX Offset: 0Hz
I: 2018-01-01 04:37:14.962     RX DC Offset: 0
I: 2018-01-01 04:37:14.962     TX DC Offset: 0
I: 2018-01-01 04:37:14.962     DMR Delay: 0 (0.0ms)
I: 2018-01-01 04:37:14.962     RX Level: 50.0%
I: 2018-01-01 04:37:14.962     CW Id TX Level: 50.0%
I: 2018-01-01 04:37:14.962     D-Star TX Level: 50.0%
I: 2018-01-01 04:37:14.962     DMR TX Level: 50.0%
I: 2018-01-01 04:37:14.962     YSF TX Level: 50.0%
I: 2018-01-01 04:37:14.962     P25 TX Level: 50.0%
I: 2018-01-01 04:37:14.962     RX Frequency: 439150000Hz (439150000Hz)
I: 2018-01-01 04:37:14.962     TX Frequency: 439150000Hz (439150000Hz)
M: 2018-01-01 04:37:14.962 Opening the MMDVM
```

So I can't see it sending the RF Power with my modem section set to

```
[Modem]
# Port=/dev/ttyACM0
Port=\\.\COM9
TXInvert=1
RXInvert=0
PTTInvert=0
TXDelay=100
DMRDelay=0
RXLevel=50
TXLevel=50
RXDCOffset=0
TXDCOffset=0
RFLevel=100
# CWIdTXLevel=50
# D-StarTXLevel=50
# DMRTXLevel=50
# YSFTXLevel=50
# P25TXLevel=50
RSSIMappingFile=RSSI.dat
SamplesDir=.
Trace=0
Debug=0
```

jamesbrownfr, 2018-01-01:
Hi rogerclarkmelbourne

upgrade your MMDVMHost with the last release because you don't have it
in Modem Parameters, you must see a line with RF Level : 100% (Not RX)

rogerclarkmelbourne, 2018-01-01:
OK
I pulled the latest sources a few days ago, so I thought it was up to date

But I'll pull again and see if anything changes

KD4EG, 2018-01-09:
I upgraded today and the RFLevel does not show in the INI file.  How do you add that line to the file?  Sorry, I am not a code writer.  I am running ZumSpot Using pi-star version 3.4.8 dashboard version 20180108. Updated 1-8-2018 this morning Zumspot: V1.1.0.

juribeparada, 2018-01-09:
Hi, you need to enter to SSH console and edit ini file directly:

sudo nano /etc/mmdvmhost

and adding the RFLevel=100 in the [Modem] section

Maybe it would be better to ask at: https://www.facebook.com/groups/pistar/

closed 2018-01-19 by juribeparada