juribeparada/MMDVM_HS #21: get bad with the latest version.

ea7gwc, 2018-01-01:
Yesterday, update to version 1.0.5 but I have seen
that my tytera 380 does not get ok.

I have returned to the version of day 30, 840462fa, and again everything OK

I also have a Hytera pd375 although this one receives but the audio
That's wrong.

73. of ea7gwc.

juribeparada, 2018-01-01:
Did you use latest version of MMDVMHost?
If so, did you set the new MMDVM.ini parameter RFLevel=100?

ea7gwc, 2018-01-01:
Yes Andy, I have the last MMDVMhost.
In fact with the version of day 30 works well.

juribeparada, 2018-01-01:
Could you enable Debug=1 in [Modem] and copy the following information? (with day 30 version and the latest):
M: 2017-12-30 14:58:45.871 Debug: MMDVM_HS FW configuration:
M: 2017-12-30 14:58:45.877 Debug: TX freq (Hz): 146999981
M: 2017-12-30 14:58:45.878 Debug: RX freq (Hz): 147000025
M: 2017-12-30 14:58:45.884 Debug: Power set: 255
M: 2017-12-30 14:58:45.884 Debug: D-Star dev (Hz): 1209
M: 2017-12-30 14:58:45.890 Debug: DMR +1 sym dev (Hz): 646
M: 2017-12-30 14:58:45.890 Debug: YSF_H +1 sym dev (Hz): 900
M: 2017-12-30 14:58:45.897 Debug: YSF_L +1 sym dev (Hz): 450
M: 2017-12-30 14:58:45.897 Debug: P25 +1 sym dev (Hz): 618

I have a MD380 and works OK with latest version.

ea7gwc, 2018-01-01:
Si en cuanto pueda te la pongo.
Mi TYT es un poco especial...........pongo el modulador en DMR en 24 y
desde todas estas ultimas versiones todo funciona correctamente tanto en tyt como
con el Hytera.

juribeparada, 2018-01-01:
OK, entonces otra prueba que puedes realizar es cambiar la desviación de DMR de 23 (por defecto) a 27. Si tu cambiabas a 24, con la versión del día 30 se producía un error de sobre-escritura del registro de desviación, y el valor resultante de "24" es 27. Claramente tu equipo está descalibrado en cuanto de desviación. Saludos y Feliz Año Nuevo.

ea7gwc, 2018-01-01:
commit 840462fafec dia 30.

2018-01-01 22:30:30.483 Debug: MMDVM_HS FW configuration:
M: 2018-01-01 22:30:30.488 Debug: TX freq (Hz): 438474993
M: 2018-01-01 22:30:30.488 Debug: RX freq (Hz): 438474925
M: 2018-01-01 22:30:30.493 Debug: Power set: 255
M: 2018-01-01 22:30:30.493 Debug: D-Star dev (Hz): 1209
M: 2018-01-01 22:30:30.498 Debug: DMR +1 sym dev (Hz): 675
M: 2018-01-01 22:30:30.498 Debug: YSF_H +1 sym dev (Hz): 900
M: 2018-01-01 22:30:30.503 Debug: YSF_L +1 sym dev (Hz): 450
M: 2018-01-01 22:30:30.503 Debug: P25 +1 sym dev (Hz): 618
M: 2018-01-01 22:30:40.580 DMR, Logged into the master successfully

ea7gwc, 2018-01-01:
Quieres decir que en vez de 24 ponga 27.....? con 24 funciona bien en la version del dia 30
Con la que falla es con la del 31.... 1.0.5
Bueno voy a probar ya te cuento.

juribeparada, 2018-01-01:
Asi, 27 con la ultima version y me cuentas

ea7gwc, 2018-01-01:
Ya esta a 27

MMDVM_HS-v1.0.5 20171231 ADF7021 FW by CA6JAU GitID #7ce1d91

018-01-01 22:17:44.702 Debug: MMDVM_HS FW configuration:
M: 2018-01-01 22:17:44.708 Debug: TX freq (Hz): 438474993
M: 2018-01-01 22:17:44.708 Debug: RX freq (Hz): 438474925
M: 2018-01-01 22:17:44.713 Debug: Power set: 63
M: 2018-01-01 22:17:44.713 Debug: D-Star dev (Hz): 1209
M: 2018-01-01 22:17:44.718 Debug: DMR +1 sym dev (Hz): 759
M: 2018-01-01 22:17:44.718 Debug: YSF_H +1 sym dev (Hz): 900
M: 2018-01-01 22:17:44.723 Debug: YSF_L +1 sym dev (Hz): 450
M: 2018-01-01 22:17:44.723 Debug: P25 +1 sym dev (Hz): 618

De momento se escucha haber que tal con el HYTERA que con ese no tenia
problemas con 23.
No se como ajustar el TYT...........antes lo que hacia era bajar la frecuencia de RX en el TYT
1000hz.

Voy a chequearlo.............
Feliz año nuevo.

ea7gwc, 2018-01-09:
Andy a 27 funciona y es la solución de compromiso para poder usar el TYT y el hytera.

Gracias.

juribeparada, 2018-01-10:
OK, realmente 27 es tu verdadero valor para tu radio. Claramente tienes una radio desajustada. Pienso en un tiempo mas habilitar el soporte para MMDVMCal en MMDVM_HS, así vas a poder modificar la desviación desde MMDVM.ini sin tener que modificar el código fuente del firmware.
Saludos

narspt, 2018-01-15:
Sorry to ask on your issue but the change you are talking about is ADF7021_DEV_DMR at ADF7021.h, right? also ea7gwc what did happen to you exactly if you didn't change deviation?

I ask it because I do also have some issue with my md380... I did calibrate frequency offsets (both, RX one by finding the lowest BER value and the TX one by using TEST_TX, and they both match same offset -439), TX from my radio is received by the hotspot OK with stable 0.0-0.2 BER and I do get perfect reports... however RX on my md380 from the hotspot is not perfect there are frequent glitches on the audio that I'm not sure how to solve... sometimes I hear people like having "tremulous" voice, is that anything similar to your problem?

ea7gwc, 2018-01-15:
My problems that my tyt is not able to receive at
Hostpot. RX LED green but without sound

In order for me to receive I have to configure the tyt on an RX frequency
1000hz lower.

Following the instructions of Andy I have changed the file ADF7021.h.

// Deviation of modulator (REG 02)
#define ADF7021_DEV_DSTAR 43U
#define ADF7021_DEV_DMR 23U <--------- Changed to 27
#define ADF7021_DEV_YSF_L 16U
#define ADF7021_DEV_YSF_H 32U

and already receives ok.

I have no problems with my hytera pd375.

There is another friend in my city with the same problem and this solution
It is also valid.

It's a problem with this particular TYT.

In TX my tyt is ok and I only have a deviation of 150hz in RXOffset
and the BER It is very low.

73.

narspt, 2018-01-16:
Thanks for your reply, my problem is a bit different then, maybe, but I will test to change this value anyway and see... In my case it works but I get "tremulous" voices (confirmed not happening listening at local repeater) and from time to time there are glitches... seems more an wrong timing thing (maybe?)... you don't get any of these symptoms, right? Thanks again. 73.

narspt, 2018-01-17:
Update: I did also try to change my ADF7021_DEV_DMR from 23 to 27 (didn't test with other values) and... perfect! The "tremulous" voices and glitches I was having all the time are gone, now my TYT MD380 finally listens to my hotspot perfectly, like when listening to the local repeater. Then it seems it's not just your TYT and your friends one... add mine on the list :)  However it's curious that (at least mine) is able to listen at "real" repeaters with no issues at all... strange why not at the hotspot with the default settings...
Many thanks, fortunately I found this issue and the mentioned fix worked with me also.

ea7gwc, 2018-01-17:
I'm glad it works.

My tyt also has no problems with the use of the local repeater and
either using my other DMO with motorola GM340 and plate
MMDVM.
This is adjusted with SDR and MMDVMCal.

The fact is that by my QTH have passed several TYT or RETEVIS and work
Fine, but mine does not. In TX my TYT is ok.

73

narspt, 2018-01-17:
Maybe that typical adjustment made on MMDVM using MMDVMCal+SDR provides greater deviation than the default 23 of MMDVM_HS? (that would explain why we can listen repeaters ok and not the hotspot)

But surely the TYT's are not all equal... while yours didn't get audio at all (just green led) with the default value, mine actually worked (but with "tremulous" voice and some glitches), then I guess your TYT probably needs even higher deviation value than my TYT, but mine needs some increase for sure... I could/may surely test with lower values between 23 and 27, but it seems really perfect so far with 27.
Pity that we can't get BER value on the TYT, it would help a lot.

73.

juribeparada, 2018-01-17:
Another possible explanation is some 7021 chips or modules are out of specs. The default DMR deviation value 23 is according with the ETSI standard. Apart from that, be sure you have corrected the frequency offset between hotspot and radio before modifying deviation. Remember this hotspot doesn't have AFC, and it is very common to confuse frequency offset problems with deviation problems. Some people have reported offsets of 400 Hz or even 800 Hz.

narspt, 2018-01-17:
Yes, I know that there is no AFC for DMR and I'm sure about my offset setting (-439), I did found it with TEST_TX and did set it to match my TYT carrier freq. (that is also not exactly accurate...), also it matched the value initially found (thoroughly) for hotspot rx that provides best BER (consistent 0.0 to 0.2). Anyway I did test before with other tx offsets and there was no difference, actually I guess the TYT haves some AFC, as it seems to work the same even with large offsets. Only really changing deviation on ADF7021_DEV_DMR cured my problem.

Btw, Andy: I did notice that a lot of values are different (incl. deviation) on defines for 12.2880MHz TCXO... that means they are also somehow dependent of TCXO freq? And if so, aren't they affected also by these TCXO offsets?

juribeparada, 2018-01-17:
OK, then probably we have differences with some 7021 chips. That is not good, but I hope to have in the near future a calibration software to try to fix particular cases.

neutralinsomniac, 2018-02-05:
Just for another piece of data, I also have one of these boards in the form of a JumboSpot, and when my JumboSpot is tuned to 431.075MHz, I have to tune my MD380 to 431.074MHz in order to reliably receive transmissions. I also have `#define ADF7021_DEV_DMR 27U` set in ADF7021.h, which I'm not sure was necessary or not.

juribeparada, 2018-02-05:
Well, that means a frequency offset of 1000 Hz !, it's nothing to do with deviation. Probably because a large frequency offset due to poor quality of the TCXO of your board, another reason to NOT buy MMDVM_HS clones. This board is a clone of DB9MAT & DF2ET MMDVM_HS_Hat board design. Please DO NOT buy or recommend to others MMDVM_HS clones, this kind of things damage open source initiatives in amateur radio, and at the end forces to stop sharing the knowledge, that means less and less innovation in amateur radio.

neutralinsomniac, 2018-02-05:
Yeah lesson learned. I was only recommended this board by another ham when I asked for advice on what to get. :( I will definitely get the "official" boards in the future

juribeparada, 2018-02-05:
Thank you for your understanding.

rogerclarkmelbourne, 2018-02-06:
I know someone else who recently bought one of these dodgy clones, and I think he may have access to a good frequency counter, so I'll email him and see if he can verify the TXO frequency

I suspect that none of the Chinese made clones have a 2.5ppm TXO, as there are none for sale on AliExpress (though I know China also has an internal "marketplace" website though which they may be able to source a 2.5ppm TXO.

But I think they probably ignored the links to Mouser that are in the BOM (in github) for these boards, and simply looked at the component values and not the tolerances.

ea7gwc, 2018-02-18:
Hello again Andy.
With the last update, it is possible to calculate to what value of DMRTxlevel
Would the value of 27 correspond ...?

I have tried the TXlevel and I start receiving in 55 now I have put it in 60
in the MMDVM.INI.

Up to now he modified the adf7021.h to 27.

Can you calculate 27 which would correspond in txlevel ...?

I have not tried the new MMDVMCal yet.

Thank you.

juribeparada, 2018-02-18:
Hello, TXLevel=59 is the same for 27, then you were very close. As MMDVM, TXLevel affect all modes, you have to use specific TXLevel for each mode if you want to modify only DMR.
Anyway, you should try MMDVMCal with a SDR, and then see if the deviation problem comes from your radio or your hotspot.

ea7gwc, 2018-02-18:
Gracias por tu rápida respuesta.
Entiendo que los valores a 50 de txlevel en los distintos modos, equivalen a los definidos
en el adf7021.h

Gracias de nuevo.

narspt, 2018-02-18:
I tried the deviation testing procedure on mine with MMDVMCal and an rtl_sdr, and on 50% I still see the carrier, only on 52.5% it goes down (not fully down but almost...), then on 55% carrier goes up again.
See screenshots: https://postimg.org/gallery/353usjmmu/

I could notice that it changes in steps, probably due to conversion to integer deviation value, but increasing it slowly and noticing when it makes difference I guess 23=50%, 24=52.5, 25=55%, 26=57% and 27=59.5% (59 should actually still be 26 if I'm correct?)

I was using 27 with great success until now, need to do some testing on next days with the 52.5% (actually any value between 52.5%-54.5%) and check if it works perfect with that value also.

juribeparada, 2018-02-18:
Well, there is not fine resolution for deviation values in ADF7021, we can't change that. You have to work with the closer value only, just that.

narspt, 2018-02-18:
Yes, I understand what you mean. But that 52.5-54.5% should supposedly be the proper one for my hotspot, right?

juribeparada, 2018-02-18:
According with your pictures, 50% should be the correct 2.75 kHz of deviation (center below 19 dB approx., before to reach Bessel null). You should change only DMR. Anyway, other modes are more tolerant for deviation issues.

narspt, 2018-02-18:
Sorry I'm not understanding, you say that 50% is the correct value for my hotspot then? (i.e. not out of specs and meaning my md380 haves the problem?) I really thought I should calibrate it to the point that the carrier goes fully down (and before going up again).

juribeparada, 2018-02-18:
With carrier null you have 2.88 kHz of deviation, going down until -19 dB you will get the expected 2.75 kHz for DMR. That produces the exact 648 Hz for +1 symbol deviation. According with your pictures, your hotspot is OK. I tested here with several md380/390 and not problems at the moment. Anyway, if you need more deviation for your radio, that is OK, and probably will be OK for other radios too.

narspt, 2018-02-18:
I see. Thanks again.

closed 2018-02-22 by juribeparada