juribeparada/MMDVM_HS #22: Random cuts RXing Dstar

EA1JM, 2018-01-08:
Hello, I have a lot of ramdom cuts rxing in Dstar mode, is there any recommendation to fix  it? Using latest MMDVM_HS firmware 1.1.0 and a zumspot librekit made by Blas EA7GIB, TNX in advance, Andy!

juribeparada, 2018-01-08:
The only thing you have to do is adjust the RX frequency offset. Could you copy a log of your MMDVMHost (with Debug=1 activated inside Modem)?

ea7gwc, 2018-01-09:
This has been happening for a long time.
Andy I do not know if you've seen this .........

https://github.com/g4klx/MMDVMHost/issues/354

73.

EA1JM, 2018-01-09:
Hello both,
Here is the log when the cut is produced: (exactly in M: 2018-01-09 22:57:38.847 Debug: DStarRX: Found end sync in Data) it seems it lost the sync, I'm using pistar image.

M: 2018-01-09 22:57:35.039 Debug: DStarRX: found data sync in Data
M: 2018-01-09 22:57:35.455 Debug: DStarRX: found data sync in Data
M: 2018-01-09 22:57:35.875 Debug: DStarRX: found data sync in Data
M: 2018-01-09 22:57:36.298 Debug: DStarRX: found data sync in Data
M: 2018-01-09 22:57:36.715 Debug: DStarRX: found data sync in Data
M: 2018-01-09 22:57:37.138 Debug: DStarRX: found data sync in Data
M: 2018-01-09 22:57:37.559 Debug: DStarRX: found data sync in Data
M: 2018-01-09 22:57:37.976 Debug: DStarRX: found data sync in Data
M: 2018-01-09 22:57:38.399 Debug: DStarRX: found data sync in Data
M: 2018-01-09 22:57:38.819 Debug: DStarRX: found data sync in Data
M: 2018-01-09 22:57:38.847 Debug: DStarRX: Found end sync in Data
M: 2018-01-09 22:57:38.852 D-Star, received RF end of transmission, 54.3 seconds, BER: 0.0%
M: 2018-01-09 22:57:39.237 Debug: DStarRX: found data sync in None, exact
M: 2018-01-09 22:57:39.604 D-Star, received RF late entry from EA1JM   /FRAN to CQCQCQ
M: 2018-01-09 22:57:41.160 Debug: DStarRX: data sync timed out, lost lock
M: 2018-01-09 22:57:41.165 D-Star, transmission lost, 0.6 seconds, BER: 13.7%
M: 2018-01-09 22:57:41.336 Debug: DStarRX: found data sync in None, exact
M: 2018-01-09 22:57:41.400 Debug: DStarRX: Found end sync in Data
M: 2018-01-09 22:57:50.882 Debug: DStarRX: found frame sync in None, fuzzy
M: 2018-01-09 22:58:00.892 Debug: DStarRX: found frame sync in None, fuzzy
M: 2018-01-09 22:58:01.980 Debug: Mode set to Idle

EA1JM, 2018-01-09:
Here is another example: the cut comes in M: 2018-01-09 23:16:30.737 Debug: DStarRX: found data sync in None, exact

Informes-Logs
Starting logging, please wait...
M: 2018-01-09 23:16:20.496 Debug: DStarRX: found frame sync in None, fuzzy
M: 2018-01-09 23:16:20.641 D-Star, received RF header from EA1JM /FRAN to CQCQCQ 
M: 2018-01-09 23:16:20.647 Debug: Mode set to D-Star
M: 2018-01-09 23:16:20.658 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:21.075 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:21.496 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:21.915 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:22.336 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:22.755 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:23.177 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:23.596 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:24.015 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:24.433 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:24.854 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:25.275 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:25.697 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:26.114 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:26.537 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:26.956 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:27.373 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:27.794 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:28.215 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:28.636 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:29.057 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:29.476 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:29.895 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:30.316 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:30.545 Debug: DStarRX: Found end sync in Data
M: 2018-01-09 23:16:30.551 D-Star, received RF end of transmission, 9.9 seconds, BER: 0.0%
M: 2018-01-09 23:16:30.737 Debug: DStarRX: found data sync in None, exact
M: 2018-01-09 23:16:31.102 D-Star, received RF late entry from EA1JM /FRAN to CQCQCQ 
M: 2018-01-09 23:16:31.157 Debug: DStarRX: found data sync in Data, early 1
M: 2018-01-09 23:16:31.574 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:31.998 Debug: DStarRX: found data sync in Data
M: 2018-01-09 23:16:32.202 Debug: DStarRX: Found end sync in Data
M: 2018-01-09 23:16:32.208 D-Star, received RF end of transmission, 1.1 seconds, BER: 0.0%
M: 2018-01-09 23:16:35.483 Debug: DStarRX: found frame sync in None, fuzzy
M: 2018-01-09 23:16:46.405 Debug: DStarRX: found frame sync in None, fuzzy

juribeparada, 2018-01-10:
Thank you Francisco to refer the MMDVMHost issue. Then, it seems to me this bug is common between MMDVM and MMDVM_HS. It is a false End of Transmission (EOT) detection (see the line: "M: 2018-01-09 23:16:30.545 Debug: DStarRX: Found end sync in Data"). I've tried to reproduce this bug here, but at the moment I haven't, it should happen not so often. I'm going to reduce the bit error number for the EOT sequence, please test this change with your D-Star radios. This could be useful also for MMDVM firmware.

EA1JM, 2018-01-10:
Ok!, many thanks Andy! I will recompile tomorrow the firmware and test, will post here the results! I also have a MMDVM 1.0.1 and the newest MMDVM for STMF446 nucleo with the same issues...hope your efforts will solve the issue...Best regards from Salamanca, Spain

juribeparada, 2018-01-10:
You welcome Fran, let's see what happen now.

EA1JM, 2018-01-10:
Hello Andy, all working flawlessly now!, I have compiled the firmware again and uploaded to the bluepill and no more cuts detected! Thanks so much!!!! Please make these changes to the MMDVM firmware as well!

juribeparada, 2018-01-10:
Excellent Fran!, I will do with the MMDVM firmware during night here.

EA1JM, 2018-01-10:
Ok many many thanks, it's so incredible that changing one value has fixed the issue! I close the ticket, many thanks for your support! Best 73 and keep up the good work!

closed 2018-01-10 by EA1JM

iw9grl, 2018-01-10:
Hi dear, same problems with nucleo/Due, which kind of fixes I have to do? Tnx Andy

juribeparada, 2018-01-10:
I will fix the MMDVM firmware soon, please wait.

iw9grl, 2018-01-10:
Tnx Andy

EA1JM, 2018-01-11:
Hello again, many thanks for the update of the MMDVM firmware! now working without cuts in D-Star with a Nucleo 446 based  board!