juribeparada/MMDVM_HS #23: Got a NAK from the modem

EA1JM, 2018-01-08:
Hello I'm trying to adjust the RX in DSTAR with the latest version of MMDVMCal with the Latest commit f89561e and I've got this response:

pi@raspberrypi:/opt/MMDVMCal $ sudo ./MMDVMCal /dev/ttyACM0
Version: 1 "MMDVM_HS-v1.1.0 20180106 ADF7021 FW by CA6JAU GitID #40cb087"
Got a NAK from the modem

Any clue?

The aim is to leave next to 0 the centre position while I receive in Dstar Mode moving the value with R or Rx and then put that level in the RXLevel of the MMDVM.ini , is it ok or RXlevel value does not have effect on the zumspots librekits?

Thanks in advance!

juribeparada, 2018-01-08:
Hi, MMDVMCal is for MMDVM, not for MMDVM_HS. There is not any analog interface with ADF7021 chips, then modem parameters like RXLevel, TXLevel, DCOffsets, etc don't make any sense with ZUMspot, MMDVM_HS or any hotspot based on ADF7021. You can't control those parameters. Maybe in the future I will enable support for MMDVMCal, but only for test purposes.

EA1JM, 2018-01-08:
Ok many thanks Andy for the response!

closed 2018-01-08 by EA1JM