juribeparada/MMDVM_HS #24: Help，Failed to init device.

hoky, 2018-01-12:
pi-star@h-pi-star(rw):scripts$ ls
build_fw.sh            install_fw_custom.sh  install_fw_hshat.sh     install_fw_rpi.sh  STM32F10X_Lib
install_buildtools.sh  install_fw_duplex.sh  install_fw_librekit.sh  install_fw_usb.sh  zumspot_rpi_fw.bin
pi-star@h-pi-star(rw):scripts$ ./install_fw_rpi.sh 
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   608    0   608    0     0    164      0 --:--:--  0:00:03 --:--:--   164
100 49752  100 49752    0     0   7453      0  0:00:06  0:00:06 --:--:-- 26690
Raspberry Pi 2 or Pi Zero W detected
stm32flash Arduino_STM32_0.9

http://github.com/rogerclarkmelbourne/arduino_stm32

Using Parser : Raw BINARY
Interface serial_posix: 57600 8E1
Failed to init device.

juribeparada, 2018-01-12:
What MMDVM_HS board do you have?

hoky, 2018-01-12:
@juribeparada 
It shows "JumboSpot" 

[image]
[image]

IS0FFD, 2018-01-12:
to upgrade firmware use 

`cd ~
curl -OL https://raw.github.com/juribeparada/MMDVM_HS/master/scripts/install_fw_hshat.sh`

make the script executable:
`chmod +x install_fw_hshat.sh`

stop your MMDVMHost process and run (you will need the root password):
`./install_fw_hshat.sh`

and wait to complete the upgrading process.

juribeparada, 2018-01-12:
@hoky, you have to understand I can’t support all the hardware that are based on this firmware, specially clones of well known boards. The board manufacturer should provide that support, specially if there are differences from my original design. If you can’t find that information, you could read BUILD.md for detailed information of the original MMDVM_HS project.

rogerclarkmelbourne, 2018-01-12:
@hoky 

The board could be DOA.

You could try posting to www.stm32duino.com as this repo uses my core Arduino code, and www.stm32duino.com is the community for that core (and some other Arduino cores)

With the generic STM32 boards, people on the forum often find they have not been tested at all by the manufacturer and there are soldering faults etc, which prevent the board from working.
And even if the code uploads, it may not run if the crystal is faulty

hoky, 2018-01-13:
Thank you @rogerclarkmelbourne  @juribeparada 
I have two hotspot, i will try one for test.

closed 2018-01-21 by juribeparada

rogerclarkmelbourne, 2018-01-26:
@juribeparada 

FYI. I've been trying to track down who makes these clones and what they consist of.

From my research so far, 

It seems to be this

https://www.aliexpress.com/item/MMDVM-hotspot-Support-P25-DMR-YSF-raspberry-pi-OLED-Antenna-Black-Case-READY-TO-QSO/32848396592.html?spm=2114.search0104.3.42.3ee70729OHfKbB&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10344_10130_10068_10342_10547_10343_10340_10548_10341_10084_10083_10618_10615_10307_10131_10132_10133_10313_10059_10534_100031_10103_441_10624_442_10623_10622_10621_10620_10142,searchweb201603_1,ppcSwitch_5&algo_expid=d6621be0-a6f3-47bd-88cf-1453bd5784ca-6&algo_pvid=d6621be0-a6f3-47bd-88cf-1453bd5784ca&transAbTest=ae803_4&priceBeautifyAB=1

They seem to be a modified version of this board

https://github.com/HS8JCV/MMDVM_HS_Hat

e.g. sold here

https://www.aliexpress.com/item/UHF-VHF-MMDVM-hotspot-Support-P25-DMR-YSF-for-raspberry-pi/32835555617.html?spm=2114.search0104.3.23.3ee70729OHfKbB&ws_ab_test=searchweb0_0,searchweb201602_4_10065_10344_10130_10068_10342_10547_10343_10340_10548_10341_10084_10083_10618_10615_10307_10131_10132_10133_10313_10059_10534_100031_10103_441_10624_442_10623_10622_10621_10620_10142,searchweb201603_1,ppcSwitch_5&algo_expid=d6621be0-a6f3-47bd-88cf-1453bd5784ca-3&algo_pvid=d6621be0-a6f3-47bd-88cf-1453bd5784ca&transAbTest=ae803_4&priceBeautifyAB=1

The HS8JCV is supposed to be a VHF not UHF version, but these are being sold as both VHF and UHF

But whether thats possible I don't know

(The original master of all of these is  https://github.com/mathisschmieder/MMDVM_HS_Hat)

Do you know if there are sources for the version with the OLED display , as I cant find a fork of this board which has a display

juribeparada, 2018-01-27:
Hi Roger,

Yes, somebody took the Mathis design of the MMDVM_HS (MMDVM_HS_Hat) and they did a copy and began to sell. Some of those board are exactly similar with the original design, and other ones have the name JumboSpot, but they are MMDVM_HS_Hat or maybe ZUMspot copies. The worse thing some of them didn’t put the full header connector, or there are problems with the component selection for the antenna matching network.
With a external 18 nH in L1 and L2 terminals of ADF7021, you will have VHF support (with UHF also). But generally you put your antenna matching/filter network for one band (UHF), and then VHF operation is not so good, but is ok for short distances.

I don’t know the sources for the version with OLED display.

Regards,
Andy.

rogerclarkmelbourne, 2018-01-27:
Hi Andy

Thanks.,

The VHF version is a fork

https://github.com/HS8JCV/MMDVM_HS_Hat

But I don't know if that fork is VHF only or would also work with UHF.

I think you are saying that its designed for VHF but works on UHF with shorter range.

juribeparada, 2018-01-31:
I think those board will do VHF and UHF, but probably UHF with limited output power and a reduced RX performance due to the different filter+matching net. The only way is using two different group of filters and selected by RF switches, but that increasing the final price of the board.

Regards,
Andy.