juribeparada/MMDVM_HS #26: PTT led keeps on

jcyfkimi, 2018-01-17:
I assembled my own librekit with blue pill and rf7021se module on prototype board with jump wires, however the PTT led will keeps on after 1st transmit(no matter what the content is, cw id or real DMR network qso) while testing with MMDVMHost, also I can hear nothing with my RTL-SDR even without signals. It seems that thr rf7021 module didn't get the command to transmit. I'm pretty sure that I modified the TCXO successfully and had no clue on what's wrong with the module. Any comments?

juribeparada, 2018-01-17:
This is a common hardware assembly issue and actually not a firmware problem. That happens when the stm32 is not receiving the proper clock signal from the 7021. There are many reasons of a missing clock signal:
- Check for proper firmware: MMDVM_HS (not ZUMspot!)
- No TxRxCLK signal connection
- No SPI bus connection between ADF7021 and STM32, mainly SLE, SDATA, SCLK (a 7021 not configured)
- CE with 0 V (7021 always in reset state)
- No proper power to 7021
- Bad 7021
- Bad soldering, short circuits, etc

jcyfkimi, 2018-01-18:
Yes I definitely know this should be a hardware issue, I double checked the items that you listed as below:
    Check for proper firmware: MMDVM_HS (not ZUMspot!)
          ====> I'm using the install_fw_librekit.sh script to load the firmware to blue pill, this should be all right.
    No TxRxCLK signal connection
    No SPI bus connection between ADF7021 and STM32, mainly SLE, SDATA, SCLK (a 7021 not configured)
          ====> I checking the wire connection using multimeter, all good.
    CE with 0 V (7021 always in reset state)
          ====> I measured the voltage of CE pin and get about 0.9v, maybe something wrong here?
    No proper power to 7021
          ====> I connected VCC to 3.3v, should I connected to 5V?
    Bad 7021
          ====> Unfortunately I only got 1 7021 module and unable to check if it's good or not.
    Bad soldering, short circuits, etc

juribeparada, 2018-01-19:
The 7021 uses 3.3 V, and CE pin should be high (it's low during a very short time for reseting)

jcyfkimi, 2018-01-20:
I will change a pin for CE to test if the PC14 pin has some issues, honestly I don't understand why it's get 0.9V

closed 2018-02-05 by juribeparada