juribeparada/MMDVM_HS #28: Problem after update

SA7BNT, 2018-02-24:
Hallo after i´m updating to commit cf557ad there is no more audio on my Hytera PD785.
RX seems to working fine on hs_hut but TX is not working for me. 
Last working commit for me is "commit 0256d3c"

73 
SA7BNT de Tim

juribeparada, 2018-02-24:
You need to have all TXLevel=50 in your MMDVM.ini in order to use default deviation values in latest firmware. Now you can adjust deviation for each mode from MMDVM.ini. Please see release notes.

SA7BNT, 2018-02-24:
Still same issue. No audio.
Thats my modem sektion

`[Modem]
Port=/dev/ttyAMA0
#Port=\\.\COM3
TXInvert=0
RXInvert=0
PTTInvert=0
TXDelay=50
RXOffset=-640 #-635
TXOffset=-550 #550 1400 Dummyload Antenne HS_HUT
DMRDelay=0
RXLevel=3
TXLevel=50
RXDCOffset=0
TXDCOffset=0
RFLevel=100
# CWIdTXLevel=50
# D-StarTXLevel=50
DMRTXLevel=83.5
# YSFTXLevel=50
# P25TXLevel=50
RSSIMappingFile=/home/pi/Applications/MMDVMHost/RSSI/mmdvm_hs_hut.dat
Trace=0
Debug=1`

juribeparada, 2018-02-24:
As I said, all TXLevels should be 50, you have DMRTXLevel=83.5, this should be set to:
DMRTXLevel=50
or
#DMRTXLevel=83.5

and TXLevel=50

SA7BNT, 2018-02-24:
Damm missed the "all" :( 
Now it´s working 👍  

Thanks

closed 2018-02-24 by SA7BNT