juribeparada/MMDVM_HS #36: Failed to read firmware version after 6 attempts

M0XZT, 2018-05-21:
Mmdvmhost fails to read firmware version after updating mmdvm_hs_dual_hat(DF2ET) to 1.3.5

phl0, 2018-05-21:
Do the LEDs blink at startup?

M0XZT, 2018-05-21:
No, just the power LED, rolled back to 1.3.3 and it's working ok

juribeparada, 2018-05-21:
Uhmmm, bad thing, probably we are running out of RAM for duplex mode

juribeparada, 2018-05-21:
I will fix reducing ring buffers again soon in the next release :(

juribeparada, 2018-05-21:
OK, it should be fix now in v1.3.6. I deleted v1.3.5

M0XZT, 2018-05-21:
Working fine now. Thanks Andy

closed 2018-05-21 by M0XZT

juribeparada, 2018-05-21:
Thank you Roger, but that is not so much, specially for the amount that I tried first time (the same amount that works OK for G4KLX MMDVM firmware). Now it is solved, I reduced a little more, but its still an improvement compared to v1.3.3. This ring buffer increment is only to reduce some modem NAK warnings (buffer overflow) that is produced for long network delays (too many DMR silence fills for example), then it is not so critical. Anyway, this affected only duplex devices, because firmware needed more resources.

rogerclarkmelbourne, 2018-05-21:
OK.

f4grl, 2018-05-23:
Bonjour à tous Mmdvmhost ne parvient pas à lire la version du micrologiciel après la mise à jour mmdvm_hs_dual de EA7GIB merci

juribeparada, 2018-05-23:
Hello, have you tried to compile latest source code with "make" method using a Raspberry Pi or a computer with Linux, with latest arm gcc tools?