juribeparada/MMDVM_HS #38: Compile problem

SA7BNT, 2018-05-22:
Hallo,
getting this fail when i´m trying to compile with STM32_USB_HOST

Thats my Config.h

#if !defined(CONFIG_H)
#define  CONFIG_H
#define MMDVM_HS_DUAL_HAT_REV10
#define ENABLE_ADF7021
#define DUPLEX
#define ADF7021_14_7456
#define STM32_USB_HOST
#define SEND_RSSI_DATA
#define SERIAL_REPEATER

And here is the last output where the fail comes

/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv7-m/libstdc++_nano.a(new_opv.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv7-m/libstdc++_nano.a(new_op.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv7-m/libstdc++_nano.a(new_handler.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
obj_f1/SerialSTM.o: In function `CSerialPort::beginInt(unsigned char, int)':
SerialSTM.cpp:(.text._ZN11CSerialPort8beginIntEhi+0x18): undefined reference to `usbserial'
SerialSTM.cpp:(.text._ZN11CSerialPort8beginIntEhi+0xc): undefined reference to `USBSerial::begin()'
obj_f1/SerialSTM.o: In function `CSerialPort::availableInt(unsigned char)':
SerialSTM.cpp:(.text._ZN11CSerialPort12availableIntEh+0x20): undefined reference to `usbserial'
SerialSTM.cpp:(.text._ZN11CSerialPort12availableIntEh+0x16): undefined reference to `USBSerial::available()'
obj_f1/SerialSTM.o: In function `CSerialPort::readInt(unsigned char)':
SerialSTM.cpp:(.text._ZN11CSerialPort7readIntEh+0x14): undefined reference to `USBSerial::read()'
SerialSTM.cpp:(.text._ZN11CSerialPort7readIntEh+0x20): undefined reference to `usbserial'
obj_f1/SerialSTM.o: In function `CSerialPort::writeInt(unsigned char, unsigned char const*, unsigned short, bool)':
SerialSTM.cpp:(.text._ZN11CSerialPort8writeIntEhPKhtb+0x18): undefined reference to `USBSerial::write(unsigned char const*, unsigned long)'
SerialSTM.cpp:(.text._ZN11CSerialPort8writeIntEhPKhtb+0x3c): undefined reference to `usbserial'
SerialSTM.cpp:(.text._ZN11CSerialPort8writeIntEhPKhtb+0x24): undefined reference to `USBSerial::flush()'
collect2: error: ld returned 1 exit status
Makefile:290: recipe for target 'bin/mmdvm_f1.elf' failed
make: *** [bin/mmdvm_f1.elf] Error 1

Something i´m doing wrong??
Trying to compile for USB use instead of GPIO
73 
SA7BNT

juribeparada, 2018-05-22:
Yes, you need to compile with bootloader support if you want to use USB: make bl

SA7BNT, 2018-05-22:
Works fine thanks for fast support.

closed 2018-05-22 by SA7BNT