juribeparada/MMDVM_HS #40: MMDVM HS Hat (GPIO) Lastest Firmware verison, Dashboard Can't read Firmware

hsojhcoom, 2018-05-31:
I updated to the lastest Dashbaord (3.4.15), and my MMDVM HS Hat is not fully booting. Fails to read the firmware after 6 attempts. I tried to flash the firmware again  and i am getting a Failed to init device. I have done nothing to the board (ie no damaged). I posted on the PI-Star user group with very little luck getting it solved. I noticed that 1.3.7 just released a couple of days ago. So maybe a bug, i wish i had some more time to dig into the problem but i figured i would report it first.

Thanks for all the hard work.

hsojhcoom, 2018-05-31:
Ended up being a bad RPI Zero W

closed 2018-05-31 by hsojhcoom

juribeparada, 2018-05-31:
OK, I see. I've just updated to PiStar 3.4.15 and with the latest firmware (1.3.7) for MMDVM_HS_Hat, and all working fine. Sorry about your RPI Zero W.