juribeparada/MMDVM_HS #52: Missing POCSAG Start Tone

robrobinette, 2018-07-18:
Another user and I, both using dual_hs_hat boards using 1.4.3 and Pi-Star 3.4.16 are seeing this behavior. It seems about 1 in 30 POCSAG transmissions are transmitted without the start tone. When this happens our SDR + PDW software will not decode the POCSAG page. The other guy even turned off all other modes and still had the problem. I monitor the hotspot's transmissions using a handi-talkie so I can hear the POCSAG tones. Another user has reported the same issue using a standard simplex ZUMspot with 1.4.3 and 3.4.16.

juribeparada, 2018-07-18:
I will need your MMDVMHost log (with log configured with FileLevel=1). This happens after you received too many POCSAG messages at once?

robrobinette, 2018-07-18:
[Pi-Star_MMDVM-2018-07-18.log](https://github.com/juribeparada/MMDVM_HS/files/2207315/Pi-Star_MMDVM-2018-07-18.log)

juribeparada, 2018-07-18:
Probably the modem has a buffer overflow due to too many POCSAG messages at once. I don't want to increase the buffer size because we are on the RAM limits for duplex boards in the firmware, actually I want to reduce some buffers. I will try to reproduce your issue and try to fix this in the MMDVMHost side, instead of simply increase buffers.

picchiosat, 2018-07-19:
Adding this info for your help

Multiple frame are in queue for DMR traffic delay.

Tx start correctly and all messages are decoded in one single tx time.
If during this trasmission another frame is added, tx stop and restart immediately without sync tone.
Last tx is lost. 

Roby IV3JDV

juribeparada, 2018-07-19:
Thank you for the info. Yes, yesterday I could reproduce the problem and I found the exact behavior that you describe. Also, if you send long menssages (several) at once, the problem happens.
A quick fix is to increase the IO buffer (Global.h) for example:

const uint16_t TX_RINGBUFFER_SIZE = 2048U;
const uint16_t RX_RINGBUFFER_SIZE = 2048U;

but I want to work a better solution, that I hope to be ready for the next release.

dk5ras, 2018-07-19:
I also can confirm it. So we're looing forward to the next release :) TNX!!

g4klx, 2018-07-20:
It's an odd one. I can't understand why this makes a difference. The output buffers are tested before the output signal is created (at least in the main MMDVM firmware) to ensure that the data buffers are not overrun. The start tone is only created when there is data to send, but the transmitter is off, which is detected by the presence of data to send in the interrupt routine. In fact this is the same logic as used by all the other modes to send the start synchronisation signal (i.e. tone). I'm watching this with interest.

Lawmanwilly, 2018-07-20:
I have seen this issue on my STM32 ZUM RADIO 0.9 Board. Is the fix relevant to this board as well ? 73s Lawrence DL1FLW

g4klx, 2018-07-20:
I hope so.

g4klx, 2018-07-20:
Andy, could you add your change to the main MMDVM firmware? I wonder if this has implications for any of the other modes?

juribeparada, 2018-07-20:
I will do Jonathan, but after a little more tests. This happens because in POCSAG mode, the modem switch to IDLE too quickly, and if this happen when IO buffers are full, the TX process exit and in not executed in the next (and last) iteration to empty the m_poBuffer. I think other modes are not affected by this because they have the hang time.

juribeparada, 2018-07-21:
I think this issue is fixed now. I also included this fix into MMDVM firmware. Please test and report.

picchiosat, 2018-07-21:
Amazing job... One shot...“One kill” 😂😂

g4klx, 2018-07-21:
> Hello Andy, i just ran an Update for the ZUM Radio Board it is a STM32 Board, It pulls from https://github.com/g4klx/MMDVM but the version was still from July 5th.
>
> Do i need pull from somewhere else ?

I don’t update the date unless it represents an update that everyone should get. The Git Id is the guide to the software revision.

Lawmanwilly, 2018-07-21:
Ok, new info for me..many thanks Ill do some testing now..law

closed 2018-07-24 by juribeparada