juribeparada/MMDVM_HS #55: Unreliable send/receive SMS on dual_hat

hb9eue, 2018-07-29:
Hi

I'm not sure how to further debug the problem. I have a dvmega based hotspot. SMS and GPS Positions from a Hytera PD685 work flawlessly. But I wanted to be able to use both time slots, as I plan using the hotspot at event like JOTI.

So I purchased the HS_DUAL_HAT and this also works flawlessly for audio, TalkerAlias and GPS Positions included into the Audio stream. But it is very unreliably (almost not at all) for SMS and GPS Positions sent to 5057 or other services like the RPT or WX Service. I have not been trying handheld to handheld SMS yet.

When an SMS is sent from APRS to my Hytera, I see in the MMDVMHost Log, that it is received and 3 transmit attempts made. The Hytera also reacts to them and is starting to send, but I suppose something goes wrong, as the message is not being received. So I suppose the issue is mainly in the direction MMDVMHost to Handheld and not Handheld to MMDVMHost.

I've been playing around with TXinvert, RXinvert etc. no change. The TX/RX Offset seem to be quite right as usually I am somewhere between 0% and 0.5% BER. Also changing the jitter buffer has no effect at all.

Any help or hints how to solve the issue are very appreciated. Could it be the preambles are too short or similar so the handheld is not able to sync up to the transmission and misses the start of the message?

Benoit / HB9EUE

juribeparada, 2018-08-15:
Hi,

I don't have any Hytera UHF radio to test this, but loss of short data packets is a common problem in duplex MMDVM_HS. At the moment I don't have a clue why this happens, it seems to me hardware has a part of this also (maybe RF filter transient response?).

TXinvert, RXinvert have no effect in MMDVM_HS boards.

I believe that the problem is in the oposite direction (RX): Handheld to MMDVM_HS board.

Be sure you have adjusted well TX/RX Offsets, select only DMR mode (disable mode scanning) and just in case, compile the firmware with AGC off (try the option #define AD7021_GAIN_HIGH in Config.h)

hb9eue, 2018-08-18:
Hi Andy

Thank you, I had a go at compiling the firmware myself. I'm no developer and I am stuck at this linker error. Any suggestion how to circumvent?

arm-none-eabi-g++ obj_f1/DMRDMOTX.o obj_f1/IOArduino.o obj_f1/POCSAGTX.o obj_f1/MMDVM_HS.o obj_f1/SerialPort.o obj_f1/DStarRX.o obj_f1/IO.o obj_f1/YSFTX.o obj_f1/DMRTX.o obj_f1/DStarTX.o obj_f1/DMRIdleRX.o obj_f1/P25TX.o obj_f1/BitRB.o obj_f1/NXDNRX.o obj_f1/Utils.o obj_f1/IOSTM.o obj_f1/SerialRB.o obj_f1/YSFRX.o obj_f1/DMRSlotType.o obj_f1/DMRDMORX.o obj_f1/SerialArduino.o obj_f1/DMRSlotRX.o obj_f1/DMRRX.o obj_f1/CalDMR.o obj_f1/ADF7021.o obj_f1/NXDNTX.o obj_f1/CalRSSI.o obj_f1/SerialSTM.o obj_f1/P25RX.o obj_f1/CWIdTX.o obj_f1/stm32f10x_rcc.o obj_f1/stm32f10x_usart.o obj_f1/misc.o obj_f1/stm32f10x_iwdg.o obj_f1/stm32f10x_gpio.o obj_f1/stm32f10x_exti.o obj_f1/system_stm32f10x.o obj_f1/startup_stm32f10x.o  -Os --specs=nano.specs --specs=nosys.specs -T normal.ld -mcpu=cortex-m3 -march=armv7-m -mthumb -Wall -Wextra  -o bin/mmdvm_f1.elf

/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv7-m/libstdc++_nano.a(new_opv.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail

[...]
collect2: error: ld returned 1 exit status
Makefile:290: recipe for target 'bin/mmdvm_f1.elf' failed
make: *** [bin/mmdvm_f1.elf] Error 1

phl0, 2018-08-18:
There should be a more descriptive error message in the output. Maybe you cut it in your paste?

phl0, 2018-08-18:
Which command did you execute to compile? It should be "make bl" if you want the one with bootloader.

hb9eue, 2018-08-18:
I did just 'make'.

hb9eue, 2018-08-18:
Ok, make bl compiles without error. But now I don't understand how I flash the firmware for the dual_hat.

MMDVM_HS# make mmdvm_hs_hat
/usr/bin/stm32flash -v -w bin/mmdvm_f1.bin -g 0x0 -R -i 20,-21,21:-20,21 /dev/ttyAMA0
stm32flash 0.4

http://stm32flash.googlecode.com/

Intel HEX ERROR: System Error
bin/mmdvm_f1.bin: No such file or directory

Makefile:418: recipe for target 'zumspot-pi' failed

Well I didn't build for a zumspot...

MMDVM_HS# make dfu
././STM32F10X_Lib/utils/rpi32/dfu-util -D bin/mmdvm_f1bl.bin -d 1eaf:0003 -a 2 -R -R
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
Makefile:429: recipe for target 'dfu' failed
make: *** [dfu] Error 74

Well that is the correct bin file, but do I need dfu?

If I look at the script to install the binary file, it just does:
$STM32FLASH -v -w mmdvm_hs_dual_hat_fw.bin -g 0x0 -R -i 20,-21,21:-20,21 /dev/ttyAMA0

So can I just have a go and try that? Is there any chance of briking my dual_hat?

-Benoit-

hb9eue, 2018-08-18:
Ok, I believe I found my mistake. I had to enable STM32_USART1_HOST instead of the USB one. Testing if AD7021_GAIN_HIGH makes SMS work.

hb9eue, 2018-08-18:
I: 2018-08-18 13:38:40.780 MMDVM protocol version: 1, description: MMDVM_HS_Dual_Hat-v1.4.6 20180818 high gain 14.7456MHz dual ADF7021 FW by CA6JAU GitID #23f1326

No, no difference...
Requesting RPT from 262994 works. But Requeinst WX GPS from 262993 still does not.

For me it looks like it is the packet back from the NET to RF which does not make it through. I see three attempts at re-sending:

: 2018-08-18 13:40:13.999 Downlink Activate received from HB9EUE
M: 2018-08-18 13:40:14.198 DMR Slot 2, received RF data header from HB9EUE to 262993, 2 blocks
M: 2018-08-18 13:40:14.319 DMR Slot 2, ended RF data transmission
M: 2018-08-18 13:40:14.634 DMR Slot 2, received network Data Preamble CSBK (1 to follow) from 262993 to HB9EUE
M: 2018-08-18 13:40:14.695 DMR Slot 2, received network data header from 262993 to HB9EUE, 0 blocks
M: 2018-08-18 13:40:14.697 DMR Slot 2, ended network data transmission
M: 2018-08-18 13:40:17.457 DMR Slot 2, received network Data Preamble CSBK (23 to follow) from 262993 to HB9EUE
M: 2018-08-18 13:40:17.519 DMR Slot 2, received network data header from 262993 to HB9EUE, 22 blocks
M: 2018-08-18 13:40:18.863 DMR Slot 2, ended network data transmission
M: 2018-08-18 13:40:20.701 DMR Slot 2, received network Data Preamble CSBK (23 to follow) from 262993 to HB9EUE
M: 2018-08-18 13:40:20.758 DMR Slot 2, received network data header from 262993 to HB9EUE, 22 blocks
M: 2018-08-18 13:40:22.104 DMR Slot 2, ended network data transmission
M: 2018-08-18 13:40:23.934 DMR Slot 2, received network Data Preamble CSBK (23 to follow) from 262993 to HB9EUE
M: 2018-08-18 13:40:23.995 DMR Slot 2, received network data header from 262993 to HB9EUE, 22 blocks
M: 2018-08-18 13:40:25.353 DMR Slot 2, ended network data transmission
M: 2018-08-18 13:40:25.779 DMR Slot 2, received RF data header from HB9EUE to 262993, 1 blocks
M: 2018-08-18 13:40:25.838 DMR Slot 2, ended RF data transmission
M: 2018-08-18 13:40:26.159 DMR Slot 2, received network Data Preamble CSBK (5 to follow) from 262993 to HB9EUE
M: 2018-08-18 13:40:26.222 DMR Slot 2, received network data header from 262993 to HB9EUE, 4 blocks
M: 2018-08-18 13:40:26.497 DMR Slot 2, ended network data transmission
M: 2018-08-18 13:40:26.978 DMR Slot 2, received RF data header from HB9EUE to 262993, 1 blocks
M: 2018-08-18 13:40:27.039 DMR Slot 2, ended RF data transmission
M: 2018-08-18 13:40:27.366 DMR Slot 2, received network Data Preamble CSBK (5 to follow) from 262993 to HB9EUE
M: 2018-08-18 13:40:27.425 DMR Slot 2, received network data header from 262993 to HB9EUE, 4 blocks
M: 2018-08-18 13:40:27.697 DMR Slot 2, ended network data transmission

juribeparada, 2018-08-26:
hb9eue, are you using DMRGateway?, if so, can you connect MMDVMHost directly to BM?

hb9eue, 2018-08-27:
Hi Andy

No, MMDVMHost is directly connecting to BM. But I'm further investigating in the issue. Maybe something changed on the 262993 side or on the BM Network, as last time I compared to my dvmega board, this also had the same problem.

-Benoit-

rogerclarkmelbourne, 2018-08-27:
SMS works Ok on my dual hat

hb9eue, 2018-08-28:
Hmm, I still am not lucky... (Just upgraded to 1.4.7)
Checked again on: https://wiki.brandmeister.network/index.php/Hytera/Radios
Well I can't find any obvious config mistake on my radio.
Changed the Data Bearer Service to compressed IP, makes no difference.

Testing mostly with 262993 service: http://wiki.bm262.de/doku.php?id=servicenummer

I got a Hytera PD685.
Sending via my DVMega Hotspot on a PI-2 works about 9 of 10 times.
Sending via my HS_Dual_Hat on a Pi-Zero-w works maybe 1 of 10 times.

RPT command to 262994 works more reliably on both, but on the Dual_Hat it's still about every 3rd attempt which is getting a reply.

I also set 'CONFIG DELAY 10' to get delayed reply SMS from 262993, no change.

Could it be a internet timing problem? My PI-2/DVMega is connected via ethernet interface and therefore hardly has any jitter.
Over the Pi-Zero-W (Wlan) the 'internet' side jitter is higher and I have observed packetloss on receiving transmissions FROM the net (well when taking it with me and connecting to public wlan on vacation and such...)

-Benoit-

juribeparada, 2018-09-02:
I don't believe this issue is an internet timing problem. Can you enable debug messages in firmware compilation (#define ENABLE_DEBUG) and enable debug of the modem in Pi-Star (or MMDVM.ini), and then copy the log here?

hb9eue, 2018-09-03:
Here you go...

The whole startup of MMDVMhost and right after it is up and connected to the swiss BM master, I request 'WX GPS' to 262993, shortly afterwards I see there is a reply coming in from 262993, but that does not make it to my Hytera. Any clues?

I: 2018-09-03 18:56:54.302 MMDVM protocol version: 1, description: MMDVM_HS_Dual_Hat-v1.4.7 20180821 14.7456MHz dual ADF7021 FW by CA6JAU GitID #ceb6a4d
I: 2018-09-03 18:56:54.323 Display Parameters
I: 2018-09-03 18:56:54.323     Type: Nextion
I: 2018-09-03 18:56:54.323     Port: modem
I: 2018-09-03 18:56:54.323     Brightness: 50
I: 2018-09-03 18:56:54.323     Clock Display: yes
I: 2018-09-03 18:56:54.323     Display UTC: no
I: 2018-09-03 18:56:54.323     Idle Brightness: 1
I: 2018-09-03 18:56:54.323     Screen Layout: G4KLX (Default)
I: 2018-09-03 18:56:54.323 Interfaces Info
I: 2018-09-03 18:56:54.326     IPv4: lo:127.0.0.1
I: 2018-09-03 18:56:54.328     IPv4: wlan0:192.168.57.253
I: 2018-09-03 18:56:54.330     IPv6: lo:::1
I: 2018-09-03 18:56:54.332     IPv6: wlan0:2001:4060:dead:babe:a68d:fee4:3d94:30f8
I: 2018-09-03 18:56:54.334     IPv6: wlan0:fe80::19a9:af84:967a:64da%wlan0
I: 2018-09-03 18:56:54.337     Default interface is : wlan0
I: 2018-09-03 18:56:54.339     IP to show: wlan0:192.168.57.253
I: 2018-09-03 18:56:54.340 DMR Network Parameters
I: 2018-09-03 18:56:54.342     Address: 128.65.196.21
I: 2018-09-03 18:56:54.344     Port: 62031
I: 2018-09-03 18:56:54.346     Local: random
I: 2018-09-03 18:56:54.346     Jitter: 500ms
I: 2018-09-03 18:56:54.346     Slot 1: enabled
I: 2018-09-03 18:56:54.347     Slot 2: enabled
I: 2018-09-03 18:56:54.348     Mode Hang: 20s
I: 2018-09-03 18:56:54.349 Info Parameters
I: 2018-09-03 18:56:54.351     Callsign: HB9EUE
I: 2018-09-03 18:56:54.352     RX Frequency: 430725000Hz
I: 2018-09-03 18:56:54.354     TX Frequency: 438325000Hz
I: 2018-09-03 18:56:54.355     Power: 1W
I: 2018-09-03 18:56:54.356     Latitude: 47.442749deg N
I: 2018-09-03 18:56:54.357     Longitude: 8.329050deg E
I: 2018-09-03 18:56:54.358     Height: 0m
I: 2018-09-03 18:56:54.359     Location: "Neuenhof,JN47DK"
I: 2018-09-03 18:56:54.359     Description: "Switzerland"
I: 2018-09-03 18:56:54.360     URL: "http://www.qrz.com/db/HB9EUE"
M: 2018-09-03 18:56:54.361 DMR, Opening DMR Network
I: 2018-09-03 18:56:54.362 RSSI
I: 2018-09-03 18:56:54.362     Mapping File: /usr/local/etc/RSSI.dat
I: 2018-09-03 18:56:54.365 Loaded 14 RSSI data mapping points from /usr/local/etc/RSSI.dat
I: 2018-09-03 18:56:54.367 DMR Id Lookups
I: 2018-09-03 18:56:54.369     File: /usr/local/etc/DMRIds.dat
I: 2018-09-03 18:56:54.370     Reload: 24 hours
I: 2018-09-03 18:56:55.469 Loaded 106766 Ids to the DMR callsign lookup table
I: 2018-09-03 18:56:55.471 DMR RF Parameters
I: 2018-09-03 18:56:55.472     Id: 228502102
I: 2018-09-03 18:56:55.472     Color Code: 1
I: 2018-09-03 18:56:55.473     Self Only: no
I: 2018-09-03 18:56:55.474     Embedded LC Only: no
I: 2018-09-03 18:56:55.477     Dump Talker Alias Data: yes
I: 2018-09-03 18:56:55.479     Prefixes: 0
I: 2018-09-03 18:56:55.481     Call Hang: 3s
I: 2018-09-03 18:56:55.483     TX Hang: 4s
I: 2018-09-03 18:56:55.485     Mode Hang: 20s
M: 2018-09-03 18:56:55.487 MMDVMHost-20180802_Pi-Star is running
M: 2018-09-03 18:56:55.489 Debug: MMDVM_HS FW configuration:
M: 2018-09-03 18:56:55.492 Debug: TX freq (Hz): 438325031
I: 2018-09-03 18:56:55.492 Started the DMR Id lookup reload thread
M: 2018-09-03 18:56:55.498 Debug: RX freq (Hz): 430725700
M: 2018-09-03 18:56:55.498 Debug: Power set: 63
M: 2018-09-03 18:56:55.504 Debug: D-Star dev (Hz): 1209
M: 2018-09-03 18:56:55.505 Debug: DMR +1 sym dev (Hz): 646
M: 2018-09-03 18:56:55.511 Debug: YSF +1 sym dev (Hz): 450
M: 2018-09-03 18:56:55.512 Debug: P25 +1 sym dev (Hz): 618
M: 2018-09-03 18:56:55.518 Debug: NXDN +1 sym dev (Hz): 365
M: 2018-09-03 18:56:55.519 Debug: POCSAG dev (Hz): 4500
[... 28 lines cut ...]
M: 2018-09-03 18:57:52.811 DMR Slot 2, received RF data header from HB9EUE to 5057, 0 blocks
M: 2018-09-03 18:57:52.812 DMR Slot 2, ended RF data transmission
M: 2018-09-03 18:58:12.841 Debug: Mode set to Idle
M: 2018-09-03 18:58:17.489 DMR Slot 2, received network Data Preamble CSBK (34 to follow) from 262993 to HB9EUE
M: 2018-09-03 18:58:17.506 Debug: Mode set to DMR
M: 2018-09-03 18:58:17.583 DMR Slot 2, received network data header from 262993 to HB9EUE, 33 blocks
M: 2018-09-03 18:58:19.622 DMR Slot 2, ended network data transmission
M: 2018-09-03 18:58:20.575 Debug: SYNC corr MS Data found slot/pos/start/end: 2 454 299 562
M: 2018-09-03 18:58:21.449 DMR Slot 2, received network Data Preamble CSBK (34 to follow) from 262993 to HB9EUE
M: 2018-09-03 18:58:21.509 DMR Slot 2, received network data header from 262993 to HB9EUE, 33 blocks
M: 2018-09-03 18:58:23.524 DMR Slot 2, ended network data transmission
M: 2018-09-03 18:58:23.935 Debug: SYNC corr MS Data found slot/pos/start/end: 2 458 303 566
M: 2018-09-03 18:58:23.947 Debug: DMRSlot2RX: data header found pos 458
M: 2018-09-03 18:58:23.957 DMR Slot 2, received RF data header from HB9EUE to 262993, 1 blocks
M: 2018-09-03 18:58:23.995 Debug: SYNC corr MS Data found slot/pos/start/end: 2 458 303 566
M: 2018-09-03 18:58:24.007 Debug: DMRSlot2RX: data payload found pos 458
M: 2018-09-03 18:58:24.016 DMR Slot 2, ended RF data transmission
M: 2018-09-03 18:58:24.333 DMR Slot 2, received network Data Preamble CSBK (10 to follow) from 262993 to HB9EUE
M: 2018-09-03 18:58:24.389 DMR Slot 2, received network data header from 262993 to HB9EUE, 9 blocks
M: 2018-09-03 18:58:24.966 DMR Slot 2, ended network data transmission
M: 2018-09-03 18:58:25.615 Debug: SYNC corr MS Data found slot/pos/start/end: 2 462 307 570
M: 2018-09-03 18:58:25.627 Debug: DMRSlot2RX: data header found pos 462
M: 2018-09-03 18:58:25.675 Debug: SYNC corr MS Data found slot/pos/start/end: 2 462 307 570
M: 2018-09-03 18:58:25.687 Debug: DMRSlot2RX: data payload found pos 462
M: 2018-09-03 18:58:45.004 Debug: Mode set to Idle

juribeparada, 2018-09-07:
Something that is different of this radio compared with others is in this log I can't see any "Data Preamble CSBK" from the RF side. It looks like your Hytera radio does not send "Data Preamble CSBK" before a data transmission. Because I am not a Hytera user, I can't say if this is normal for Hytera radios or not, but other radios transmit a lot of data preambles before sending data, which increases the data RX reliability. I will test with my other radios and compare.

pjao, 2018-09-20:
Any news about this? I've the same problem with uv-md380 working in uhf.

rogerclarkmelbourne, 2018-09-20:
Did you need to change the DMRTxLevel setting to make your MD-380 work with the duplex hotspot?

I need to change this value to 55 (instead of 50) otherwise the MD-380 will not connect at all.

I've also received reports that people can not make the MD-380 work with the duplex hotspot, at all, even if they change the DMRTxLevel

From what Andy has said in response to other issues. There can be problems with the MD-380 even on Simplex hotspots.

Did you try changing your board to work in simplex mode and see if the problem persists?

pjao, 2018-09-20:
I was talking about the vhf/uhf version of the md380, the uv-380.

Let me tell what i've done with some success.

First compile the firmware with AGC off (option #define AD7021_GAIN_HIGH in Config.h)
Then changed the DMRTXLevel not the TXLevel to 55.

After this I'm able to send/receive SMS on slot2 most of the times. Nothing changed on slot1, unable to use it for SMS.

Anything else to test?

rogerclarkmelbourne, 2018-09-20:
I only have the MD-380 , but I think perhaps it has the same problem

I will need to change my DMRTxLevel to approx 55 and retest, because at the moment, I have it set to 50 for use with GD-77

I did not need to change any compile options to make the MD-380 work for digital audio.

I think it received SMS OK, but I don't think I tested sending, I also only used TS2, so I'd need to test TS1

BTW. 
I have a problem sometimes where PiStar shows MD-380 transmitting on TS1 but its set to TS2 (for audio)

juribeparada, 2018-09-21:
I don't have any news about this. In general, duplex boards have problems decoding short data transmissions, then it's important that the radio transmits a long preamble in order to activate the DMR mode. Increasing the deviation level of TX helps with the DMR wake-up activation, in some specific radio models.

rogerclarkmelbourne, 2018-09-21:
Andy.

Do you know what value of "preamble" is OK ?
I've checked my the MD-380/390 CPS and the default preamble seems to be set to 300mS
On the GD-77 the default seems to be 360mS (I've checked the official latest CPS and also my CPS. Both have the same value)

So perhaps setting the preamble on the MD-380 to the same as the GD-77 would be worth trying

juribeparada, 2018-09-21:
I think more than 300 msec is OK

pjao, 2018-09-21:
> I don't have any news about this. In general, duplex boards have problems decoding short data transmissions, then it's important that the radio transmits a long preamble in order to activate the DMR [...]

I think the problem could be the preamble of the board, not the radio (if radio as a value greater then 300). How it is possible to increase

hb9eue, 2018-09-21:
I also think the problem is the board => radio way, not the other way round. I seem to be perfectly able to SEND sms from the radio, but mostly the radio is not able to receive the answer. So increasing the preamble length in the firmware would be worth a try...

juribeparada, 2018-09-21:
Sorry but in that case firmware has nothing to do with preamble length, firmware is transparent for data from MMDVMHost/network. I can't remember if MMDVMHost add more preamble or just copy the preamble from DMR network, I need to take a look to MMDVMHost code when I have some time for that.

hb9eue, 2018-09-23:
That too could be a hint to the problem. I noticed that DMR to APRS SMS work more or less and also the Services on 262994 work more or less, but the real problem is with servies on 262993, so maybe this service answers with a different preamble. Or what I also suspect. I see that long replies are being sent in chunks. Sometimes bigger, sometimes smaller chunks (... more to follow) message in the debug log. So I suspect this also could be part of the issue.

rogerclarkmelbourne, 2018-09-23:
FYI.
WX service does not work for me, and I was told by VK5QS it does not work for him either.
But VK5QS uses a board from “repeater builder”, which AFIK uses the analog version of MMDVM (modem) , but also AFIK uses MMDVMHost 

I presumed the problem was the WX service, but perhaps it’s also partially a problem in MMDVMHost

hb9eue, 2018-09-24:
With the information that the preamble length could be involved in the problem. I checked if I could configure any parameters on the Hytera PD685 via CPS.

Yes, the preamble lenght was set to 960ms. I shortened this to 480ms.

And the help to that topic also mentioned, that the battery saver function affects the preamble length. So to see if this changes anything, I set the delay until battery saving kicks in from 5 seconds to 30 seconds, long enough to get a reply from most services.

Somehow with those settings I get the feeling that SMS reception is much more reliable. And yes, after more tests I can confirm now, the problem mainly occurs while receiving from service 262993. All other SMS, like sending a qrz.com lookup request to WHO-15 on 5057 work fine now.

I suppose in battery saving mode, the radio is going into deep sleep and only listens for a preamble all 500ms or so, so the preamble has to have a certain length to be able to wake up the radio.

juribeparada, 2018-10-07:
Ok, finally this issue seems to be related with network or radio specific configuration. I will close this issue, but feel free to re-open if you find this could be produced by MMDVM_HS firmware.

closed 2018-10-07 by juribeparada

K2ELI, 2022-12-06:
Greetings and hope it's ok to re-open this. I searched and only found this thread on SMS. If I should open a new thread, just let me know.
I'm able to send and receive SMS over the BM network, both on my duplex hotspots and the repeater. But I can't get my Connect Systems CS800Ds or AnyTone 878s to ACK the receipt of these message when using the MB Text Capture feature. Once and a while the ACK will get to BM and the repeat SMSs will stop. This tells me it's some type of timing issue. I tried to increase the preamble times on both radios and that doesn't work. Curious if anyone has been able to get BM Text Capture to work with a CS800D or AnyTone 878.