juribeparada/MMDVM_HS #56: compile fail

picchiosat, 2018-07-29:
today I receive this message when try to compile the firmware for BI7JTA HS_DUAL_HAT

obj_f1/SerialSTM.o: In function `CSerialPort::beginInt(unsigned char, int)':
SerialSTM.cpp:(.text._ZN11CSerialPort8beginIntEhi+0x18): undefined reference to `usbserial'
SerialSTM.cpp:(.text._ZN11CSerialPort8beginIntEhi+0xc): undefined reference to `USBSerial::begin()'
obj_f1/SerialSTM.o: In function `CSerialPort::availableInt(unsigned char)':
SerialSTM.cpp:(.text._ZN11CSerialPort12availableIntEh+0x20): undefined reference to `usbserial'
SerialSTM.cpp:(.text._ZN11CSerialPort12availableIntEh+0x16): undefined reference to `USBSerial::available()'
obj_f1/SerialSTM.o: In function `CSerialPort::readInt(unsigned char)':
SerialSTM.cpp:(.text._ZN11CSerialPort7readIntEh+0x14): undefined reference to `USBSerial::read()'
SerialSTM.cpp:(.text._ZN11CSerialPort7readIntEh+0x20): undefined reference to `usbserial'
obj_f1/SerialSTM.o: In function `CSerialPort::writeInt(unsigned char, unsigned char const*, unsigned short, bool)':
SerialSTM.cpp:(.text._ZN11CSerialPort8writeIntEhPKhtb+0x16): undefined reference to `USBSerial::write(unsigned char const*, unsigned long)'
SerialSTM.cpp:(.text._ZN11CSerialPort8writeIntEhPKhtb+0x3c): undefined reference to `usbserial'
SerialSTM.cpp:(.text._ZN11CSerialPort8writeIntEhPKhtb+0x22): undefined reference to `USBSerial::flush()'
collect2: error: ld returned 1 exit status
Makefile:290: recipe for target 'bin/mmdvm_f1.elf' failed
make: *** [bin/mmdvm_f1.elf] Error 1

Roby, IV3JDV

juribeparada, 2018-07-30:
Your are not compiling right:
1) If you enable USB (see your Config.h), you must use: make bl
2) If your board use GPIO port, you must change at Config.h and then compile with: make

picchiosat, 2018-07-30:
Tnx for answer,
until other week everything worked well.
I use the GPIO and the Config.h is always the same and I have not changed anything
Today I will still do some tests

Roby

picchiosat, 2018-07-30:
SOLVED!! Tnx and sorry for the useless question

Roby

closed 2018-08-04 by juribeparada