juribeparada/MMDVM_HS #57: How would I support a PA module which needs a Tx enable input

rogerclarkmelbourne, 2018-07-30:
Hi

Could you give me a rough idea of how to implement turning on a GPIO pin when the tx module is actually transmitting, as I'd like to implement this to control a 3W PA module, which I bought from eBay.

The PA has some notes which suggest it should not be powered when there is no signal input, so I'll need to use a FET to switch its power on and off, under GPIO control.

Also, for use with a single Rf module hotspot, it would require a switching voltage for the receiver bypass switching system consisting of a quarter wave of coax and some pin dioded

I had a look though the code but couldn't easily see any variables which were set when the RF module was set to transmit.

juribeparada, 2018-07-30:
It should work with simplex and duplex boards.

rogerclarkmelbourne, 2018-07-30:
Thanks

I'll close this issue, and investigate how to use that pin.

closed 2018-07-30 by rogerclarkmelbourne