juribeparada/MMDVM_HS #59: Led ws2812b

ea7gwc, 2018-08-12:
I ask from my ignorance ....... it would be possible
the use of a led ws2812b for mode indication. this would use a
single line and no 6 led and 6 resistor for mode indication.

That is, a single LED with color change.

73

rogerclarkmelbourne, 2018-08-12:
The problem with those ws2812b  LED's is that they are actually quite difficult to control.
They require very precise pulse widths of serial data to be sent to them, which either requires the interrupts to be disabled or to use the SPI bus.

Turning off the interrupts probably isn't an option, even when just changing the LED colour as I suspect the transceiver chips interrupts the STM32 processor when they have data.

On the single transceiver hotspot boards, the 2nd SPI channel could be used to send the data, but I don't know if thats possible on the duplex hotspot, because they may be using the 2nd SPI channel for the second transceiver  

Also. I'm not sure if you mean as string of 6 WS2812b LEDs or whether you mean one LED and you change its colour ?
In terms of User Interface design, replacing 6 separate LED's with one multi coloured LED would make it very difficult to know what the board was doing, as you'd need to understand what the different colours mean.

If you mean 6 x WS2812B, then again, I'm not sure of the benefit. Yes it saves some GPIO, but the STM32 is not short of GPIO's

phl0, 2018-08-12:
I have some WS2812bs on the bench ... need to expermiment with those...

closed 2018-08-13 by ea7gwc