juribeparada/MMDVM_HS #60: MMDVM_HS_HAT_DUPLEX For Raspberry pi

A61BN, 2018-08-14:
i have just got my hotpot model MMDVM_HS_HAT_DUPLEX For Raspberry pi and i have problem to make it tx and rx no UHF and the offset is wrong it says 800hz only no - no + i have tried bought but no luck and i new to the DMR world so can any one help me and point me in to the right direction 

thank you all 

mohammed A61bn

rogerclarkmelbourne, 2018-08-15:
Try running it in Simplex mode.

My MD-380 won't connect to that board, but my GD-77 will.
On my GD-77 it only works for 60 secs in Duplex mode and then it stops working, and I have to release the PTT and try again.

But both my MD-380 and GD-77 work fine if I put the board into simplex mode.

In PiStar, use the menu option above the one you selected for the Duplex modem, 
call

You also need to change the Controller Mode to Simplex in the Control Software section at the top of the page.

BTW.
If you have a SDR receiver, you can use use MMDVMCal to transmit just a carrier and compare that with the carrier frequency on your radio (i.e transmit on FM with no audio) and compare the frequency.

FYI.
I think these boards are not very reliable. I have spoken to several people who have them but have problems using them.

awasser1, 2018-08-18:
That's very odd Andy.  we have reported this to Mr. Winters in China.  that's strange he hasnt mentioned it.  also timing like calls on slot 1 will come out on slot 2 also.  that could be different issue.  Now that you do know I am sure it can be fixed.  You do awesome work and thank you very much for everything you do.

Alec
N1AJW

closed 2018-09-02 by juribeparada