juribeparada/MMDVM_HS #61: Feature request. Duplex HAT, swap Tx and Rx transceivers

rogerclarkmelbourne, 2018-08-15:
Hi Andy.

I order to debug problems on Duplex hotspots, I wonder if its worth having some way to swap the Tx and Rx transceivers.

I've had a look at the dual hat schematics and they seem to share the same SPI bus, but use a different chip select for each transcriver.  

I'm actually a bit surprised that they did this, rather than using separate SPI busses, but I guess it saves memory as you only need one SPI object.

juribeparada, 2018-08-15:
Hi Roger,

I don't believe that swapping TX and RX transceiver could help for debugging problems with duplex boards. At the moment SPI buses work fine, as far as I know. Even a small problem in data transfer with SPI bus will cause serious problems, like TX audio complete lost. 
It is not a problem to share the SPI bus, actually is the way that SPI bus standard is designed, you can have one master and several slaves devices, where you use several CS lines to select the slave device.
Remember that SPI bus is used for ADF7021 register configurations (low speed), and the ADF7021 digital data streams are through separate synchronous lines.

rogerclarkmelbourne, 2018-08-15:
Hi Andy

OK. I didn't realise the data was sent over separate lines.

I was hoping to swap the Tx and Rx over completely.  So I'll need to look at the code that defines the data stream pins as well as the SPI CS pin

closed 2018-08-18 by rogerclarkmelbourne