juribeparada/MMDVM_HS #62: Add support for CC1101 radio module

sp2ong, 2018-08-15:
Is it possible add support for use radio module CC1101 with hotspot instead ADF7021 ?

http://www.ti.com/lit/ds/symlink/cc1101.pdf

juribeparada, 2018-08-15:
I don't have plans to add new radio interfaces for now, due to my limited free time and the big effort that adding a new radio IC means. I prefer to work improving the current implementation (fixing bugs) and adding new features. Not sure if CC1101 is a good candidate for MMDVM_HS, it has not support for 2m band for example, and an ideal RF IC must support transparent data (not sync word detection, not packet handling.

rogerclarkmelbourne, 2018-08-15:
I had the same thought 2m (144-148Mhz) not being supported by that chip.

Also its focus seems on low power operation, which is also not necessary for a hotspot.

sp2ong, 2018-08-17:
Thank you for replay but for your argument for not support 2m band is not clear. If I am wrong that hotspot is generally for the purpose of personal access to DV network and has very low power radio like 10 mW for this reason 70 cm band it is OK.  If anybody would be like setup hotspot widerange rather should be used regular radio like Motorola GMxxx with output power 10/25 Watts

juribeparada, 2018-08-17:
Hi, it is not a matter of RF power, it's a matter of DV radio availability. Some people only have VHF handheld radios in some part of the world where UHF is not popular. Anyway, if I had sufficient spare time to add a new RF IC support, I would select an equal of better IC than ADF7021, with the same bands support at least.

sp2ong, 2018-08-17:
Hi Andy,

Thank you for the explanation. So, I think we can close this issue.

closed 2018-08-17 by sp2ong