juribeparada/MMDVM_HS #63: Some transceivers will not connect to the duplex hotspot

rogerclarkmelbourne, 2018-08-20:
(Note, this issue is in the official release 1.4.6 not just in the latest commit)

I have been testing my duplex hotspot with 2 different GD-77 and both connect without any problems

However I am not able to connect at all from my MD-380

With the MD-380, the PTT LED on the hotspot lights up immediatly, and I can hear the hotspot transmitting, if I listen on FM to its output frequency.

However after about 1 second, the MD-380 beeps and stops transmitting.

The COS LED does not illuminate on the hotspot.

On the GD-77 the COS LED illuminates about 0.25 secs after the PTT.

Can you suggest anything I can do to debug this issue ?

Thanks

juribeparada, 2018-08-20:
I think this is an issue only for MD380. I tried to solve this for long time without luck. The problem is the duplex board doesn't activate fast enough after MD380 sends a wake-up request, then the radio tries two times more without realize that the HS is already activated. At the moment I can't speed up the TX response, probably latency introduced by ADF7021 is too much for the total delay. It would be very easy a MD380 FW fix (adding a little more wait after sending the wake-up request), but it seems to me we don't have more official MD380 FW releases. Be aware this is a complete different situation compared with other radios with "similar" problems.

I think I will add soon a section in the README with the "known issues" and some explanations. As you see, duplex mode introduces some inherent problems in this design, and probably people must know those issues before acquiring those boards.

rogerclarkmelbourne, 2018-08-20:
Do you know where the latency occurs? 

Is it in the STM or perhaps just the time it takes for the ADF7021 to start transmitting.

I presume you are running the STM32 at 72MHz (9 x PLL multiplier), but the STM32F103 overclocks quite well, and you could try changing the PLL to its max value (I think its 16)

Normally overclocking isnt possible as USB will only work on 48 and 72MHz main clocks, but as the comms is via UART, its possible to overclock to 128Mhz 

Or if processor speed was really an issue, the F103 could be replaced by a F4 series with the same pinout, However the low level register access is a lot different on the F4 so there would need to be a lot of firmware changes at the low level.

Or possibly look at the compiler optimisation setting.

It looks like you are using -Os.

I guess I could try -O2 or -O3

BTW.
Still possible to compile this using the Arduino IDE?
I initially tried compiling using the IDE but it didnt seem to work, so I installed the gcc toolchain on my RPi 3B+ and used your make file

juribeparada, 2018-08-21:
AFAIK Arduino IDE should work, but I haven't tested lately with the duplex code. I remember some problems with large code with Arduino IDE (probably compilation flags).
Not sure about where the latency comes. It should be the sum of all contributions. But I think about MMDVM-POG, which uses a STM32F105, running at 72 MHz also, and I think that board doesn't have activation problems in duplex mode. At least buffering is almost the same between MMDVM and MMDVM_HS, then ADF7021 might be the problem. But the latency of ADF7021 is also very little according with the datasheet.
Another possibility is an initial TX "abnormality" of ADF7021. I remember that reducing RF power of TX or changing antennas positions improves the MD380 activation. I will try to reproduce this latter and see.

rogerclarkmelbourne, 2018-08-21:
AFIK the STM32F105 is very similar to the F103.

Using -03 has not helped on my MD-380, so I'll look at overclocking.
But I'll need to carefully look at the code., as all the delays may be wrong (if you use any), and also the baud rate will be incorrect on the UART, and if you are using SPI etc that will also be going faster and may be too fast for the ADF7021 etc etc

Re: Tx power and antenna position etc

I've tried 3 different antennas and it didnt make any difference, but I"ve not tied adjusting the power

juribeparada, 2018-08-21:
I am probably wrong about my first hypothesis. I increased the TX deviation (TXLevel=55) a little and it seems to me it is working. I remembered some issues with deviation and MD380, and probably this is the case, again. Please test different DMR deviation values. I will test more tomorrow, I have to sleep now.

rogerclarkmelbourne, 2018-08-21:
In PiStar, I tried changing the TXLevel to 55 but it didnt seem to make any difference.

But I noticed there was a separate DMRTXLevel setting, and when I changed this to 55 it seemed to fix the problem on the MD380

I now have them both set to 55

What does this setting do ?

rogerclarkmelbourne, 2018-08-21:
One time when I pressed the PTT, PiSTar showed that the GD-77 was on TS 1 not TS2.

Is this something to do with TXLevel (Deviation) or is it another problem :-(

rogerclarkmelbourne, 2018-08-21:
Andy

There seems to be some problems with changing DMRTXLevel

My GD-77 seems to now have problem accessing the hotspot.

Also although the MD-380 can access the hotspot all the time, I noticed in 10% of the time, PiStar shows that its transmitting on TS1 when its always transmitting on TS2.

I think I will set DMRTXLevel back to 50 and change back to using my GD-77 as its more stable

juribeparada, 2018-08-21:
Roger, of course, the deviation problem is only for MD380 AFAIK, not GD77. This could be related with some reports where users need to increase a little the DMR deviation for simplex HS and MD380, to avoid audio losses. Original settings (50 %) is adjusted according with the DMR standard. Maybe (just a hypothesis), ADF7021 might need a time to transmit at correct deviation at the beginning.
The TS problem also could be related with the same problem (increase a little more the deviation and test), maybe MD380 might decode wrong the first TX from HS and transmit at incorrect TS. Look at the log.

BD4QHI, 2018-08-28:
上述问题我是用的Hyterapd680 未出现此问题，硬件使用精度1%的电阻，电容，电感使用高频绕线电感，ECS14.7456晶振。
Hyterapd 680, which I used to solve the above problems, does not have this problem. The hardware uses 1% precision resistor, capacitor, inductor uses high frequency winding inductor, ECS 14.7456 crystal oscillator.

juribeparada, 2018-09-02:
Added a comment for MD380 in README.md. I will close this issue.

closed 2018-09-02 by juribeparada