juribeparada/MMDVM_HS #64: FYI. Compiler optimisation level

rogerclarkmelbourne, 2018-08-21:
Andy

Just a FYI.
As far as I can tell all build targets compile OK with compiler optimization setting  of -O3 (instead of -Os)

The duplex hotspot fw seems to run fine when built using -O3, but it does not seem to give any benefit to the operation.

But the firmware size is around 50% bigger when using this option.

The duplex hotspot firmware would only just fit in a 64k STM32F103C8 (albeit virtually all STM32F103C8 actually have 128k as I doubt STM really make the C8 version and simply label the CB version as C8)

There may potentially be less RAM usage when using -O3, but I've not checked, and I think generally -O3 just gives more speed.

juribeparada, 2018-08-22:
OK, thank you, I will have that in consideration if we need more speed in the future. For now, I think if that produces a RAM optimization, that would be very nice.

rogerclarkmelbourne, 2018-08-22:
I'm not sure about the RAM.

And its normally quite hard to determine how much RAM is free.

closed 2018-09-02 by juribeparada