juribeparada/MMDVM_HS #66: install_fw_nanodv_usb.sh

vk7hse, 2018-08-30:
MMDVM_HS/scripts  the file install_fw_nanodv_usb.sh is not marked as executable

`pi-star@pi-star-duplex(rw):MMDVM_HS$ ls -la scripts/
total 72
drwxr-xr-x 2 pi-star pi-star 4096 Aug 22 10:53 .
drwxr-xr-x 8 pi-star pi-star 4096 Aug 30 00:57 ..
-rwxr-xr-x 1 pi-star pi-star 3130 Jul 21 06:37 build_fw.sh
-rwxr-xr-x 1 pi-star pi-star 1228 Feb  9  2018 install_buildtools.sh
-rwxr-xr-x 1 pi-star pi-star 3172 Feb 27  2018 install_fw_custom.sh
-rwxr-xr-x 1 pi-star pi-star 2928 Aug 22 10:53 install_fw_duplex_gpio.sh
-rwxr-xr-x 1 pi-star pi-star 3015 Aug 22 10:53 install_fw_duplex.sh
-rwxr-xr-x 1 pi-star pi-star 2915 Aug 22 10:53 install_fw_gen_gpio.sh
-rwxr-xr-x 1 pi-star pi-star 2952 Aug 22 10:53 install_fw_hsdualhat-12mhz.sh
-rwxr-xr-x 1 pi-star pi-star 2940 Aug 22 10:53 install_fw_hsdualhat.sh
-rwxr-xr-x 1 pi-star pi-star 2963 Aug 22 10:53 install_fw_hshat-12mhz.sh
-rwxr-xr-x 1 pi-star pi-star 2907 Aug 22 10:53 install_fw_hshat.sh
-rwxr-xr-x 1 pi-star pi-star 3013 Aug 22 10:53 install_fw_librekit.sh
-rwxr-xr-x 1 pi-star pi-star 2893 Aug 22 10:53 install_fw_nanodv_npi.sh
-rw-r--r-- 1 pi-star pi-star 3001 Aug 22 10:53 install_fw_nanodv_usb.sh
-rwxr-xr-x 1 pi-star pi-star 2901 Aug 22 10:53 install_fw_nanohs.sh
-rwxr-xr-x 1 pi-star pi-star 2911 Aug 22 10:53 install_fw_rpi.sh
-rwxr-xr-x 1 pi-star pi-star 3003 Aug 22 10:53 install_fw_usb.sh
pi-star@pi-star-duplex(rw):MMDVM_HS$ `

phl0, 2018-08-30:
https://github.com/juribeparada/MMDVM_HS/pull/67

juribeparada, 2018-08-30:
Merged #67, thank you Flo

closed 2018-08-30 by juribeparada