juribeparada/MMDVM_HS #72: MMDVM Dual (V1.2) Issue on update

TheOriginalFreddoFrog, 2018-11-06:
Roger,

I am receiving this on a firmware upgrade on a standard "Chinese" MMDVM Dual 1.2 (on 2 arduino bases):

_http://github.com/rogerclarkmelbourne/arduino_stm32

Using Parser : Raw BINARY
Interface serial_posix: 57600 8E1
Failed to init device._

A Version 1.0 upgrades perfectly.

Any suggestions???

73.

Steve I
VK3VM / VK3SIR

TheOriginalFreddoFrog, 2018-11-06:
Roger,

Good call. I went over the STM with an iron … thwere saoldwer that the manufacturer was using was appaling and it took quite some time to "clean up" and ensure that the poor chip was properly soldered in....

And yes, I say poor chip....

It has now updated.

73 and thanks for the tip !!!

Steve I
VK3VM / VK3SIR

closed 2018-11-06 by TheOriginalFreddoFrog

rogerclarkmelbourne, 2018-11-06:
Glad that fixed it. Poor soldering is a problem that is reported a lot on various forums including the one I run www.stm32duino.com