juribeparada/MMDVM_HS #73: Failed to Init Device to 1.4.12 update via pi-star

dhchenault, 2018-11-20:
Been running, no problems. Updated previously using the ssh shell in pi-star. Tried it tonight with same command and received a failed to init device error. After reboot, it shows back to 1.4.7.

Any ideas?
David

Here's the text from the process:

pi-star@pi-star(ro):~$ sudo pistar-mmdvmhshatflash hs_hat                            
MMDVM_HS_Hat-v1.4.7 20180821 14.7456MHz ADF7021 FW by CA6JAU GitID #b14548e          
Press any key to write the hs_hat firmware to this modem or Ctrl-C to quit...        
Flashing your hs_hat modem to the latest version                                     
--2018-11-20 01:01:45--  http://github.com/juribeparada/MMDVM_HS/releases/download/v1
.4.12/install_fw_hshat.sh                                                            
Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112                  
Connecting to github.com (github.com)|192.30.253.113|:80... connected.               
HTTP request sent, awaiting response... 301 Moved Permanently                        
Location: https://github.com/juribeparada/MMDVM_HS/releases/download/v1.4.12/install_
fw_hshat.sh [following]                                                              
--2018-11-20 01:01:46--  https://github.com/juribeparada/MMDVM_HS/releases/download/v
1.4.12/install_fw_hshat.sh                                                           
Connecting to github.com (github.com)|192.30.253.113|:443... connected.              
HTTP request sent, awaiting response... 302 Found                                    
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/80589711/a3
81d300-e050-11e8-8ba3-5272f575399f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=
AKIAIWNJYAX4CSVEH53A%2F20181120%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20181120T0
70146Z&X-Amz-Expires=300&X-Amz-Signature=76e5cd2cc70273f285fb24b38f333d7aaaad7c125da2
3cb87fd958368f126696&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition
=attachment%3B%20filename%3Dinstall_fw_hshat.sh&response-content-type=application%2Fo
ctet-stream [following]                                                              
--2018-11-20 01:01:46--  https://github-production-release-asset-2e65be.s3.amazonaws.
com/80589711/a381d300-e050-11e8-8ba3-5272f575399f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-
Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20181120%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-
Date=20181120T070146Z&X-Amz-Expires=300&X-Amz-Signature=76e5cd2cc70273f285fb24b38f333
d7aaaad7c125da23cb87fd958368f126696&X-Amz-SignedHeaders=host&actor_id=0&response-cont
ent-disposition=attachment%3B%20filename%3Dinstall_fw_hshat.sh&response-content-type=
application%2Foctet-stream                                                           
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-
release-asset-2e65be.s3.amazonaws.com)... 52.216.128.107                             
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-product
ion-release-asset-2e65be.s3.amazonaws.com)|52.216.128.107|:443... connected.         
HTTP request sent, awaiting response... 200 OK                                       
Length: 2908 (2.8K) [application/octet-stream]                                       
Saving to: ‘/tmp/mmdvmhshatfirmware/flash.sh’                                        
                                                                                     
/tmp/mmdvmhshatfirmwa 100%[========================>]   2.84K  --.-KB/s   in 0.001s  
                                                                                     
2018-11-20 01:01:46 (3.16 MB/s) - ‘/tmp/mmdvmhshatfirmware/flash.sh’ saved [2908/2908
]                                                                                    
                                                                                     
FINISHED --2018-11-20 01:01:46--                                                     
Total wall clock time: 1.9s                                                          
Downloaded: 1 files, 2.8K in 0.001s (3.16 MB/s)                                      
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   609    0   609    0     0   1407      0 --:--:-- --:--:-- --:--:--  1409
100 48216  100 48216    0     0  53026      0 --:--:-- --:--:-- --:--:-- 53026       
Cloning into 'STM32F10X_Lib'...                                                      
remote: Enumerating objects: 5, done.                                                
remote: Counting objects: 100% (5/5), done.                                          
remote: Compressing objects: 100% (5/5), done.                                       
remote: Total 178 (delta 0), reused 3 (delta 0), pack-reused 173                     
Receiving objects: 100% (178/178), 3.27 MiB | 921.00 KiB/s, done.                    
Resolving deltas: 100% (60/60), done.                                                
Checking connectivity... done.                                                       
Raspberry Pi 2 or Pi Zero W detected                                                 
stm32flash Arduino_STM32_0.9                                                         
                                                                                     
http://github.com/rogerclarkmelbourne/arduino_stm32                                  
                                                                                     
Using Parser : Raw BINARY                                                            
Interface serial_posix: 57600 8E1                                                    
Failed to init device.                                                               
                                                                                     
                                                                                     
Flashing your hs_hat modem complete, press any key to reboot your Pi-Star System...  
Session closed.

rogerclarkmelbourne, 2018-11-21:
Someone else had a similar problem, and it was caused by a dry joint coming loose on their board.

The upload program uses the internal bootloader in the STM32, which is a hardware feature in the chip, which is activated when the chip is reset while the Boot0 pin is pulled high.
The data is sent from the RPi to the hotspot via hardware serial (Tx and Rx pins).

So generally the only way this can fail is if one of those connections is broken.

Or if perhaps there is a problem with the hardware serial or GPIO pin drivers on the RPi

dhchenault, 2019-01-06:
SOLVED - Found a couple of similar reports on the net. I am running a Nextion screen on the Modem interface which apparently conflicted with the firmware update. Unplugged the screen (could not seem to disable it via Pi-Star) and then successfully updated firmware.

closed 2019-01-19 by juribeparada

porpup, 2025-05-23:
[image]

[image]

porpup, 2025-05-23:
[mmdvm_firmware_upgrade.txt](https://github.com/user-attachments/files/20402398/mmdvm_firmware_upgrade.txt)