juribeparada/MMDVM_HS #74: After upgrade to  1.4.12 (via Pi-STAR) i receive NAK from modem

sm0tsc, 2018-12-10:
I: 2018-12-10 09:43:24.136     RX Frequency: 433450000Hz (433450000Hz)
I: 2018-12-10 09:43:24.137     TX Frequency: 433450000Hz (433450000Hz)
M: 2018-12-10 09:43:24.138 Opening the MMDVM
I: 2018-12-10 09:43:26.149 MMDVM protocol version: 1, description: ZUMspot-v1.4.     12 20181104 14.7456MHz ADF7021 FW by CA6JAU GitID #dec7051
E: 2018-12-10 09:43:26.160 Received a NAK to the SET_FREQ command from the modem

Even thou that QRG is outside sat allocations..

Please advise

//SM0TSC

juribeparada, 2018-12-10:
Did you check POCSAG frequency?

sm0tsc, 2018-12-10:
> Did you check POCSAG frequency?

Pocsag was disabled but indeed it was in the 437 range and after enabled and changed it works

closed 2018-12-16 by juribeparada