juribeparada/MMDVM_HS #75: 10Mhz reference clock support

stonegray, 2018-12-11:
Hi all,

Is it possible to modify the firmware to use a 10Mhz clock in place of the 14/12Mhz oscillator?  This would allow use of an external clock from standard lab reference clocks, inexpensive [atomic clocks](https://www.ebay.com/p/Fe-5680a-Rubidium-Oscillator-10mhz-Programmable-Version/12026884989?iid=113435040692), [homemade](https://www.eevblog.com/forum/projects/lars-diy-gpsdo-with-arduino-and-1ns-resolution-tic/) or  [off the shelf GPSDOs](https://www.ebay.com/itm/PLL-GPSDO-GPS-Tame-Disciplined-Clock-Sine-Wave-GPS-Receiver-10M-1PPS/112691569933?hash=item1a3cf0e50d:g:mkYAAOSwVtZaM2Cb:rk:11:pf:0) (GPS disciplined oscillators), and cheap high precision [OCXO](https://www.ebay.com/itm/New-ultra-precision-high-quality-UCT-10-Mhz-Double-Oven-OCXO-crystal-oscillator/202508294657?hash=item2f266f5a01:g:kCQAAOSwFe5X0fV0:rk:7:pf:0), or the sync output from other open source radio devices that support clock I/O such as the HackRF, all which use a 10Mhz standard.

While I don't think everybody wants to go spend $100 on an rubidium atomic clock to keep their timing right, I'm sure lots of us would consider trying a [$13 GPS module](http://la3za.blogspot.com/2015/10/just-good-enough-10-mhz-reference.html) and a PLL to get the timing from one in space. 

Also I imagine there's a few of us with a 'house clock' in their shack :)

juribeparada, 2018-12-12:
Hi,
there is a reason of using frequencies 14.7456 MHz and 12.2880 MHz (and a few other non-integer frequencies that I will not implement): basically because those frequencies produce a proper baudrate for the implemented digital modes, other common frequencies like 10 MHz can't produce proper baurate with ADF7021, then will not work.

stonegray, 2018-12-12:
Darn, I’ll add a PLL then. Thank you.

closed 2018-12-12 by stonegray