juribeparada/MMDVM_HS #78: Getting partial tx audio echo on duplex board (MMDVM_HS_Dual_Hat)

af5bc, 2019-01-13:
After every transmit, I receive a partial audio echo of my last transmission on my MMDVM_HS_Dual_Hat based hot spot.

This may be similar to the issue documented in: https://github.com/g4klx/MMDVM/issues/181

Note, I do not see this issue on my simplex (MMDVM_HS_Hat) boards.

I running a duplex MMDVM_HS_Hat DB9MAT DF2ET DO7EN board running firmware 1.4.14, and Pi-Star:3.4.17 / Dashboard: 20190109.

here is the output of "pistar-mmdvmhshatflash":
Selected Modem:     MMDVM_HS_Dual_Hat (DB9MAT, DF2ET & DO7EN) for Pi (GPIO)
Connected Hardware: MMDVM_HS_Dual_Hat-v1.4.14 20181209 14.7456MHz dual ADF7021 FW by CA6JAU GitID #8d77ff3
TCXO Value:         14.7456MHz

rogerclarkmelbourne, 2019-01-13:
I presumed this is unavoidable, caused by internal buffering inside the duplex hotspot.
i.e its re transmitting your audio all the time with a slight delay, which you can head if you listen to the output frequency of the hotspot.

Hence when you release the PTT, your transceiver will start listening to the output from the hotspot and hear a small amount of delayed audio.

I presume whats being transmitted comes from MMDVMHost (running as part of PiStar), which is likely to be where the majority of the delay comes from. 

However if the received audio data from MMDVMHost into the hotspot comes from the server, then network latency will make this issue worse.

af5bc, 2019-01-14:
So, my intended use is is Half-duplex on hotspots. Is there a special configuration setting to disable the audio re-transment?

rogerclarkmelbourne, 2019-01-14:
I'm not sure exactly what you mean by Half-Duplex. AFIK DMR transceivers don't operate in full duplex, like a landline or mobile phone, where you are both receiving and transmitting audio at the same time.

Do you mean you don't want the hotspot to transmit when you are transmitting and its receiving your signal. If thats the case, you have to select that its a Simplex hotspot in PiStar and also select the MMDVM_HS_Hat as the "Radio/Modem type" in PiStar

In this configuration, it only uses the ADF7021 connected to the SMD socket labelled Rx

I don't know of a way to use one ADF7021 for Rx and the other one for Tx, when it not operating in duplex mode.

af5bc, 2019-01-14:
I purchased duplex boards to solve the problem where I do not have access to my dashboard (mobile) and I'm stuck on a talk group that has a couple of rag chewers who refuse to leave any gaps in their transmissions.
 
With a simplex board, I'm stuck until they run out of hot air. But with the duplex board I can send a de-link command (TG4000). So, I'm not really intending to use use the MMDVM_HS_Dual_Hat to create a repeater. Instead, using it to give me an independent rx path so I can change/drop active talk groups.

juribeparada, 2019-02-14:
The partial audio echo is a "common" issue of MMDVM duplex mode. For now you have to live with that, it is related with the buffering scheme in MMDVM firmware and MMDVMHost, not easy to solve properly. Probably will be solved in the future.

closed 2019-02-14 by juribeparada