juribeparada/MMDVM_HS #79: Offset -4000 required after firmware update

NF0Q, 2019-01-20:
I have two S56AL MMDVM HS Hat boards that I bought from Ebay. One is a version 1 board and the other is a version 2 board. I have talked to Alexsander(S56AL) and confirmed both are china clones. Other than replacing the TCXO (which I did) he said the boards should work fine and they do. 

I had no problem updating either board to the latest firmware (v1.4.14). However on the version 1 board after the update, I have to set the offset to -4000 in order to get it on frequency. Other than the odd offset, the board works fine with the new firmware. The version 2 board also works fine after updating but doesn't require any offset adjustments. The version 1 board required no offset prior to the update.

Any idea why the version 1 board would require a -4000 offset after the update?

juribeparada, 2019-01-20:
No idea, only I can tell you that I haven't modified the VCO/PLL code or anything regarding frequency adjustment. Personally I've never used such high frequency offsets with my boards. But probably some strange BER or "no RX" problems are related with that. I wonder if a "fake" ADF7021 may produce that (since you replaced TCXO), and a pre-installed firmware (with a frequency fix) is shipped with some MMDVM_HS clones. I can't prove that, but it is common that after a firmware update (any firmware version, not just v1.4.14) of a pre-installed firmware, the user has to apply a big offset. Maybe if you try to install the exact previous fw version you can test if you still need -4000 Hz of offset or not.

NF0Q, 2019-01-20:
Thanks Andy! I too suspect something was up with the pre-installed firmware.

closed 2019-01-27 by juribeparada