juribeparada/MMDVM_HS #81: compile issue on git commit ed487bcad455158a2c1f0703ee5a5f09e809ae49

vk7hse, 2019-02-07:
Getting the following error,

/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv7-m/libstdc++_nano.a(new_opv.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv7-m/libstdc++_nano.a(new_op.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv7-m/libstdc++_nano.a(new_handler.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
obj_f1/SerialSTM.o: In function `CSerialPort::beginInt(unsigned char, int)':
SerialSTM.cpp:(.text._ZN11CSerialPort8beginIntEhi+0x18): undefined reference to `usbserial'
SerialSTM.cpp:(.text._ZN11CSerialPort8beginIntEhi+0xc): undefined reference to `USBSerial::begin()'
obj_f1/SerialSTM.o: In function `CSerialPort::availableInt(unsigned char)':
SerialSTM.cpp:(.text._ZN11CSerialPort12availableIntEh+0x20): undefined reference to `usbserial'
SerialSTM.cpp:(.text._ZN11CSerialPort12availableIntEh+0x16): undefined reference to `USBSerial::available()'
obj_f1/SerialSTM.o: In function `CSerialPort::readInt(unsigned char)':
SerialSTM.cpp:(.text._ZN11CSerialPort7readIntEh+0x14): undefined reference to `USBSerial::read()'
SerialSTM.cpp:(.text._ZN11CSerialPort7readIntEh+0x20): undefined reference to `usbserial'
obj_f1/SerialSTM.o: In function `CSerialPort::writeInt(unsigned char, unsigned char const*, unsigned short, bool)':
SerialSTM.cpp:(.text._ZN11CSerialPort8writeIntEhPKhtb+0x18): undefined reference to `USBSerial::write(unsigned char const*, unsigned long)'
SerialSTM.cpp:(.text._ZN11CSerialPort8writeIntEhPKhtb+0x3c): undefined reference to `usbserial'
SerialSTM.cpp:(.text._ZN11CSerialPort8writeIntEhPKhtb+0x24): undefined reference to `USBSerial::flush()'
collect2: error: ld returned 1 exit status
Makefile:317: recipe for target 'bin/mmdvm_f1.elf' failed
make: *** [bin/mmdvm_f1.elf] Error 1

Seems to be cause from /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv7-m/libstdc++_nano.a(new_handler.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail

phl0, 2019-02-07:
Please show the Config.h and the command that you used for building. Which version / git commit do you use?

vk7hse, 2019-02-07:
Well I have to admit that I had pulled in the source but I failed to copy over from the config directory the correct Config.h for the Zumspot! So now I have just corrected the oversight on my behalf all is good and this is invalid!

closed 2019-02-07 by vk7hse