juribeparada/MMDVM_HS #82: make dfu problem

SA7BNT, 2019-02-10:
Hallo, i try to flash via USB but dfu cant find a device.
I´m running 
make clean
make bl 
make dfu
and getting that issue
[image]
Best 
SA7BNT

juribeparada, 2019-02-14:
When you use "make dfu", you also need to specify the serial-usb device port to the command, to perform the reset automatically, and use "sudo". For example:

sudo make dfu devser=/dev/ttyACM0

If that doesn't work, you can use:
1) Press reset button of the MMDVM_HS board
2) Wait 1 or 2 second
3) Execute: sudo make dfu

For more details, see BUILD.md: "Makefile options"

dl1bz, 2019-02-20:
Don't work for me too.
pi-star@mmdvm-dl1bz02(rw):MMDVM_HS$ sudo make dfu devser=/dev/ttyACM0
././STM32F10X_Lib/utils/rpi32/upload-reset /dev/ttyACM0 750
././STM32F10X_Lib/utils/rpi32/dfu-util -D bin/mmdvm_f1bl.bin -d 1eaf:0003 -a 2 -R -R
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
Makefile:468: recipe for target 'dfu' failed
make: *** [dfu] Error 74

It's not important if I press RESET or not. **The only way for success is the the modem connect with GPIO, set BOOT0 to 1 , press RESET and flash with sudo make serial-bl-old devser=/dev/ttyAMA0** .
sudo make serial-bl don't work too - only make serial-bl-old.
But flashing over USB - no success. And yes, I read the BUILD.md not only one time :)
Check with Pi 3B+ and a Linux-PC with debian, modem is MMDVM_HS_DUAL V1.3

Best, Heiko, DL1BZ

juribeparada, 2019-02-20:
Unfortunately, the USB bootloader used in MMDVM_HS has issues with Pi 3B+ and some USB host controllers and doesn't work correctly with all USB. You can test two options:
1) Old bootloader (serial-bl-old), in that case, you need to have LONG_USB_RESET disabled in Config.h (you will need to compile the firmware)
2) Current bootloader (serial-bl), with LONG_USB_RESET enabled in Config.h (default binary firmware).
For your 3B+, you need to use serial-bl in order to install the new bootloader (not serial-bl-old, see README.md). You could even try to run "sudo make dfu" 1 or 2 seconds after plug your device into the USB port (if reset doesn't work).

dl1bz, 2019-02-20:
Ok news:
now it works with the new bootloader on GPIO, but only if I insert a "sleep 3" in your Makefile between flashing the loader and in second stage the firmware. Without a short break, it will not write the firmware and stop. It's simply to fast between flashing the two parts.

dl1bz, 2019-02-20:
again success:
with the new bootloader now the USB flashing works on Pi after pressing RESET:
pi-star@mmdvm-dl1bz02(rw):MMDVM_HS$ sudo make dfu devser=/dev/ttyACM0
././STM32F10X_Lib/utils/rpi32/upload-reset /dev/ttyACM0 750
././STM32F10X_Lib/utils/rpi32/dfu-util -D bin/mmdvm_f1bl.bin -d 1eaf:0003 -a 2 -R -R
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1eaf:0003
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download        [=========================] 100%        62216 bytes
Download done.
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode
pi-star@mmdvm-dl1bz02(rw):MMDVM_HS$

Thanks for assistance,
73 Heiko, DL1BZ

juribeparada, 2019-02-20:
Could you tell me where did you put the "sleep 3"?, maybe could be useful to add it to the repository.

dl1bz, 2019-02-20:
in Makefile:
serial-bl:
$(STM32FLASH) -v -w $(F1_LIB_PATH)/utils/bootloader/generic_boot20_pc13_long_rst.bin -g 0x0 $(devser)
**sleep 3**
$(STM32FLASH) -v -w bin/$(BINBIN_F1BL) -g 0x0 -S 0x08002000 $(devser)

with the sleep 3 the time is long enough for the second stage flashing will be continue.

phl0, 2019-02-20:
Like that? https://github.com/juribeparada/MMDVM_HS/compare/master...phl0:sleep

dl1bz, 2019-02-20:
yes 👍

phl0, 2019-02-20:
Roger. @juribeparada Shall I PR that?

juribeparada, 2019-02-20:
Flo: yes, please, thank you. And also thank you Heiko for that suggestion.

dl1bz, 2019-02-20:
perfect. I check this with 3 different modems with my Pi3B+, in all cases flashing was sucessful.
thanks guys, 73 Heiko, DL1BZ

phl0, 2019-02-20:
See https://github.com/juribeparada/MMDVM_HS/pull/86

closed 2019-02-20 by SA7BNT