juribeparada/MMDVM_HS #83: The generated firmware image using the Ubuntu 18.04 gcc-arm (6.3.1) toolchain doesn't work

af5bc, 2019-02-10:
Here is the default gcc-arm packages installed on Ubuntu 18.04 LTS:
gcc-arm-none-eabi is already the newest version (15:6.3.1+svn253039-1build1).
libnewlib-arm-none-eabi is already the newest version (2.4.0.20160527-3).
libstdc++-arm-none-eabi-newlib is already the newest version (15:6.3.1+svn253039-1+10).

I would like to downgrade my tools to match the version you are using for the firmware releases. Which version is that?

af5bc, 2019-02-10:
Update, tried installing  the latest from ppa:team-gcc-arm-embedded/ppa (7.3.1) and that seems to work.

What is the recommended release of the gcc-arm-embedded tool chain>

juribeparada, 2019-02-14:
Normally I use the gcc-arm-embedded tool chain version 5.4.1, 20160919, for macOS. Normally works OK, and I build all the binaries releases using this version. Good to know that latest version 7.3.1 also works, probably I will update too.

juribeparada, 2019-02-20:
The latest binary firmware release is now built using the gcc-arm tool version 8.2.1 (20181213)

closed 2019-02-20 by af5bc