juribeparada/MMDVM_HS #85: 1.4.16: Overflow in the DMR slot RF queue with 12MHz Dual HS-Hat

szikora, 2019-02-19:
I'm using a Rpi3 with Pi-Star (3.4.17) and a v1.3 MMDVM_HS_Dual_Hat (12MHz TCXO version). Works without any trouble with the 1.4.14 firmware. Flashed the 1.4.16 with the pistar tool and the RF is not usable anymore with an "overflow in the DMR slot RF queue" message in the mmdvmhost log. Reflashed several time with 1.4.16 release firmware, no change. Need to downgrade back to 1.4.14 in order to have again a working HS.

Thanks,

Jean-Pierre

juribeparada, 2019-02-19:
When you receive that message, it means your flashed with a wrong firmware type. Please tell me what command did you use for flashing, or contact the board manufacturer to know the correct firmware type, I don't have any 12 MHz board for testing.

szikora, 2019-02-19:
With just a pistar-mmdvmhshatflash:
This tool is used to flash your MMDVM_HS Hat Modem
to use the command, you will need to pass the name of your modem,
please choose form the list below...

To flash HS_HAT (14.7456MHz TCXO) connected to GPIO, run: pistar-mmdvmhshatflash hs_hat
To flash HS_HAT (12.2880MHz TCXO) connected to GPIO, run: pistar-mmdvmhshatflash hs_hat-12mhz
To flash HS_DUAL_HAT (14.7456MHz TCXO) connected to GPIO, run: pistar-mmdvmhshatflash hs_dual_hat
To flash HS_DUAL_HAT (12.2880MHz TCXO) connected to GPIO, run: pistar-mmdvmhshatflash hs_dual_hat-12mhz

To flash ZumSPOT devices by please use pistar-zumspotflash
To flash MMDVM_HS Nano Hat devices please use pistar-vyehsflash
To flash your BD7KLE/BG3MDO devices please use pistar-mdoflash

Selected Modem:     MMDVM_HS_Dual_Hat (DB9MAT, DF2ET & DO7EN) for Pi (GPIO)
Connected Hardware: MMDVM_HS_Dual_Hat-v1.4.14 20181209 12.2880MHz dual ADF7021 FW by CA6JAU GitID #8d77ff3
TCXO Value:         12.2880MHz

And I flashed it with pistar-mmdvmhshatflash hs_dual_hat-12mhz.

To downgrade it back to 1.4.14, I just use
pistar-mmdvmhshatdowngrade hs_dual_hat-12mhz

I don't think I've made a mistake with the firmware type. I've updated the firmware a few times before with the same command.

juribeparada, 2019-02-19:
Please contact Florian DF2ET to see if he can reproduce the issue, I don't have a hs_dual_hat-12mhz board to see by myself.

phl0, 2019-02-19:
I think 1.4.16 binary release of the firmware has a problem. We are currently investigating.

vk4nsp, 2019-02-19:
1.4.16 works fine here on a HS_Hat 14.756MHz

Maybe only affecting Dual HS_Hat

Regards,
Shaun VK4NSP

szikora, 2019-02-19:
I've access to a Dual HS_Hat with 14.756MHz TCXO, and no problem with 1.4.16. Problems seems to be very specific to the 12.2880MHz Dual HS_Hat.

juribeparada, 2019-02-19:
Please test now. I replaced all binaries with a compilation with latest arm gcc tools. I was using an old version, and it was working good until now. It seems to me that could be the source of the issue. Anyway, I will continue to investigate more.

szikora, 2019-02-19:
Thanks! Works great again this new compilation...
From the log:
I: 2019-02-19 13:48:14.035 MMDVM protocol version: 1, description: MMDVM_HS_Dual_Hat-v1.4.16 20190211 12.2880MHz dual ADF7021 FW by CA6JAU GitID #8cf46ca
M: 2019-02-19 13:48:14.724 MMDVMHost-20181107_Pi-Star is running
M: 2019-02-19 13:48:24.772 DMR, Logged into the master successfully
M: 2019-02-19 13:51:24.813 Downlink Activate received from ON4JPS
M: 2019-02-19 13:51:25.154 DMR Slot 1, received RF voice header from ON4JPS to TG 9
M: 2019-02-19 13:51:30.675 DMR Slot 1, received RF end of voice transmission, 5.4 seconds, BER: 0.2%, RSSI: -47/-47/-47 dBm

af5bc, 2019-02-19:
Will there be a version 17 to indicate the new binary generated with latest tools, or will we need to reflash version 16 on all of our boards?

phl0, 2019-02-19:
For now it means re-flashing the boards with 1.4.16 firmware. The binaries in the release have been rebuilt and exchanged.

juribeparada, 2019-02-26:
This is solved, closing

closed 2019-02-26 by juribeparada