juribeparada/MMDVM_HS #87: Compiler warnings...

dl1bz, 2019-02-26:
with "make bl" I get compiler warnings with the linker:
arm-none-eabi-g++ obj_f1/DMRDMOTX.o obj_f1/IOSTM.o obj_f1/DMRTX.o obj_f1/POCSAGTX.o obj_f1/SerialArduino.o obj_f1/MMDVM_HS.o obj_f1/SerialPort.o obj_f1/DStarRX.o obj_f1/IO.o obj_f1/YSFTX.o obj_f1/DStarTX.o obj_f1/DMRIdleRX.o obj_f1/P25TX.o obj_f1/NXDNRX.o obj_f1/CalDMR.o obj_f1/Utils.o obj_f1/BitRB.o obj_f1/SerialRB.o obj_f1/IOArduino.o obj_f1/YSFRX.o obj_f1/DMRDMORX.o obj_f1/DMRSlotRX.o obj_f1/DMRRX.o obj_f1/DMRSlotType.o obj_f1/ADF7021.o obj_f1/NXDNTX.o obj_f1/CalRSSI.o obj_f1/I2CHost.o obj_f1/SerialSTM.o obj_f1/P25RX.o obj_f1/CWIdTX.o obj_f1/stm32f10x_i2c.o obj_f1/stm32f10x_rcc.o obj_f1/stm32f10x_usart.o obj_f1/misc.o obj_f1/stm32f10x_iwdg.o obj_f1/stm32f10x_gpio.o obj_f1/stm32f10x_exti.o obj_f1/system_stm32f10x.o obj_f1/startup_stm32f10x.o obj_f1/usb_serial.o obj_f1/usb_mem.o obj_f1/usb_cdcacm.o obj_f1/usb_init.o obj_f1/usb.o obj_f1/usb_reg_map.o obj_f1/usb_core.o obj_f1/usb_regs.o -Os --specs=nano.specs --specs=nosys.specs -T bootloader.ld -mcpu=cortex-m3 -march=armv7-m -mthumb -Wall -Wextra  -o bin/mmdvm_f1bl.elf
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: **warning: /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv7-m/libstdc++_nano.a(new_opv.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail**
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: **warning: /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv7-m/libstdc++_nano.a(new_op.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail**
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: **warning: /usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/lib/armv7-m/libstdc++_nano.a(new_handler.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail**
Linking complete!

I don't know it's important - otherwise you need to check your code :)

73 Heiko, DL1BZ

phl0, 2019-02-26:
Seen those warnings before but couldn't find a solution. So as long as its just warnings ... Ignored them.

dl1bz, 2019-02-26:
I found the correct solution by myself :)
Look here: [https://community.nxp.com/thread/389154](url)

Important to know:
"There is a standard know 'issue' with interworking code compiled with different toolchains, which is a result of the ARM AEABI documentation allowing the size of wchar_t to be compiler defined. GCC uses a wchar_t size of 4, but some other toolchains use a size of 2."

"you can solve this problem by simply adding the linker flag "-no-wchar-size-warning" " :))

73 Heiko, DL1BZ

phl0, 2019-02-26:
I see. Then we maybe should add this to he Makefile :)

dl1bz, 2019-02-26:
yes that should be clear the warnings.

phl0, 2019-02-26:
Seems not so easy. That command line option is for ld. But in Makefile g++ is used for linking. And that doesn't know "--no-wchar-size-warning" ....

juribeparada, 2019-02-26:
Interesting the "-no-wchar-size-warning"...are you testing on Raspbian?. At least using latest arm gcc tools on macOS (the system where I build all the official fw binaries), I haven't seen those warnings. I always thought this warning issue was an arm gcc version related.

phl0, 2019-02-26:
Tested on my Ubuntu 18.04 x86_64 box here. The option seems to be only option for ld not for g++ ...

juribeparada, 2019-02-26:
Are you testing on the common LDFLAGS variable?

phl0, 2019-02-26:
jepp.