juribeparada/MMDVM_HS #94: 1.4.16: Overflow in the DMR slot 2 RF queue with 14.7456 MHz Dual HS-Hat

Rafa-RGB, 2019-03-17:
Hi,
B4 all, thans for reading.

I read the issue abt 12Mhz, and double checked (update TWICE my two MMDVM boards' FW) just in case...
I've even loaded in one of them the 12Mhz version (just for fun...), same results.

Scenario:
Raspi3b (NOT +), Pi-Star:3.4.17 / Panell de Control: 20190310
Linux bitxa 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l 

There NO WAY to keep them PTT "OFF" when they receive the DMR+ introcution speech telling connected reflector....

Here you have my full log:

Starting logging, please wait...
I: 2019-03-17 21:27:32.941 MMDVM protocol version: 1, description: MMDVM_HS_Dual_Hat-v1.4.16 20190211 14.7456MHz dual ADF7021 FW by CA6JAU GitID #8cf46ca
I: 2019-03-17 21:27:32.961 Display Parameters
I: 2019-03-17 21:27:32.961 Type: None
W: 2019-03-17 21:27:32.961 No valid display found, disabling
I: 2019-03-17 21:27:32.965 DMR Network Parameters
I: 2019-03-17 21:27:32.965 Address: 127.0.0.1
I: 2019-03-17 21:27:32.965 Port: 62031
I: 2019-03-17 21:27:32.965 Local: 62032
I: 2019-03-17 21:27:32.965 Jitter: 360ms
I: 2019-03-17 21:27:32.965 Slot 1: enabled
I: 2019-03-17 21:27:32.965 Slot 2: enabled
I: 2019-03-17 21:27:32.965 Mode Hang: 1s
I: 2019-03-17 21:27:32.966 Info Parameters
I: 2019-03-17 21:27:32.966 Callsign: EA3BIL
I: 2019-03-17 21:27:32.966 RX Frequency: 433475000Hz
I: 2019-03-17 21:27:32.966 TX Frequency: 439475000Hz
I: 2019-03-17 21:27:32.966 Power: 1W
I: 2019-03-17 21:27:32.966 Latitude: 41.556023deg N
I: 2019-03-17 21:27:32.966 Longitude: 2.033664deg E
I: 2019-03-17 21:27:32.966 Height: 0m
I: 2019-03-17 21:27:32.966 Location: "Terrassa JN11an"
I: 2019-03-17 21:27:32.966 Description: "Country"
I: 2019-03-17 21:27:32.966 URL: "http://ea3bil.ddns.net:8081"
M: 2019-03-17 21:27:32.966 DMR, Opening DMR Network
I: 2019-03-17 21:27:32.966 RSSI
I: 2019-03-17 21:27:32.966 Mapping File: /usr/local/etc/RSSI.dat
I: 2019-03-17 21:27:32.967 Loaded 14 RSSI data mapping points from /usr/local/etc/RSSI.dat
I: 2019-03-17 21:27:32.967 DMR Id Lookups
I: 2019-03-17 21:27:32.967 File: /usr/local/etc/DMRIds.dat
I: 2019-03-17 21:27:32.967 Reload: 24 hours
I: 2019-03-17 21:27:33.668 Loaded 125330 Ids to the DMR callsign lookup table
I: 2019-03-17 21:27:33.668 DMR RF Parameters
I: 2019-03-17 21:27:33.668 Started the DMR Id lookup reload thread
I: 2019-03-17 21:27:33.668 Id: 214338103
I: 2019-03-17 21:27:33.668 Color Code: 2
I: 2019-03-17 21:27:33.668 Self Only: no
I: 2019-03-17 21:27:33.668 Embedded LC Only: no
I: 2019-03-17 21:27:33.668 Dump Talker Alias Data: yes
I: 2019-03-17 21:27:33.668 Prefixes: 0
I: 2019-03-17 21:27:33.668 Call Hang: 1s
I: 2019-03-17 21:27:33.668 TX Hang: 1s
I: 2019-03-17 21:27:33.668 Mode Hang: 1s
M: 2019-03-17 21:27:33.669 MMDVMHost-20181107_Pi-Star is running
M: 2019-03-17 21:27:43.706 DMR, Logged into the master successfully
M: 2019-03-17 21:27:54.247 DMR Slot 2, received network late entry from 4714 to TG 8
E: 2019-03-17 21:27:56.048 DMR Slot 2, overflow in the DMR slot RF queue
E: 2019-03-17 21:27:56.058 DMR Slot 2, overflow in the DMR slot RF queue
E: 2019-03-17 21:27:56.104 DMR Slot 2, overflow in the DMR slot RF queue
E: 2019-03-17 21:27:56.104 DMR Slot 2, overflow in the DMR slot RF queue
E: 2019-03-17 21:27:56.104 DMR Slot 2, overflow in the DMR slot RF queue
E: 2019-03-17 21:27:56.104 DMR Slot 2, overflow in the DMR slot RF queue
E: 2019-03-17 21:27:56.115 DMR Slot 2, overflow in the DMR slot RF queue
E: 2019-03-17 21:27:56.115 DMR Slot 2, overflow in the DMR slot RF queue
E: 2019-03-17 21:27:56.166 DMR Slot 2, overflow in the DMR slot RF queue
E: 2019-03-17 21:27:56.166 DMR Slot 2, overflow in the DMR slot RF queue
[... 573 lines cut ...]
M: 2019-03-17 21:42:59.392 Downlink Activate received from EA3BIL
M: 2019-03-17 21:42:59.678 DMR Slot 2, received RF voice header from EA3BIL to TG 8
M: 2019-03-17 21:43:05.256 DMR Slot 2, received RF end of voice transmission, 5.4 seconds, BER: 0.6%, RSSI: -47/-47/-47 dBm
M: 2019-03-17 21:46:03.333 DMR Slot 2, received network voice header from EB3TC to TG 8
M: 2019-03-17 21:46:22.617 DMR Slot 2, received network end of voice transmission, 19.6 seconds, 1% packet loss, BER: 1.5%
M: 2019-03-17 21:46:44.789 DMR Slot 2, received network voice header from EB3TC to TG 8
M: 2019-03-17 21:47:03.403 DMR Slot 2, received network end of voice transmission, 18.5 seconds, 0% packet loss, BER: 1.4%
M: 2019-03-17 21:48:03.756 DMR Slot 2, received network voice header from EB3TC to TG 8
M: 2019-03-17 21:48:08.676 DMR Slot 2, received network end of voice transmission, 4.8 seconds, 0% packet loss, BER: 1.4%
M: 2019-03-17 21:48:24.321 DMR Slot 2, received network voice header from EB3TC to TG 8
M: 2019-03-17 21:48:30.691 DMR Slot 2, received network end of voice transmission, 6.2 seconds, 0% packet loss, BER: 1.5%

It HANGED with PTT ON after the second RX (no OVERFLOW showed on log...).
I was unable to access my MMDVM.
Asked the colleague to PTT again,... My boards do release PTT and stop bursting.

I hope this helps to clarify where the problem could be.

If you need the boards, It will be my pleasure to ship them...

NOTE:
BTW, It did NOT happened with 1.4.14

Thanks,
Rafa.

Rafa-RGB, 2019-03-17:
New try... Set Display port to NONE (just in case...)

LOG:
Starting logging, please wait...
M: 2019-03-17 22:15:14.942 Closing the MMDVM
I: 2019-03-17 22:15:15.787 Stopped the DMR Id lookup reload thread
M: 2019-03-17 22:15:15.787 DMR, Closing DMR Network
I: 2019-03-17 22:15:15.787 MMDVMHost-20181107_Pi-Star exited on receipt of SIGTERM
I: 2019-03-17 22:15:20.433 This software is for use on amateur radio networks only,
I: 2019-03-17 22:15:20.433 it is to be used for educational purposes only. Its use on
I: 2019-03-17 22:15:20.433 commercial networks is strictly prohibited.
I: 2019-03-17 22:15:20.433 Copyright(C) 2015-2018 by Jonathan Naylor, G4KLX and others
M: 2019-03-17 22:15:20.433 MMDVMHost-20181107_Pi-Star is starting
M: 2019-03-17 22:15:20.433 Built 09:24:10 Nov 12 2018 (GitID #9444eca)
I: 2019-03-17 22:15:20.433 General Parameters
I: 2019-03-17 22:15:20.433 Callsign: EA3BIL
I: 2019-03-17 22:15:20.433 Id: 214338103
I: 2019-03-17 22:15:20.433 Duplex: yes
I: 2019-03-17 22:15:20.433 Timeout: 240s
I: 2019-03-17 22:15:20.433 D-Star: disabled
I: 2019-03-17 22:15:20.434 DMR: enabled
I: 2019-03-17 22:15:20.434 YSF: disabled
I: 2019-03-17 22:15:20.434 P25: disabled
I: 2019-03-17 22:15:20.434 NXDN: disabled
I: 2019-03-17 22:15:20.434 POCSAG: disabled
I: 2019-03-17 22:15:20.434 Modem Parameters
I: 2019-03-17 22:15:20.434 Port: /dev/ttyAMA0
I: 2019-03-17 22:15:20.434 Protocol: uart
I: 2019-03-17 22:15:20.434 RX Invert: no
I: 2019-03-17 22:15:20.434 TX Invert: yes
I: 2019-03-17 22:15:20.434 PTT Invert: no
I: 2019-03-17 22:15:20.434 TX Delay: 100ms
I: 2019-03-17 22:15:20.434 RX Offset: 0Hz
I: 2019-03-17 22:15:20.434 TX Offset: 0Hz
I: 2019-03-17 22:15:20.434 RX DC Offset: 0
I: 2019-03-17 22:15:20.434 TX DC Offset: 0
I: 2019-03-17 22:15:20.434 RF Level: 100.0%
I: 2019-03-17 22:15:20.434 DMR Delay: 0 (0.0ms)
I: 2019-03-17 22:15:20.434 RX Level: 50.0%
I: 2019-03-17 22:15:20.434 CW Id TX Level: 50.0%
I: 2019-03-17 22:15:20.434 D-Star TX Level: 50.0%
I: 2019-03-17 22:15:20.434 DMR TX Level: 50.0%
I: 2019-03-17 22:15:20.434 YSF TX Level: 50.0%
I: 2019-03-17 22:15:20.434 P25 TX Level: 50.0%
I: 2019-03-17 22:15:20.434 NXDN TX Level: 50.0%
I: 2019-03-17 22:15:20.434 POCSAG TX Level: 50.0%
I: 2019-03-17 22:15:20.434 RX Frequency: 433475000Hz (433475000Hz)
I: 2019-03-17 22:15:20.434 TX Frequency: 439475000Hz (439475000Hz)
M: 2019-03-17 22:15:20.434 Opening the MMDVM
I: 2019-03-17 22:15:22.445 MMDVM protocol version: 1, description: MMDVM_HS_Dual_Hat-v1.4.16 20190211 14.7456MHz dual ADF7021 FW by CA6JAU GitID #8cf46ca
I: 2019-03-17 22:15:22.465 Display Parameters
I: 2019-03-17 22:15:22.465 Type: None
W: 2019-03-17 22:15:22.465 No valid display found, disabling
I: 2019-03-17 22:15:22.465 DMR Network Parameters
I: 2019-03-17 22:15:22.465 Address: 127.0.0.1
I: 2019-03-17 22:15:22.465 Port: 62031
I: 2019-03-17 22:15:22.465 Local: 62032
I: 2019-03-17 22:15:22.465 Jitter: 360ms
I: 2019-03-17 22:15:22.465 Slot 1: enabled
I: 2019-03-17 22:15:22.465 Slot 2: enabled
I: 2019-03-17 22:15:22.465 Mode Hang: 1s
I: 2019-03-17 22:15:22.466 Info Parameters
I: 2019-03-17 22:15:22.466 Callsign: EA3BIL
I: 2019-03-17 22:15:22.466 RX Frequency: 433475000Hz
I: 2019-03-17 22:15:22.466 TX Frequency: 439475000Hz
I: 2019-03-17 22:15:22.466 Power: 1W
I: 2019-03-17 22:15:22.466 Latitude: 41.556023deg N
I: 2019-03-17 22:15:22.466 Longitude: 2.033664deg E
I: 2019-03-17 22:15:22.466 Height: 0m
I: 2019-03-17 22:15:22.466 Location: "Terrassa JN11an"
[... 100 lines cut ...]
I: 2019-03-17 22:17:27.990 Mapping File: /usr/local/etc/RSSI.dat
I: 2019-03-17 22:17:27.991 Loaded 14 RSSI data mapping points from /usr/local/etc/RSSI.dat
I: 2019-03-17 22:17:27.991 DMR Id Lookups
I: 2019-03-17 22:17:27.991 File: /usr/local/etc/DMRIds.dat
I: 2019-03-17 22:17:27.991 Reload: 24 hours
I: 2019-03-17 22:17:28.324 Loaded 125330 Ids to the DMR callsign lookup table
I: 2019-03-17 22:17:28.324 DMR RF Parameters
I: 2019-03-17 22:17:28.324 Id: 214338103
I: 2019-03-17 22:17:28.324 Started the DMR Id lookup reload thread
I: 2019-03-17 22:17:28.324 Color Code: 2
I: 2019-03-17 22:17:28.324 Self Only: no
I: 2019-03-17 22:17:28.324 Embedded LC Only: no
I: 2019-03-17 22:17:28.324 Dump Talker Alias Data: yes
I: 2019-03-17 22:17:28.324 Prefixes: 0
I: 2019-03-17 22:17:28.324 Call Hang: 1s
I: 2019-03-17 22:17:28.324 TX Hang: 1s
I: 2019-03-17 22:17:28.324 Mode Hang: 1s
M: 2019-03-17 22:17:28.324 MMDVMHost-20181107_Pi-Star is running
M: 2019-03-17 22:17:38.361 DMR, Logged into the master successfully
M: 2019-03-17 22:17:48.724 DMR Slot 2, received network late entry from 4714 to TG 8
M: 2019-03-17 22:17:55.327 DMR Slot 2, received network end of voice transmission, 6.6 seconds, 0% packet loss, BER: 0.0%

NOT HANGED.

Ghosts ???

Rafa-RGB, 2019-03-17:
BTW...

Here you have one of the FW updates log dump:

pi-star@bitxa(rw):~$ sudo pistar-mmdvmhshatflash hs_dual_hat
MMDVM_HS_Dual_Hat-v1.4.16 20190211 14.7456MHz dual ADF7021 FW by CA6JAU GitI
D #8cf46ca                                                                  
Press any key to write the hs_dual_hat firmware to this modem or Ctrl-C to q
uit...                                                                      
Flashing your hs_dual_hat modem to the latest version                       
--2019-03-17 23:20:55--  http://github.com/juribeparada/MMDVM_HS/releases/do
wnload/v1.4.16/install_fw_hsdualhat.sh                                      
Resolving github.com (github.com)... 140.82.118.4, 140.82.118.3             
Connecting to github.com (github.com)|140.82.118.4|:80... connected.        
HTTP request sent, awaiting response... 301 Moved Permanently               
Location: https://github.com/juribeparada/MMDVM_HS/releases/download/v1.4.16
/install_fw_hsdualhat.sh [following]                                        
--2019-03-17 23:20:55--  https://github.com/juribeparada/MMDVM_HS/releases/d
ownload/v1.4.16/install_fw_hsdualhat.sh                                     
Connecting to github.com (github.com)|140.82.118.4|:443... connected.       
HTTP request sent, awaiting response... 302 Found                           
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/80
589711/c6a35c00-32a8-11e9-8b96-6725abb78633?X-Amz-Algorithm=AWS4-HMAC-SHA256
&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20190317%2Fus-east-1%2Fs3%2Faws4_re
quest&X-Amz-Date=20190317T222055Z&X-Amz-Expires=300&X-Amz-Signature=358683cf
3e5e350bc417e86a4a969fb78fc10594784f36a097b74f5c5cc9dc37&X-Amz-SignedHeaders
=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dins
tall_fw_hsdualhat.sh&response-content-type=application%2Foctet-stream [follo
wing]                                                                       
--2019-03-17 23:20:56--  https://github-production-release-asset-2e65be.s3.a
mazonaws.com/80589711/c6a35c00-32a8-11e9-8b96-6725abb78633?X-Amz-Algorithm=A
WS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20190317%2Fus-east-1
%2Fs3%2Faws4_request&X-Amz-Date=20190317T222055Z&X-Amz-Expires=300&X-Amz-Sig
nature=358683cf3e5e350bc417e86a4a969fb78fc10594784f36a097b74f5c5cc9dc37&X-Am
z-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%2
0filename%3Dinstall_fw_hsdualhat.sh&response-content-type=application%2Focte
t-stream                                                                    
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-pr
oduction-release-asset-2e65be.s3.amazonaws.com)... 52.216.144.35            
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (githu
b-production-release-asset-2e65be.s3.amazonaws.com)|52.216.144.35|:443... co
nnected.                                                                    
HTTP request sent, awaiting response... 200 OK                              
Length: 2942 (2.9K) [application/octet-stream]                              
Saving to: ‘/tmp/mmdvmhshatfirmware/flash.sh’                               
                                                                            
/tmp/mmdvmhshatfir 100%[==================>]   2.87K  --.-KB/s   in 0.001s  
                                                                            
2019-03-17 23:20:56 (5.13 MB/s) - ‘/tmp/mmdvmhshatfirmware/flash.sh’ saved [
2942/2942]                                                                  
                                                                            
FINISHED --2019-03-17 23:20:56--                                            
Total wall clock time: 2.1s                                                 
Downloaded: 1 files, 2.9K in 0.001s (5.13 MB/s)                             
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Curr
ent                                                                         
                                 Dload  Upload   Total   Spent    Left  Spee
d                                                                           
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--    
100   614    0   614    0     0   1344      0 --:--:-- --:--:-- --:--:--  13
46                                                                          
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--    
100 50260  100 50260    0     0  40861      0  0:00:01  0:00:01 --:--:-- 985
49                                                                          
Cloning into 'STM32F10X_Lib'...                                             
remote: Enumerating objects: 15, done.                                      
remote: Counting objects: 100% (15/15), done.                               
remote: Compressing objects: 100% (14/14), done.                            
remote: Total 193 (delta 1), reused 6 (delta 1), pack-reused 178            
Receiving objects: 100% (193/193), 3.29 MiB | 3.03 MiB/s, done.             
[... 11 lines cut ...]
Option 2     : 0x00                                                         
Device ID    : 0x0410 (Medium-density)                                      
- RAM        : 20KiB  (512b reserved by bootloader)                         
- Flash      : 128KiB (sector size: 4x1024)                                 
[... cut ...]
ar System...                                                                

And... Surprise:
Got this on the LOG winodw while FW updating:

E: 2019-03-17 22:21:00.719 Error from select(), errno=4
E: 2019-03-17 22:21:00.719 Error when reading from the modem
M: 2019-03-17 22:21:00.719 Closing the MMDVM
I: 2019-03-17 22:21:01.342 Stopped the DMR Id lookup reload thread
M: 2019-03-17 22:21:01.342 DMR, Closing DMR Network
I: 2019-03-17 22:21:01.342 MMDVMHost-20181107_Pi-Star exited on receipt of SIGTERM

juribeparada, 2019-03-24:
Can you enable another additional mode (to enable scan mode), and test again?
I still can't reproduce your problem with a duplex GPIO board that I have.

Rafa-RGB, 2019-03-24:
Hi Andy,

I'll do, even I have no other DV modes than DMR equipment...

Thanks for help.

Rafa.
EA3BIL.

Rafa-RGB, 2019-03-25:
Hi Andy,

Actived YSF as mentioned...

Unfortunatelly same results.

It was working for a few hours and then, after last entry on the log, it
"hooked ptt in DMR".

M: 2019-03-25 16:54:44.647 DMR Slot 2, Embedded Talker Alias Block 1
M: 2019-03-25 16:54:44.647 0000: 05 00 00 20 20 20 00 00 00 *... ...*
M: 2019-03-25 16:56:23.459 DMR Slot 2, received network end of voice
transmission, 100.6 seconds, 0% packet loss, BER: 0.0%
M: 2019-03-25 16:56:31.348 DMR Slot 2, received network voice header from
EA2OZ to TG 21460
M: 2019-03-25 16:56:32.075 DMR Talker Alias (Data Format 1, Received 6/12
char): 'EA2OZ '
M: 2019-03-25 16:56:32.076 DMR Slot 2, Embedded Talker Alias Header
M: 2019-03-25 16:56:32.076 0000: 04 00 58 45 41 32 4F 5A 20 *..XEA2OZ *
M: 2019-03-25 16:56:32.801 DMR Talker Alias (Data Format 1, Received 12/12
char): 'EA2OZ Javier'
M: 2019-03-25 16:56:32.801 DMR Slot 2, Embedded Talker Alias Block 1
M: 2019-03-25 16:56:32.801 0000: 05 00 4A 61 76 69 65 72 00 *..Javier.*
M: 2019-03-25 16:59:00.865 DMR Slot 2, received network end of voice
transmission, 150.2 seconds, 0% packet loss, BER: 0.0%
M: 2019-03-25 16:59:06.040 DMR Slot 2, received network voice header from
EA3HCL to TG 21460
M: 2019-03-25 16:59:06.708 DMR Talker Alias (Data Format 1, Received 6/10
char): 'EA3HCL'
M: 2019-03-25 16:59:06.709 DMR Slot 2, Embedded Talker Alias Header
M: 2019-03-25 16:59:06.709 0000: 04 00 54 45 41 33 48 43 4C *..TEA3HCL*
M: 2019-03-25 16:59:07.426 DMR Talker Alias (Data Format 1, Received 6/10
char): 'EA3HCL'
M: 2019-03-25 16:59:07.426 DMR Slot 2, Embedded Talker Alias Block 1
M: 2019-03-25 16:59:07.426 0000: 05 00 00 20 20 20 00 00 00 *... ...*
M: 2019-03-25 16:59:56.120 DMR Slot 2, received network end of voice
transmission, 50.2 seconds, 0% packet loss, BER: 0.0%
M: 2019-03-25 17:00:03.821 DMR Slot 2, received network voice header from
EA2OZ to TG 21460
M: 2019-03-25 17:00:04.558 DMR Talker Alias (Data Format 1, Received 6/12
char): 'EA2OZ '
M: 2019-03-25 17:00:04.558 DMR Slot 2, Embedded Talker Alias Header
M: 2019-03-25 17:00:04.558 0000: 04 00 58 45 41 32 4F 5A 20 *..XEA2OZ *
M: 2019-03-25 17:00:05.276 DMR Talker Alias (Data Format 1, Received 12/12
char): 'EA2OZ Javier'
M: 2019-03-25 17:00:05.277 DMR Slot 2, Embedded Talker Alias Block 1
M: 2019-03-25 17:00:05.277 0000: 05 00 4A 61 76 69 65 72 00 *..Javier.*
M: 2019-03-25 17:00:24.123 DMR Slot 2, received network end of voice
transmission, 20.3 seconds, 0% packet loss, BER: 0.0%
M: 2019-03-25 17:00:28.860 DMR Slot 2, received network voice header from
EA3HCL to TG 21460
M: 2019-03-25 17:00:29.530 DMR Talker Alias (Data Format 1, Received 6/10
char): 'EA3HCL'
M: 2019-03-25 17:00:29.530 DMR Slot 2, Embedded Talker Alias Header
M: 2019-03-25 17:00:29.530 0000: 04 00 54 45 41 33 48 43 4C *..TEA3HCL*
M: 2019-03-25 17:00:30.241 DMR Talker Alias (Data Format 1, Received 6/10
char): 'EA3HCL'
M: 2019-03-25 17:00:30.241 DMR Slot 2, Embedded Talker Alias Block 1
M: 2019-03-25 17:00:30.241 0000: 05 00 00 20 20 20 00 00 00 *... ...*
M: 2019-03-25 17:00:31.548 DMR Slot 2, received network end of voice
transmission, 2.8 seconds, 0% packet loss, BER: 0.0%
M: 2019-03-25 17:00:35.590 DMR Slot 2, received network voice header from
EA2OZ to TG 21460
M: 2019-03-25 17:00:36.331 DMR Talker Alias (Data Format 1, Received 6/12
char): 'EA2OZ '
M: 2019-03-25 17:00:36.331 DMR Slot 2, Embedded Talker Alias Header
M: 2019-03-25 17:00:36.331 0000: 04 00 58 45 41 32 4F 5A 20 *..XEA2OZ *
M: 2019-03-25 17:00:37.050 DMR Talker Alias (Data Format 1, Received 12/12
char): 'EA2OZ Javier'
M: 2019-03-25 17:00:37.050 DMR Slot 2, Embedded Talker Alias Block 1
M: 2019-03-25 17:00:37.050 0000: 05 00 4A 61 76 69 65 72 00 *..Javier.*
M: 2019-03-25 17:00:37.892 DMR Slot 2, received network end of voice
transmission, 2.3 seconds, 0% packet loss, BER: 0.0%

 Once again...

Thanks for your patiente and helping.

Rafa.

*De:* Andy CA6JAU <notifications@github.com>
*Enviado el:* domingo, 24 de marzo de 2019 16:37
*Para:* juribeparada/MMDVM_HS <MMDVM_HS@noreply.github.com>
*CC:* Rafa-RGB <ea3bil@gmail.com>; Author <author@noreply.github.com>
*Asunto:* Re: [juribeparada/MMDVM_HS] 1.4.16: Overflow in the DMR slot 2 RF
queue with 14.7456 MHz Dual HS-Hat (#94)

Can you enable another additional mode (to enable scan mode), and test
again?
I still can't reproduce your problem with a duplex GPIO board that I have.

juribeparada, 2019-03-25:
I thought your problem happens very often, not after a "few hours", that is very different...and difficult to reproduce. It looks like a firmware crash. What is your MMDVM_HS hardware?. What are the LEDs status when the problem happens?. Have you tried to compile the latest firmware by yourself?

Rafa-RGB, 2019-03-26:
Hi Andy,
Yes it shows quite often but also after a long perior of time working fine (more than 12 hours).
I assume it's difficult to reproduce...

MMDVM HW:
The same "chinese" as Roger.
LED Stastus:
PWR: ON
SVC: Flashing (as usual)
COS: OFF
PTT: ON
DMR: ON
all others OFF.

I've not yet tryed to compile as it was working "fine" on previous versions.

BTW.
Here you have the log for a new try:
BOLD the RX that "hooked" the system.
ITALIC the one that "un-hooked" it.

M: 2019-03-26 11:52:57.432 DMR Slot 2, received network late entry from 4714 to TG 8
M: 2019-03-26 11:53:04.023 DMR Slot 2, received network end of voice transmission, 13.1 seconds, 0% packet loss, BER: 0.0%
M: 2019-03-26 11:53:12.124 Downlink Activate received from EA3BIL
M: 2019-03-26 11:53:12.384 DMR Slot 2, received RF voice header from EA3BIL to TG 8
M: 2019-03-26 11:53:14.362 DMR Slot 2, received RF end of voice transmission, 1.8 seconds, BER: 0.4%, RSSI: -47/-47/-47 dBm
M: 2019-03-26 11:53:50.314 DMR Slot 2, received network voice header from EA3BIL to TG 8
M: 2019-03-26 11:53:51.215 DMR Slot 2, received network end of voice transmission, 0.8 seconds, 0% packet loss, BER: 1.1%
**M: 2019-03-26 11:53:57.661 DMR Slot 2, received network voice header from EA3BIL to TG 8
M: 2019-03-26 11:53:58.205 DMR Slot 2, received network end of voice transmission, 0.5 seconds, 0% packet loss, BER: 0.9%**
_M: 2019-03-26 11:58:12.890 DMR Slot 2, received network voice header from EA3BIL to TG 8
M: 2019-03-26 11:58:14.154 DMR Slot 2, received network end of voice transmission, 1.2 seconds, 0% packet loss, BER: 1.3%_

juribeparada, 2019-03-26:
Well, if SVC is still flashing, then we can say it is not a firmware crash (so far), this is more close to a firmware bug.
Please, try to compile the latest firmware source code and see. We need to be sure your problem is not a compiler related issue. Depending on what you will find, we can continue narrowing.

Rafa-RGB, 2019-05-14:
Hi...

After all this time testing and cheking I found the following:

It depends on What YSF StartUp Host I try to connect I get those overflows...
Even another weird behave:
Connecting notification from DMR+ reflector does NOT take the usual time to get TX'd but about 2 or even 3 times longer, getting up to 20 seconds for a speech not longer than 7 seconds (when it all works fine)...

So, I decided to not using DMR<->YSF crossmode (even PiStar shows system connected to YSF server) and... Surprise: 
No overflows showing, no slow-motion notifications from DMR+ Conection notification, NO PTT hooked never again.

Having reached this point (and as Bugs Bunny used to say)...
Thats all folks!!

Research of defective working has no reason to keep taking time/efforts.

Thank you all for your patiente, comments and helping.

73 & Radio !!

closed 2019-05-14 by Rafa-RGB