juribeparada/MMDVM_HS #95: Upgrading from 1.0 to 1.3 dual hat board, not entering boot loader

hb9eue, 2019-03-20:
Hello

As mentioned in a previous bug report, my 1.0 board has swapped TX/RX Antennas and poor reception. So I ordered a new 1.3 board from Aliexpress, which now has arrived. It seems to work and mmdvmhost sees ancient firmware 1.3.7 which does not support POCSAG. So time to flash 1.4.16...

# ./STM32F10X_Lib/utils/rpi32/stm32flash -v -w mmdvm_hs_dual_hat_fw.bin -g 0x0 -R -i 20,-21,21:-20,21 /dev/ttyAMA0
stm32flash Arduino_STM32_0.9

http://github.com/rogerclarkmelbourne/arduino_stm32

Using Parser : Raw BINARY
Interface serial_posix: 57600 8E1
Failed to init device.

Trying to flash my 1.0 hat from 1.4.14 to 1.4.16 on the same pi zero works perfectly fine.

Google tells me, that this could be due to the GPIO Port 20/21 not being connected, that is not the case.
I also read the issue about poor solder joints on the STM32 chip itself. I'm not sure if I shall try to re-solder them as they are very small!

I see there is a 'boot' header and a 'reset' switch. I did try the 'reset' swtich, but this does not enter boot mode. Is there a way to enter the boot mode via boot header?

Also there is a USB socket on the 1.3 board, what is it's purpose?

Any other suggestions how this hat could be flashed?
[image]

-Benoit- / HB9EUE

hb9eue, 2019-03-20:
Lesson learned, NEVER try to re-solder an STM32. Only with the littlest pressure the pins get bend and you lost.... ordering the next one :-(

hb9eue, 2019-03-22:
Just stumbled over a facebook post of another user of the same board with exact the same problem. There somebody pointed out, that you have to enable boot mode by bridging the JP1 solder joints. Well I guess I'll have to wait for a couple of weeks for the next board to be shipped to me and then try again.

ecolive19, 2019-04-24:
I have both boards and I confirm you need to short JP1 to be able to update v1.3 board's firmware. Just solder JP1 two points together and you should be good to go.

badtime, 2020-01-15:
> 
> 
> I have both boards and I confirm you need to short JP1 to be able to update v1.3 board's firmware. Just solder JP1 two points together and you should be good to go.

Is a little bit incorrect, because before the lastest update i flashed the dual hat (the one with usb) without phisically jump the boot!
In the flashing command the gpio pin 20 is to enable boot mode and 21 to reset the board, and this worked fine until fw1.3.x.
Now..if gpio pin 20 is pulled up, the board wont reset with 21 enabled, otherwise, if i do the flash command without 20 the board with 21 raised up reset correctly the board.
Something is changed in last release.

closed 2020-11-09 by juribeparada