juribeparada/MMDVM_HS #100: Could not flash image with bootloader enabled

shawnchain, 2019-05-14:
```
././STM32F10X_Lib/utils/macosx/st-flash write ./STM32F10X_Lib/utils/bootloader/generic_boot20_pc13_long_rst.bin 0x8000000
2019-05-14T15:54:48 INFO /Users/kuwatay/src/stlink-master/src/common.c: Loading device parameters....
2019-05-14T15:54:48 INFO /Users/kuwatay/src/stlink-master/src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2019-05-14T15:54:48 INFO /Users/kuwatay/src/stlink-master/src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2019-05-14T15:54:48 INFO /Users/kuwatay/src/stlink-master/src/common.c: Attempting to write 7180 (0x1c0c) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08001c00 erased
2019-05-14T15:54:48 INFO /Users/kuwatay/src/stlink-master/src/common.c: Finished erasing 8 pages of 1024 (0x400) bytes
2019-05-14T15:54:48 INFO /Users/kuwatay/src/stlink-master/src/common.c: Starting Flash write for VL/F0/F3 core id
2019-05-14T15:54:48 INFO /Users/kuwatay/src/stlink-master/src/flash_loader.c: Successfully loaded flash loader in sram
  7/7 pages written
2019-05-14T15:54:49 INFO /Users/kuwatay/src/stlink-master/src/common.c: Starting verification of write complete
2019-05-14T15:54:49 INFO /Users/kuwatay/src/stlink-master/src/common.c: Flash written and verified! jolly good!
sleep 3
././STM32F10X_Lib/utils/macosx/st-flash write bin/mmdvm_f1bl.bin 0x8002000
2019-05-14T15:54:52 INFO /Users/kuwatay/src/stlink-master/src/common.c: Loading device parameters....
2019-05-14T15:54:52 INFO /Users/kuwatay/src/stlink-master/src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2019-05-14T15:54:52 INFO /Users/kuwatay/src/stlink-master/src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2019-05-14T15:54:52 INFO /Users/kuwatay/src/stlink-master/src/common.c: Attempting to write 57912 (0xe238) bytes to stm32 address: 134225920 (0x8002000)
2019-05-14T15:54:52 ERROR /Users/kuwatay/src/stlink-master/src/common.c: addr too high
stlink_fwrite_flash() == -1
make: *** [stlink-bl] Error 255
```

shawnchain, 2019-05-14:
It seems `57912 + 8192 = 66,104` which exceeds the max 64KiB of stm32f1 flash

rogerclarkmelbourne, 2019-05-14:
Its a bit of a hack, but virtually all STM32F103C8 devices are actually STM32F103CB and have 128k flash, because ST's production yields are so high.

You could change the linker script and assume that the MCU has 128k

closed 2020-11-09 by juribeparada