juribeparada/MMDVM_HS #102: error compiling with arduino ide

ua6ahp, 2019-06-06:
Hi!
I compile with arduino ide 1.8.5
and 
I get an error when choosing #define STM32_I2C_HOST
what could be the reason?

listing of error:

sketch\I2CHost.cpp: In function 'void I2C2_ER_IRQHandler()':

I2CHost.cpp:32: error: 'I2C2' was not declared in this scope

     if (I2C_GetITStatus(I2C2, I2C_IT_AF)) {

                         ^

I2CHost.cpp:32: error: 'I2C_IT_AF' was not declared in this scope

     if (I2C_GetITStatus(I2C2, I2C_IT_AF)) {

                               ^

I2CHost.cpp:32: error: 'I2C_GetITStatus' was not declared in this scope

     if (I2C_GetITStatus(I2C2, I2C_IT_AF)) {

                                        ^

I2CHost.cpp:33: error: 'I2C_ClearITPendingBit' was not declared in this scope

       I2C_ClearITPendingBit(I2C2, I2C_IT_AF);

                                            ^

sketch\I2CHost.cpp: In member function 'void CI2CHost::Init()':

I2CHost.cpp:44: error: 'GPIO_InitTypeDef' was not declared in this scope

   GPIO_InitTypeDef  GPIO_InitStructure;

   ^

I2CHost.cpp:44: error: expected ';' before 'GPIO_InitStructure'

   GPIO_InitTypeDef  GPIO_InitStructure;

                     ^

I2CHost.cpp:45: error: 'NVIC_InitTypeDef' was not declared in this scope

   NVIC_InitTypeDef NVIC_InitStructure;

   ^

I2CHost.cpp:45: error: expected ';' before 'NVIC_InitStructure'

   NVIC_InitTypeDef NVIC_InitStructure;

                    ^

I2CHost.cpp:46: error: 'I2C_InitTypeDef' was not declared in this scope

   I2C_InitTypeDef  I2C_InitStructure;

   ^

I2CHost.cpp:46: error: expected ';' before 'I2C_InitStructure'

   I2C_InitTypeDef  I2C_InitStructure;

                    ^

I2CHost.cpp:48: error: 'RCC_APB1Periph_I2C2' was not declared in this scope

   RCC_APB1PeriphClockCmd(RCC_APB1Periph_I2C2, ENABLE);

                          ^

I2CHost.cpp:48: error: 'ENABLE' was not declared in this scope

   RCC_APB1PeriphClockCmd(RCC_APB1Periph_I2C2, ENABLE);

                                               ^

I2CHost.cpp:48: error: 'RCC_APB1PeriphClockCmd' was not declared in this scope

   RCC_APB1PeriphClockCmd(RCC_APB1Periph_I2C2, ENABLE);

                                                     ^

I2CHost.cpp:49: error: 'RCC_APB2Periph_GPIOB' was not declared in this scope

   RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOB, ENABLE);

                          ^

I2CHost.cpp:49: error: 'RCC_APB2PeriphClockCmd' was not declared in this scope

   RCC_APB2PeriphClockCmd(RCC_APB2Periph_GPIOB, ENABLE);

                                                      ^

I2CHost.cpp:50: error: 'RCC_APB2Periph_AFIO' was not declared in this scope

   RCC_APB2PeriphClockCmd(RCC_APB2Periph_AFIO, ENABLE);

                          ^

I2CHost.cpp:53: error: 'GPIO_InitStructure' was not declared in this scope

   GPIO_InitStructure.GPIO_Pin =  GPIO_Pin_10 | GPIO_Pin_11;

   ^

I2CHost.cpp:53: error: 'GPIO_Pin_10' was not declared in this scope

   GPIO_InitStructure.GPIO_Pin =  GPIO_Pin_10 | GPIO_Pin_11;

                                  ^

I2CHost.cpp:53: error: 'GPIO_Pin_11' was not declared in this scope

   GPIO_InitStructure.GPIO_Pin =  GPIO_Pin_10 | GPIO_Pin_11;

                                                ^

I2CHost.cpp:54: error: 'GPIO_Speed_50MHz' was not declared in this scope

   GPIO_InitStructure.GPIO_Speed = GPIO_Speed_50MHz;

                                   ^

I2CHost.cpp:55: error: 'GPIO_Mode_AF_OD' was not declared in this scope

   GPIO_InitStructure.GPIO_Mode = GPIO_Mode_AF_OD;

                                  ^

I2CHost.cpp:56: error: 'GPIO_Init' was not declared in this scope

   GPIO_Init(GPIOB, &GPIO_InitStructure);

                                       ^

I2CHost.cpp:59: error: 'NVIC_InitStructure' was not declared in this scope

   NVIC_InitStructure.NVIC_IRQChannel                   = I2C2_EV_IRQn;

   ^

I2CHost.cpp:59: error: 'I2C2_EV_IRQn' was not declared in this scope

   NVIC_InitStructure.NVIC_IRQChannel                   = I2C2_EV_IRQn;

                                                          ^

I2CHost.cpp:63: error: 'NVIC_Init' was not declared in this scope

   NVIC_Init(&NVIC_InitStructure);

                                ^

I2CHost.cpp:66: error: 'I2C2_ER_IRQn' was not declared in this scope

   NVIC_InitStructure.NVIC_IRQChannel = I2C2_ER_IRQn;

                                        ^

I2CHost.cpp:70: error: 'I2C_InitStructure' was not declared in this scope

   I2C_InitStructure.I2C_Mode = I2C_Mode_I2C;

   ^

I2CHost.cpp:70: error: 'I2C_Mode_I2C' was not declared in this scope

   I2C_InitStructure.I2C_Mode = I2C_Mode_I2C;

                                ^

I2CHost.cpp:71: error: 'I2C_DutyCycle_2' was not declared in this scope

   I2C_InitStructure.I2C_DutyCycle = I2C_DutyCycle_2;

                                     ^

I2CHost.cpp:73: error: 'I2C_Ack_Enable' was not declared in this scope

   I2C_InitStructure.I2C_Ack = I2C_Ack_Enable;

                               ^

I2CHost.cpp:74: error: 'I2C_AcknowledgedAddress_7bit' was not declared in this scope

   I2C_InitStructure.I2C_AcknowledgedAddress = I2C_AcknowledgedAddress_7bit;

                                               ^

I2CHost.cpp:78: error: 'I2C2' was not declared in this scope

   I2C_Cmd(I2C2, ENABLE);

           ^

I2CHost.cpp:78: error: 'I2C_Cmd' was not declared in this scope

   I2C_Cmd(I2C2, ENABLE);

                       ^

I2CHost.cpp:80: error: 'I2C_Init' was not declared in this scope

   I2C_Init(I2C2, &I2C_InitStructure);

                                    ^

I2CHost.cpp:82: error: 'I2C_IT_EVT' was not declared in this scope

   I2C_ITConfig(I2C2, I2C_IT_EVT, ENABLE);

                      ^

I2CHost.cpp:82: error: 'I2C_ITConfig' was not declared in this scope

   I2C_ITConfig(I2C2, I2C_IT_EVT, ENABLE);

                                        ^

I2CHost.cpp:83: error: 'I2C_IT_BUF' was not declared in this scope

   I2C_ITConfig(I2C2, I2C_IT_BUF, ENABLE);

                      ^

I2CHost.cpp:84: error: 'I2C_IT_ERR' was not declared in this scope

   I2C_ITConfig(I2C2, I2C_IT_ERR, ENABLE);

                      ^

sketch\I2CHost.cpp: In member function 'void CI2CHost::I2C_EVHandler()':

I2CHost.cpp:94: error: 'I2C2' was not declared in this scope

   uint32_t event = I2C_GetLastEvent(I2C2);

                                     ^

I2CHost.cpp:94: error: 'I2C_GetLastEvent' was not declared in this scope

   uint32_t event = I2C_GetLastEvent(I2C2);

                                         ^

I2CHost.cpp:97: error: 'I2C_EVENT_SLAVE_RECEIVER_ADDRESS_MATCHED' was not declared in this scope

     case I2C_EVENT_SLAVE_RECEIVER_ADDRESS_MATCHED:

          ^

I2CHost.cpp:99: error: 'I2C_EVENT_SLAVE_BYTE_RECEIVED' was not declared in this scope

     case I2C_EVENT_SLAVE_BYTE_RECEIVED:

          ^

I2CHost.cpp:101: error: 'I2C_ReceiveData' was not declared in this scope

         rxFIFO[rxFIFO_head] = I2C_ReceiveData(I2C2);

                                                   ^

I2CHost.cpp:106: error: 'I2C_ReceiveData' was not declared in this scope

         I2C_ReceiveData(I2C2);

                             ^

I2CHost.cpp:108: error: 'I2C_EVENT_SLAVE_TRANSMITTER_ADDRESS_MATCHED' was not declared in this scope

     case I2C_EVENT_SLAVE_TRANSMITTER_ADDRESS_MATCHED:

          ^

I2CHost.cpp:109: error: 'I2C_EVENT_SLAVE_BYTE_TRANSMITTED' was not declared in this scope

     case I2C_EVENT_SLAVE_BYTE_TRANSMITTED:

          ^

I2CHost.cpp:111: error: 'I2C_SendData' was not declared in this scope

         I2C_SendData(I2C2, txFIFO[txFIFO_tail]);

                                               ^

I2CHost.cpp:116: error: 'I2C_SendData' was not declared in this scope

         I2C_SendData(I2C2, 0U);

                              ^

I2CHost.cpp:118: error: 'I2C_EVENT_SLAVE_STOP_DETECTED' was not declared in this scope

     case I2C_EVENT_SLAVE_STOP_DETECTED:

          ^

sketch\I2CHost.cpp: In member function 'void CI2CHost::I2C2_ClearFlag()':

I2CHost.cpp:126: error: 'I2C2' was not declared in this scope

   while((I2C2->SR1 & I2C_SR1_ADDR) == I2C_SR1_ADDR) {

          ^

I2CHost.cpp:126: error: 'I2C_SR1_ADDR' was not declared in this scope

   while((I2C2->SR1 & I2C_SR1_ADDR) == I2C_SR1_ADDR) {

                      ^

I2CHost.cpp:132: error: 'I2C2' was not declared in this scope

   while((I2C2->SR1 & I2C_SR1_STOPF) == I2C_SR1_STOPF) {

          ^

I2CHost.cpp:132: error: 'I2C_SR1_STOPF' was not declared in this scope

   while((I2C2->SR1 & I2C_SR1_STOPF) == I2C_SR1_STOPF) {

                      ^

exit status 1
'I2C2' was not declared in this scope

rogerclarkmelbourne, 2019-06-06:
I don't think this is the problem, but Arduino 1.8.5 is not the current version.

I rebuilt MMSVM_HS using Arduino 1.8.7 2 days ago and it was OK.

Looking at the last errors in the list, for I2C_STR1_STOPF

This is defined here.

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/9d46a1c27d4dc9dd4df06b76a1d81684519d8acc/STM32F1/system/libmaple/include/libmaple/i2c.h#L165

So I suspect you've not actually installed the correct Arduino STM32 core.
You may have installed the new official STM Arduino core which uses the STM HAL rather than the LibMaple based core, which is a lighter weight core which was originally written by LeafLabs

ua6ahp, 2019-06-07:
Please refer to the correct Arduino STM32 core.

I used:
https://github.com/rogerclarkmelbourne/Arduino_STM32

Maybe I misunderstood the kernel?
If possible, refer to the detailed manual to configure it!

I don’t think it affects, but I’ve additionally installed support for boards digispark attiny85.

I am confused by an error:
'I2C2' was not declared in this scope
I could not understand where I2C2 is declared!

Thanks for your help and attention to my problem.

ua6ahp, 2019-06-07:
I tried on a clean win7(64). I installed the Arduino IDE 1.8.7 like you. I installed it on the Arduino SAM Boards board manager. Added vayla from https://github.com/rogerclarkmelbourne/Arduino_STM32 to the hardware folder (as in the installation description). Just in case, I placed a folder with a sketch in the root of the C: drive. Again got an error: 'I2C2' was not declared in this scope.

Full listing of errors:
In file included from sketch\Globals.h:63:0,

                 from sketch\ADF7021.cpp:28:

sketch\IO.h:28:0: warning: "LOW" redefined [enabled by default]

 #define LOW      0

 ^

In file included from C:\Users\ev\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/wirish.h:55:0,

                 from C:\Users\ev\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/Arduino.h:30,

                 from sketch\Globals.h:34,

                 from sketch\ADF7021.cpp:28:

C:\Users\ev\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/io.h:113:0: note: this is the location of the previous definition

 #define LOW  0x0

 ^

In file included from sketch\Globals.h:63:0,

                 from sketch\ADF7021.cpp:28:

sketch\IO.h:29:0: warning: "HIGH" redefined [enabled by default]

 #define HIGH     1

 ^

In file included from C:\Users\ev\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/wirish.h:55:0,

                 from C:\Users\ev\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/Arduino.h:30,

                 from sketch\Globals.h:34,

                 from sketch\ADF7021.cpp:28:

C:\Users\ev\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/io.h:112:0: note: this is the location of the previous definition

 #define HIGH 0x1

 ^

In file included from sketch\Globals.h:63:0,

                 from sketch\CWIdTX.cpp:22:

sketch\IO.h:28:0: warning: "LOW" redefined [enabled by default]

 #define LOW      0

 ^

In file included from C:\Users\ev\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/wirish.h:55:0,

                 from C:\Users\ev\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/Arduino.h:30,

                 from sketch\Globals.h:34,

                 from sketch\CWIdTX.cpp:22:

C:\Users\ev\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/io.h:113:0: note: this is the location of the previous definition
[... 1240 lines cut ...]

                 from sketch\YSFTX.cpp:21:

sketch\IO.h:29:0: warning: "HIGH" redefined [enabled by default]

 #define HIGH     1

 ^

In file included from C:\Users\ev\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/wirish.h:55:0,

                 from C:\Users\ev\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/Arduino.h:30,

                 from sketch\Globals.h:34,

                 from sketch\YSFTX.cpp:21:

C:\Users\ev\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/io.h:112:0: note: this is the location of the previous definition

 #define HIGH 0x1

 ^

exit status 1
'I2C2' was not declared in this scope

rogerclarkmelbourne, 2019-06-08:
Sounds like you have seriously screwed up your installation somehow.

The Arduino IDE has a load of hidden cached data in C:\Users\rclark\AppData\Local\Arduino15

You may have some sort of conflict with another version of the Arduino core in there.

rogerclarkmelbourne, 2019-06-08:
I'm using the Master branch of Arduino32, because the USB Serial was changed a while ago, so it should support MMDVM_HS

closed 2020-11-09 by juribeparada