juribeparada/MMDVM_HS #106: Help Selecting Correct Board For Build And Firmware

dh47, 2019-10-15:
Hello, I ordered the following board of bangood and need help making sure I build and compile and also get the right firmware for the following.

`https://www.banggood.com/USB-Communication-Duplex-MMDVM-Hotspot-Support-P25-DMR-YSF-2PCS-Antenna-For-Raspberry-Pi-p-1360474.html`

Which firmware should I select to flash?

Thanks.

rogerclarkmelbourne, 2019-10-15:
The GitHub issues are for reporting bugs.

Try posting to https://forum.pistar.uk/

dh47, 2019-10-15:
Okay.

dh47, 2019-10-15:
Reopeing as I'm having issues signing up to that forum.  Could an admin please take a look please?  User name there is Dark_Hunter.

phl0, 2019-10-15:
The admins of that forum are probably not reading github issues for the firmware.

dh47, 2019-10-15:
Okay, I'll try the email.

closed 2019-10-15 by dh47