juribeparada/MMDVM_HS #114: Crossband mode

BlueJohnnyOnly, 2020-01-08:
On my duplex hat I have tested VHF only and UHF only split frequencies and works fine.

If I try to set duplex hat TX to VHF and RX to UHF  or  TX to UHF and RX to VHF, 
the duplex will TX on set frequency but it will not RX on set frequency.

rogerclarkmelbourne, 2020-01-28:
Are you sure the problem is the Hotspot ?

Most DMR radios do not work crossband, because of errors in their firmware.

BlueJohnnyOnly, 2020-01-29:
Yes, re-tested and confirmed it the dual hat hotspot, with separate TX and RX simplex mobile radios.

rogerclarkmelbourne, 2020-01-29:
So. You tested the DMR 4FSK deviation on your transceiver when its working crossband ?

BlueJohnnyOnly, 2020-02-01:
Nothing wrong with transceivers.
The crossband radio used can be hear on other monitoring radios at the time off testing. 
It's the Hotspot Dual which fails to receive UHF signals when Transmit is VHF and fails to receive VHF signals when transmit is set to UHF. Wide range of offsets has been tested. 
Hotspot Dual all work fine as normal repeater mode with TX/RX function is UHF and work fine as normal repeater TX/RX function is VHF