juribeparada/MMDVM_HS #118: Nextion screen firmware upload / generic serial.

rawbengal, 2020-05-03:
I have a Nextion screen wired to the MMDVM modem and would like to be able to upload firmware to it without having to remove the display from its case or reconnect it. From what I can tell, the existing serial repeater code is very specific to Nextion packets that end in 3 0xFF characters. Unfortunately, the Nextion firmware upload does not send back packets formatted this way. Instead it sends back a 0x05 byte for every 4096 bytes of raw data sent during the upload.

The upload protocol also allows for one to specify a different baud rate for the data transfer.

1) Perhaps I am misreading the code that handles the modem serial port but it seems very Nextion specific rather than just acting as a generic serial interface.
2) From my reading of the code, it would not be possible to use the serial repeater to upload new firmware to the Nextion because it only handles frames that end in 3 0xFF chars.
3) There does not seem to be support for dynamic setting of the modem serial port baud rate. This means any desire to use a different rate requires recompilation.

I propose that the serial repeater interface be modified to allow more generic support of serial data. This would include being able to specify a host-specified end-frame character (or none, meaning, send characters as quickly as possible), specification of the serial interface parameters dynamically. The goal would be to support Nextion firmware updates via the serial interface.

I would be happy to code this up and submit a pull request but wanted to make sure I'm not missing something in my analysis of the current situation. Perhaps this is already implemented?

rawbengal, 2020-05-03:
Adding @plainpylut since they implemented the original Nextion receive support and might have comment.

rawbengal, 2020-05-03:
In reading the code again, I just noticed this comment in SerialPort.cpp

```
      //  TODO - BG5HHP
      //   modem serial data repeat should be generic instead of coupling with the nextion protocol.
```

Looks like he/she agrees with my concern... Adding @shawnchain for comment before I go down the rabbit-hole of making these changes.

SS396, 2020-05-03:
Hi,

I was one of the individuals who worked on getting the serial repeater function to work properly in 1.4.17.  shawnchain did some further optimizing on the code I created.  I had originally just patterned it off of similar functions, so I'm not quite sure why shawnchain optimized just my code and didn't bother to optimize all the rest of the code I patterned it after.

> I have a Nextion screen wired to the MMDVM modem and would like to be able to upload firmware to it without having to remove the display from its case or reconnect it. From what I can tell, the [...]

In order for the Nextion to have the ability to update the firmware over the serial port, you would also have to create entirely new communication packets similar to MMDVMHost.  I'm not sure if you are going to run into an issue with size or with the handoff between UARTS.

For me it is just easier to update the screen using a SD card or if you really want to have that ability to just use a USB to TTL adapter which currently supports what you are wanting.

> 1. Perhaps I am misreading the code that handles the modem serial port but it seems very Nextion specific rather than just acting as a generic serial interface.

Well, that may be the case but there isn't any other serial traffic being passed from MMDVMHost.  The messages are all formatted Nextion specific, and I don't think anyone has used a generic serial device hooked to that port for anything.

> 2. From my reading of the code, it would not be possible to use the serial repeater to upload new firmware to the Nextion because it only handles frames that end in 3 0xFF chars.

Yes, I figured that would be the best way to know for sure that the Nextion was finished sending its serial data.

Maybe it isn't the best way to handle it, but it works and that was my only goal, since earlier versions of the code just erased any serial data it received so when a touch happened on the device nothing was ever passed to MMDVMHost.

> 3. There does not seem to be support for dynamic setting of the modem serial port baud rate. This means any desire to use a different rate requires recompilation.

I don't think that is possible, I originally wanted something like that but when I started poking around at the code the speeds are hardcoded at the time of creation.  

I have tested the serial data to a Nextion at speeds from 9600 baud all the way up to 921600 baud with no issues at all.  Currently my screens run at 921600 baud.

If you figure out a way to make it dynamic then you might as well do the same for the the first UART since it is locked at 115200 too.  Since the 2nd UART can go faster, I figure the 1st one can too.  For some reason when the Nextion passthrough was created the individual locked it at 9600 on UART 2.  Shrug.
>

> I propose that the serial repeater interface be modified to allow more generic support of serial data. This would include being able to specify a host-specified end-frame character (or none, [...]

I'm not sure if that is possible.  The code loops around and only grabs one character at a time.  I'm not sure how you know when a device is done sending or that you have a valid message ready to send if you don't supply some sort of ending character to trigger from.  If you just send everything you could run into issues with messages being partial or concatenated.

> I would be happy to code this up and submit a pull request but wanted to make sure I'm not missing something in my analysis of the current situation. Perhaps this is already implemented?

I don't think anyone has implemented any of that or if they have they have not made it public.

By all means if you can improve on what others have built on, then go for it.

rawbengal, 2020-05-04:
Thanks for the feedback @SS396. I guess my general thinking is that, even though most uses of the serial repeater will be for Nextion displays, we can anticipate other ways in which people will use/misuse the modem board port. Additionally, since even Nextion doesn't hold firmly to the "every message ends in FFFFFF" rule, we cannot only use that rule if we want a full featured driver.  This feels like a place in which we don't want to hard code a specific use case or operating mode into the firmware. Essentially, the repeater should do just that... repeat.  Since RS232 is not a packet-based protocol, we probably shouldn't enforce that in the firmware. It is up to the host-side software to handle interpretation of the serial stream. This also has the benefit that the same code which handles DIRECT serial port communication with the device will also work when used through the repeater. They both just see a stream of bytes which must be interpreted or broken into packets. In the ideal situation, the modem-based serial port should just look like any other device based serial port and leave it to the host-side program to deal with formatting.  

Now, that said, I understand what you are saying about potential issues with the two UART's on the modem board having contention issues, especially when you consider that the host interface needs to add extra packet information to the stream. This could especially have impact if the baud rates are similar and we are saturating the communication channel.

I'll experiment with it and see what I learn.

SS396, 2020-05-05:
> Additionally, since even Nextion doesn't hold firmly to the "every message ends in FFFFFF" rule, we cannot only use that rule if we want a full featured driver.

In MMDVMHost there are mechanisms in place that disable all return data from the Nextion over the serial port.

https://github.com/g4klx/MMDVMHost/blob/4a923fb27e9789f031052835fdc1c7ef1815204c/Nextion.cpp#L94

So in my testing the only thing that I found on the RX port on UART 2 besides the initial power on message would be a touch screen response that ends in FFFFFF.

Also, in order for the touch screen responses to be passed back into MMDVMHost correctly you need to have the Nextion Driver from ON7LDS installed.

> Essentially, the repeater should do just that... repeat.

If that is the goal then both MMDVMHost and the Nextion Driver are going to also have to be modified in order to function that way too.

shawnchain, 2020-05-05:
I think the serial data repeat via MMDVM MCU has limitations. Small packets will be fine to be pass through, but it's not a good idea to transfer long-lasting, huge data at high speed. 

The quick reason for that is the modem firmware has no RTOS and you have to do a lot of work to support slow(compare to the mcu speed and ISR handler) USART I/O read/write.

rawbengal, 2020-05-05:
This is all good feedback. Thanks.

The current behavior of the MMDVM firmware is to buffer up all return data from the Nextion until it sees FFFFFF and then send ALL data back. The only time it does not send ALL the data back is on an empty message (just FFFFFF). With the exception of this corner case, the firmware is already sending all the bytes back. The change of behavior I am describing is to allow a host-specified mode in which the buffering on return bytes is turned off.  This would allow a single 05 character (an ACK for 4K bytes during firmware upload) to pass without being held up. The default behavior would be the current, buffered, behavior so no other software should need to change.

The current firmware, will ONLY support a Nextion device to be connected to the modem serial port and will not allow firmware update. The change I propose will allow either. The MMDVM Host software will continue to need to be Nextion-specific. "Generic" serial device support comes for free from implementing support for Nextion firmware upload.

As far as @shawnchain's concerns about speed and amount of data, it seems that there is some practical limit on both of these.

I have a simple fix to the firmware which will let me test this behavior out. I suspect Shawn is right about throughput issues though, so I think I'll hold off on any further discussion. I'm also prepping for my general class license upgrade so I need to spend some time on that. Wish me luck. :)

rd0fun, 2021-01-09:
> This is all good feedback. Thanks.
>
> The current behavior of the MMDVM firmware is to buffer up all return data from the Nextion until it sees FFFFFF and then send ALL data back. The only time it [...]

Just curious if you ever implemented the change?

The only thing requiring people to use TTL to USB converters is updating the Nextion screen. It would be awesome if a small change enabled this! Most cases do not allow for easy removal of the screen to insert (then remove) an SD Card. This would be a VERY welcome change for sure!