juribeparada/MMDVM_HS #122: Is the latest beta from Dec 2019 the last development on the MMDVM firmware?

ecolive19, 2020-07-09:
Hi, first of all many thanks for this project.
This is not a bug by itself, I am just wondering whether the development effort is now considered completed or not. I see a beta firmware dated DEC 2019 and since there were frequent updates before, I am just checking whether this is the final code for this project. Thanks.

n2aws, 2020-07-09:
@ecolive19 I'm not sure if this answer helps or not:  but if you look at the commit history for the master branch, you can see new changes were added on may 6th, of this year.  However, those changes haven't been added to an actual firmware release.

My guess is,  either this has been shifted lower on the priority list (maybe newer hardware is the priority or something) OR The feature set is close enough to what was envisioned for the project.. and there just isn't that much more to do that is a priority.  But I absolutely agree with your assesment,  it does appear that active development has slowed significantly recently.

juribeparada, 2020-07-09:
Well, actually I don't have much free time for development due to my work load from some time. Then, I am only accepting PR for supporting new boards o small improvements. Apart from that, certainly the code has reached a good "stability" and there are not big or urgent issues so far. Obviously, some improvements could do to this code, but have less priority now. Moreover, we are very close to the limits of RAM/Flash of the MCU, which translate to have difficulties to add new "big" features and maintain the compatibility of all hardware.

cyberjew, 2020-07-19:
Why not take the code out of beta and change the release from 1.5.1b to 1.5.1?

I've been running 1.5.1b on an N5BOC duplex and a couple of simplex Zumspot GPIO boards since it was released back in December and it's been rock solid for me.

Your reasons for not releasing newer builds since December are 100% legitimate. You might just want to publish that on the main page for the GitHub project so you set expectations appropriately. Otherwise, unless someone happens to read this thread, they won't know what's going on and why there haven't been any new releases.

Just my $0.02...

cyberjew, 2020-08-19:
Or not... :-)

juribeparada, 2020-11-09:
Now we have fw 1.5.2 :)

closed 2020-11-09 by juribeparada