juribeparada/MMDVM_HS #126: Revisit COS Timing Issue with Duplex MMDVMs

JeffHochberg, 2020-09-22:
The issue documented in [https://github.com/juribeparada/MMDVM_HS/issues/70] - Duplex problem when replying to over after leaving a pause for other stations - needs to be revisited.

I own a duplex MMDVM and have been experiencing an issue where it does not respond to keying after another station ends its transmission. I contacted the manufacturer and they clued me into this issue.

I initially thought it was an issue with the RXOffset/TXOffset values. I adjusted them to the point where my BER is incredibly low (at most 0.02% to 0.05%). I am using Motorola DMR radios (XPR 7550, XPR 5550e, and XPR 4550). I have verified that the alignment in all of my radios is accurate.

I've also tried adjusting the offset and hang values and I don't see any improvement. The only thing that helps is waiting a second or two for the COS LED to illuminate. Having to monitor the status of the COS LED is not realistic as it's not reasonable to assume the user will be close to their hotspot.

@juribeparada I understand that you are busy and have not been able to return to development for quite some time. The last official build was in May of 2019 and the most recent beta build was in December 2019. Is there anyone else involved in the development of the MMDVM firmware or is the MMDVM_HS project dead at this point?

rd0fun, 2021-01-19:
I have experienced the same issue.

JeffHochberg, 2021-01-22:
It's made my duplex MMDVM unusable. It would be nice to understand if it's
something that will be addressed or if it's not even being considered. I'd
prefer to know what the intent is that rather than have the issue left
open/unresolved.

stevelenti, 2021-02-06:
With the n5boc duplex board I am seeing the same issue.  I have been able to mitigate this with using these settings https://www.k2ie.net/blog/2020/05/08/duplex-hotspot-reliability-revisited/

It still happens but not nearly as much and makes the board very usable in duplex mode.  Now the n5boc boards are much better quality than their Chinese counterparts so your results might not be as good using the settings above.

I'm not a programmer but since I can adjust settings to mitigate the issue it seems to me that there must be a way to program a fix into the firmware.

-Steve

macroexp, 2022-10-07:
@stevelenti the settings you linked to made a huge difference for me. My MD-380 only fails handshake maybe 5% of the time now, compared to nearly 100% of the time before changing settings. I am going to keep fiddling to figure out what the critical setting is, but thanks for the link.

stevelenti, 2022-10-26:
> @stevelenti the settings you linked to made a huge difference for me. My MD-380 only fails handshake maybe 5% of the time now, compared to nearly 100% of the time before changing settings. I am going to keep fiddling to figure out what the critical setting is, but thanks for the link.

@macroexp Try these settings. You might need to adjust the RXOffset and TXOffset for what gives you the best BER for your board.  These settings have been the best hands down with absolutely no issues.  Let me know if it works for you.
[image]
[image]

-Steve

RdWing, 2024-08-06:
Just like to comment that I am also experiencing this issue. With a BI7JTA SMA duplex board, the issue persists, but on a BI7JTA ceramic antenna duplex mini board it almost never occurs.
Really a big pain. Not just occasionally missing a TX but I also have a hard time getting the BI7JTA SMA duplex board to key up sometimes. I have replicated this across many different radios, Motorola XPR7550, XPR6550, Anytone 878. It is not a frequency calibration issue, all radios and hotspots are aligned using my station GPSDO and a spectrum analyzer and various calibration routines.