juribeparada/MMDVM_HS #136: Unable to upgrade to  v1.5.2, keeps telling me v1.5.2 is v1.47.1

aa3jc, 2021-01-07:
Not sure if I am doing something wrong, but here is what I used in SSH and the results.   "Failed to init device"  I even did it twice, the second time checking that the folder was empty prior to downloading the file again. I've double checked that it is the correct modem and it is a mmdvm dual hat 14.7456mhz. Any ideas ??

https://github.com/juribeparada/MMDVM_HS/releases/download/v1.5.2/mmdvm_hs_dual_hat_fw.bin

pi-star@aa3jc-06(rw):~$ sudo pistar-mmdvmhshatflash-custom mmdvm_hs_dual_hat_fw.bin  
MMDVM_HS_Dual_Hat-v1.4.17 20190529 14.7456MHz dual ADF7021 FW by CA6JAU GitID #cc451c
4                                                                                    
Press any key to write the mmdvm_hs_dual_hat_fw.bin firmware to this modem or Ctrl-C 
to quit...                                                                           
Cloning into 'STM32F10X_Lib'...                                                      
remote: Enumerating objects: 193, done.                                              
remote: Total 193 (delta 0), reused 0 (delta 0), pack-reused 193                     
Receiving objects: 100% (193/193), 3.28 MiB | 6.02 MiB/s, done.                      
Resolving deltas: 100% (71/71), done.                                                
Raspberry Pi 3 detected                                                              
stm32flash Arduino_STM32_0.9                                                         
                                                                                     
http://github.com/rogerclarkmelbourne/arduino_stm32                                  
                                                                                     
Using Parser : Raw BINARY                                                            
Interface serial_posix: 57600 8E1                                                    
Failed to init device.                                                               
                                                                                     
                                                                                     
Flashing mmdvm_hs_dual_hat_fw.bin to your modem complete, press any key to reboot you
r Pi-Star System...                                                                  
Session closed.

rd0fun, 2021-01-19:
"Failed to init device."

Your flash is not set for write. You need to enable write with JP1 or the BOOT header.

IW8ELN, 2021-04-10:
If that doesn't work, you can use STLink and programmer by connecting to the appropriate PINs on the board.