juribeparada/MMDVM_HS #139: Jumbospot PCB 1.7 no VHF mode

IW8ELN, 2021-04-08:
On PCB version 1.7 the VHF mode no longer works, only UHF works.
The firmware works correctly and MMDVMHost does not detect problems, but the Jumbospot does not receive or transmit in VHF.

g4klx, 2021-04-08:
The JumboSpot is an illegal copy of someone else’s design. Please contact your supplier about it.

IW8ELN, 2021-04-08:
Interesting answer, but it doesn't solve the problem, I was expecting a productive response, not an idiotic one.

g4klx, 2021-04-08:
It’s not idiotic. If you use an illegal product don’t expect answers from those who worked hard to create the legal one originally. Contact your supplier.

IW8ELN, 2021-04-08:
Are you talking about copied or non-original things? Very strange.

Anyway, I'll take your advice and ask the manufacturer.
In the meantime I check the differences on the PCB.

g4klx, 2021-04-08:
All JumboSpots are illegal copies of a German design of MMDVM_HS. They’ll soon be useless anyway 😀 Only legal ones will be OK.

IW8ELN, 2021-04-08:
Great, then we'll buy "original" cards.

cwfuselier, 2021-04-08:
Johnathan,                 Just for the sake of commenting, the 1.7 version is a single band and I prefer the first version which does a variety of bands. I have both the original German board and the original Jumbospot and the Jumbospot out performs and is more reliable than the original German MMDVM_HS. After using the Jumbospot for awhile I personally think that the original is not a good investment. On that note it the Jumbospot also does VHF which works great without flaw, in which the original designers are to lazy to include. I do all most nothing on that worthless UHF band. It isn't right to exclude other designer from even modifying and exiting product in the free marketplace. If someone would try to exclude the other boards from being use will be hurting the Ham community as a whole. Thanks for your time. 73's"Our greatest weakness lies in giving up. The most certain way to succeed is always to try just one more time." Thomas A. Edison Sent from my T-Mobile 4G LTE DeviceThis email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to whom they are addressed. If you have received this email in error please notify the system manager. This message contains confidential information and is intended only for the individual named. If you are not the named addressee you should not disseminate, distribute or copy this e-mail. Please notify the sender immediately by e-mail if you have received this e-mail by mistake and delete this e-mail from your system. If you are not the intended recipient you are notified that disclosing, copying, distributing or taking any action in reliance on the contents of this information is strictly prohibited.

g4klx, 2021-04-08:
The JumboSpot is an illegal copy of the German design, that’s all there is to say about it. If it were an independent design, such as hardware from BI7JTA, then I would say nothing. Whether it works better or worse is irrelevant. I have a German hotspot board and it works very well, I have other boards also.

Asking for support for an illegal product is not very ethical.

cwfuselier, 2021-04-08:
Johnathan,         I don't see an ethical issue here at all maybe more of a personal opinion. And the 1st Gen Jumbospot board doesn't even have the same board layout as the German board. So I'm not sure where you get its a copy. But I can agree with the Ver 1.7 because it's not the same as the first board and does not functionthe same. V 1.7 is more like the German board as to the layout and the restrictions in functionality as the German board. Hope you have a great day. 73's"Our greatest weakness lies in giving up. The most certain way to succeed is always to try just one more time." Thomas A. Edison Sent from my T-Mobile 4G LTE DeviceThis email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to whom they are addressed. If you have received this email in error please notify the system manager. This message contains confidential information and is intended only for the individual named. If you are not the named addressee you should not disseminate, distribute or copy this e-mail. Please notify the sender immediately by e-mail if you have received this e-mail by mistake and delete this e-mail from your system. If you are not the intended recipient you are notified that disclosing, copying, distributing or taking any action in reliance on the contents of this information is strictly prohibited.

IW8ELN, 2021-04-10:
Following the problems encountered in VHF, and since with this post it was evident that no one was able to solve the problem, I studied the whole scheme of the card in question, I tried to modify the bootloader and firmware of the ST32, I replaced the Radio Chip, I checked the entire pinout of the jumbospot and now I have solved the problem.
The coil L6 placed between the PIN 44 and 46 of the radio chip ADF7021, has been inserted with the wrong value.
The right value must be 18nH and instead a 1.5uH was inserted.
I replaced it and now the MMDVM works correctly even in VHF.
I share with everyone because I have no commercial agreements with any MMDVM card manufacturer.

closed 2021-04-10 by IW8ELN