juribeparada/MMDVM_HS #143: MMDVM Does Not Hear FTM-7250D

cwawak, 2021-08-29:
Detected MMDVM_HS (GPIO): /dev/ttyAMA0 (MMDVM_HS_Hat-v1.4.12 20181104 14.7456MHz ADF7021 FW by CA6JAU GitID #dec7051)

Hi, I have a MMDVM_HS_Hat plugged into an RPi Zero. I have an FTM-7250D that is only heard by MMDVM on its first transmission after powering up. After that, no further transmissions are heard. This radio works perfectly on repeaters, simplex, etc. I also have an FT3D and FTM-300 that work perfectly with the MMDVM. I tried a factory reset, as well as rolling back to an earlier firmware (1.4.12), and up to the latest firmware. The behavior doesn't change. The FT3D and 7250 have been tested using the same antenna.

Here's what I see in the logs when I transmit on the FT3D:

```
M: 2021-08-29 21:12:47.268 YSF, received RF header from KC2IEB     to DG-ID 0
M: 2021-08-29 21:12:52.468 YSF, received RF end of transmission from KC2IEB     to DG-ID 0, 5.3 seconds, BER: 0.3%, RSSI: -47/-47/-47 dBm
M: 2021-08-29 21:12:54.587 YSF, received network data from KC2IEB     to DG-ID 0 at KC2IEB
M: 2021-08-29 21:12:59.783 YSF, received network end of transmission from KC2IEB     to DG-ID 0, 5.3 seconds, 0% packet loss, BER: 0.0%

```

Here's the first time the 7250D transmits. 

```
M: 2021-08-29 21:13:25.381 YSF, received RF header from KC2IEB     to DG-ID 0
M: 2021-08-29 21:13:26.773 YSF, transmission lost from KC2IEB     to ALL       , 1.4 seconds, BER: 3.0%, RSSI: -47/-47/-47 dBm
M: 2021-08-29 21:13:30.301 YSF, received network data from KC2IEB     to DG-ID 0 at KC2IEB
M: 2021-08-29 21:13:33.100 YSF, network watchdog has expired, 1.4 seconds, 0% packet loss, BER: 0.0%
```

Every time it fails after about 1.4 seconds no matter how long I transmit with the radio. The radio must be fully powered down and powered on for it to be "heard". Any subsequent transmissions are not acknowledged by MMDVM. Again, on the FT3D I can hear the entire transmission directly from the radio, so it's a valid data stream to the FT3D. The FT3D has no problem interacting with MMDVM even after transmissions from the 7250.

Let me know what other information I can provide to help debug this.

cwawak, 2021-08-30:
Here is the 7250 transmitting after power cycle, in mmdvm debug mode.

```
M: 2021-08-29 23:59:47.924 Debug: YSFRX: sync found in None
M: 2021-08-29 23:59:48.027 YSF, received RF header from KC2IEB     to DG-ID 0
M: 2021-08-29 23:59:48.029 Debug: YSFRX: found sync in Data, pos 0
M: 2021-08-29 23:59:48.033 Debug: Mode set to System Fusion
M: 2021-08-29 23:59:48.129 Debug: YSFRX: found sync in Data, pos 0
M: 2021-08-29 23:59:48.229 Debug: YSFRX: found sync in Data, pos 0
M: 2021-08-29 23:59:48.330 Debug: YSFRX: found sync in Data, pos 0
M: 2021-08-29 23:59:48.430 Debug: YSFRX: found sync in Data, pos 0
M: 2021-08-29 23:59:48.529 Debug: YSFRX: found sync in Data, pos 0
M: 2021-08-29 23:59:48.631 Debug: YSFRX: found sync in Data, pos 0
M: 2021-08-29 23:59:48.829 Debug: YSFRX: found sync in Data, pos 0
M: 2021-08-29 23:59:49.318 Debug: YSFRX: sync timed out, lost lock
M: 2021-08-29 23:59:49.325 YSF, transmission lost from KC2IEB     to ALL       , 1.3 seconds, BER: 4.4%, RSSI: -47/-47/-47 dBm
M: 2021-08-29 23:59:52.843 YSF, received network data from KC2IEB     to DG-ID 0 at KC2IEB
M: 2021-08-29 23:59:55.546 YSF, network watchdog has expired, 1.3 seconds, 0% packet loss, BER: 0.0%

```

cwawak, 2021-08-31:
Adjusting TXOffset in the MMDVM config seeemed to help.

RXOffset of 0, I get a 0% BER with the FT3D, and 3+% BER with the 7250. 
RXOffset of -200 I get 0.7% BER with FT3D, 4+% BER with 7250
RXOffset of -400 i get 1.7% BER with FT3D, 1,3% BER with 7250
RXOffset of -375 I get 1.4% BER with FT3D, 1.4% BER with 7250

Closing this since the solution was fiddling w/ RXOffset.

closed 2021-08-31 by cwawak

Twilight-Logic, 2021-09-01:
I am glad that sorted it. I ended up selling and replacing my FT1D with another one. The replacement worked perfectly fine.